zephyrproject-rtos/zephyr #68578: doc: Documentation issue in 'releases/migration-guide-3.6' for ili9xxx drivers

everedero, 2024-02-05:
**Describe the bug**

In the ili9xxx drivers section, the exemple shows reset and dc gpios with the same polarity before and after, whereas in the conversion commit, they are all inverted (https://github.com/zephyrproject-rtos/zephyr/commit/1cfed06929087701fec8edcabbc78f92ddcad926).

`reset-gpios = <&gpio0 6 GPIO_ACTIVE_LOW>;`
`reset-gpios = <&gpio0 6 GPIO_ACTIVE_HIGH>;`

**Environment**

* Page: `releases/migration-guide-3.6`
* Version: main
* SHA-1: 5bb8ffb51785833b26d7b39c6ac210a5f439d019

kartben, 2024-02-06:
Related to #68563

danieldegrasse, 2024-02-06:
Thanks for catching this. I noticed this issue as well, #68563 should fix the problem. Can you confirm that looks right to you @everedero?

everedero, 2024-02-06:
I checked on my board, the screen is displaying when the nRST pin is high, so reset is active low, and it now matches what is written in the device tree. Looks good to me!

closed 2024-02-06 by everedero