zephyrproject-rtos/zephyr #68580: Phython 3.10.12 venv install failure in Ubuntu 22.04

rrozario1968, 2024-02-05:
[image]

JPHutchins, 2024-02-06:
Hi @rrozario1968, I believe that venv is already installed in all recent distributions of python.  Please proceed with the command to create a venv:
```
python3 -m venv .venv
```

Note here that I named the venv `.venv`.

If that is working for you then please close this ticket.

Cheers,
JP

kartben, 2024-02-06:
What install instructions are you following? I would recommend you follow the getting started instructions to the letter, in this case it would have you install `python3-venv`, which I think should get you past the error your having by trying to force a specific version.

henrikbrixandersen, 2024-02-06:
> What install instructions are you following? I would recommend you follow the getting started instructions to the letter, in this case it would have you install `python3-venv`, which I think should get you past the error your having by trying to force a specific version.

I can confirm that following the getting started instructions and using `sudo apt-get install python3-venv` succeeds on Ubuntu 22.04.

@rrozario1968 Please reopen this issue with further details if this does not work for you.

closed 2024-02-06 by henrikbrixandersen

rrozario1968, 2024-02-06:
> Caution: This e-mail originated outside Infineon Technologies. Do not click on links or open attachments unless you validate it is safe<https://intranet-content.infineon.com/explore/aboutinfineon/rules/informationsecurity/ug/SocialEngineering/Pages/SocialEngineeringElements_en.aspx>.

Hi,
I tried it in different ways but it would not install venv. I tried to follow the guide word for word and it did not work.
Best Regards,
Robert

kartben, 2024-02-06:
As mentioned, please share more details (and actual text, not screenshots, please) if you want help. It is virtually impossible to troubleshoot your problem if all we know is that "[you] tried it in different ways but it would not install". Thanks!