zephyrproject-rtos/zephyr #68600: ATSAME70Q21 Ethernet Issue: Can't assign IP address, subnet, etc.

Julieanna97, 2024-02-06:
Hello everyone, I'm pretty new to Zephyr, so configuring it has been a bit difficult. I'm encountering an issue with my SAM E70 board running Zephyr OS. It's not getting assigned an IP address despite detecting the Ethernet connection and displaying its speed. I used one of the samples called DHCPV4_CLIENT to test (no configuration) but I get this following result: 
[image]
[image]

I've checked the system logs, and there are no error messages related to Ethernet or IP configuration. The only relevant information I see is that the Ethernet link is up and running at 100 Mbps.

I would appreciate any insights or suggestions on how to troubleshoot this issue further. Has anyone encountered a similar problem with the SAM E70 or Zephyr OS.

Thank you in advance for any help you can provide.

jukkar, 2024-02-06:
Everything looks ok from network point of view in your screen capture. Do you get connectivity with static IP addresses, could you try to flash `samples/net/sockets/echo-server` sample application and try that?

Julieanna97, 2024-02-07:
Sorry for the late response, had some trouble with the internet yesterday.

This is the output I get from running echo_server sample:

[image]

Interface 1 coming up.... that's it...

Julieanna97, 2024-02-07:
```
net stats

Interface 0x204006c0 (Ethernet) [1]
===================================
IPv4 recv      33       sent    0       drop    33      forwarded       0
IP vhlerr      0        hblener 0       lblener 0
IP fragerr     0        chkerr  0       protoer 0
ICMP recv      0        sent    0       drop    3
ICMP typeer    0        chkerr  0
IGMP recv      0        sent    0       drop    0
UDP recv       0        sent    0       drop    33
UDP chkerr     0
TCP bytes recv 0        sent    0       resent  0
TCP seg recv   0        sent    0       drop    0
TCP seg resent 0        chkerr  0       ackerr  0
TCP seg rsterr 0        rst     0
TCP conn drop  0        connrst 0
TCP pkt drop   0
Bytes received 19906
Bytes sent     0
Processing err 31
```

Julieanna97, 2024-02-07:
I noticed there was a discussion on GitHub related to a similar issue ([link](https://github.com/zephyrproject-rtos/zephyr/pull/22961)). However, the provided solution seems to be specific to an older version of Zephyr. Has anyone encountered a similar issue recently or found a workaround for this problem?

Any insights or suggestions would be greatly appreciated. Thank you!

jukkar, 2024-02-07:
Network interface is up and you can see information about it with `net iface` command.
From statistics, all the packets are dropped but difficult to say why without additional debugging activated.
You could set `CONFIG_NET_IPV4_LOG_LEVEL_DBG=y` to see if it says why the packet is dropped.
Are you able to ping other host from zephyr device? You can use `net ping` command for that.

Julieanna97, 2024-02-07:
Hi jukkar,

I am able to ping hosts according to this output, and it shows that IPv4 packet has been received and is being processed so it seems. It shows the source from 192.x.x.x and destination IP of 255.x.x.x  which seems to be from the router, but to which it sends to, i am not sure of if it's from router to my pc or to sam e70.

Pinging the source of packet:

```
net ping 192.168.1.20
PING 192.168.1.20
[00:00:16.186,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.215 to 255.255.255.255
[00:00:16.599,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.188 to 239.255.255.250
[00:00:16.600,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.188 to 239.255.255.250
[00:00:16.600,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.188 to 239.255.255.250
[00:00:16.614,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.188 to 239.255.255.250
[00:00:16.840,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.159 to 239.255.255.250
[00:00:16.871,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.159 to 224.0.0.251
[00:00:16.953,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.144 to 239.255.255.250
[00:00:17.076,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.37 to 224.0.0.251
[00:00:17.131,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.157 to 224.0.0.22
[00:00:17.131,000] <dbg> net_ipv4: net_ipv4_igmp_input: (rx_q[0]): DROP: IGMPv3 msg received but IGMP support is disabled
[00:00:17.147,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.157 to 224.0.0.22
[00:00:17.148,000] <dbg> net_ipv4: net_ipv4_igmp_input: (rx_q[0]): DROP: IGMPv3 msg received but IGMP support is disabled
[00:00:17.262,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.57 to 255.255.255.255
[00:00:17.603,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.188 to 239.255.255.250
[00:00:17.606,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.188 to 239.255.255.250
[00:00:17.607,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.188 to 239.255.255.250
[00:00:17.618,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.188 to 239.255.255.250
[00:00:17.698,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.13 to 255.255.255.255
[00:00:17.887,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.157 to 224.0.0.22
[00:00:17.887,000] <dbg> net_ipv4: net_ipv4_igmp_input: (rx_q[0]): DROP: IGMPv3 msg received but IGMP support is disabled
[00:00:17.902,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.20 to 224.0.0.251
[00:00:17.949,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.144 to 239.255.255.250
[00:00:18.611,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.188 to 239.255.255.250
[00:00:18.611,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.188 to 239.255.255.250
[00:00:18.611,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.188 to 239.255.255.250
[00:00:18.633,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.188 to 239.255.255.250
[00:00:18.818,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.198 to 224.0.0.251
[00:00:18.841,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.159 to 239.255.255.250
[00:00:18.883,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.159 to 224.0.0.251
[00:00:18.916,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.185 to 224.0.0.251
[00:00:18.917,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.176 to 224.0.0.251
[00:00:18.943,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.198 to 224.0.0.251
[00:00:18.976,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.188 to 224.0.0.251
[00:00:18.976,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.188 to 224.0.0.251
[00:00:18.978,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.144 to 239.255.255.250
[00:00:19.031,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.121 to 224.0.0.251
[00:00:19.048,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.176 to 224.0.0.251
[00:00:19.182,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.238 to 224.0.0.251
[00:00:19.207,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.12 to 224.0.0.251
[00:00:19.207,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.12 to 224.0.0.251
[00:00:19.260,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.196 to 224.0.0.251
[... cut ...]
[00:00:08.273,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.185 to 239.255.255.250
[00:00:09.471,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.159 to 239.255.255.250
[00:00:09.500,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.159 to 224.0.0.251
[00:00:10.079,000] <dbg> net_ipv4: net_ipv4_input: (rx_q[0]): IPv4 packet received from 192.168.1.57 to 255.255.255.255
```

Is there some configuration I need to do in the .overlay or .conf files for specific boards?

Thank you, very much appreciate any help.

jukkar, 2024-02-07:
The packets you receive are IPv4 multicast packets and not related to the ping command. I am not seeing any ping reply in the log. The sample application does not use the same IP address range as what you are using so the communication will probably not work.
I have SAM-E70 board which works ok, not sure if my board is exactly the same as yours.

You could try to compile and flash the sample application like this:
```
west build -p -b sam_e70_xplained samples/net/sockets/echo_server/ -d build --  \
     -DCONFIG_NET_CONFIG_MY_IPV4_ADDR="\"192.168.1.x\"" \
     -DCONFIG_NET_CONFIG_MY_IPV4_GW="\"192.168.1.1\""
```
Please set the IPv4 address in `CONFIG_NET_CONFIG_MY_IPV4_ADDR` to proper value (in this test you can choose some address that is not used in the test). The `CONFIG_NET_CONFIG_MY_IPV4_GW` should point to your default router (I am guessing it is 192.168.1.1 in this example).
Normally you should use DHCPv4 to setup the IP address as you had earlier, but for testing purposes it is nice to try the communication with static address as that leaves out one step that can go wrong.

nandojve, 2024-02-07:
Hi @Julieanna97 ,

Thank you for bring this to us. Initially I would like to mention that there is a warning about CIDR. This means that you are building your software for the from SoC version. You should try to build for sam_e70**b**_xplained. Besides that, I noted that when I build the current mainline the software do not bring up CC: @jukkar. I tested with an not to old version 0bf5cbc1c8 and confirmed that it works fine.

I usually test using the `samples/net/cloud/tagoio_http_post` and additionally put a few more configs besides what is in `prj.conf`.

```
CONFIG_HEAP_MEM_POOL_SIZE=8192
CONFIG_MAIN_STACK_SIZE=8192
CONFIG_IDLE_STACK_SIZE=2048
CONFIG_ISR_STACK_SIZE=4096
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=4096

CONFIG_NET_MAX_ROUTERS=2
CONFIG_NET_TCP_WORKQ_STACK_SIZE=4096
CONFIG_NET_MAX_CONN=8
CONFIG_NET_MAX_CONTEXTS=32
CONFIG_NET_CONTEXT_NET_PKT_POOL=y

CONFIG_NET_PKT_RX_COUNT=16
CONFIG_NET_PKT_TX_COUNT=16
CONFIG_NET_BUF_RX_COUNT=32
CONFIG_NET_BUF_TX_COUNT=32
CONFIG_NET_BUF_DATA_SIZE=256

CONFIG_DNS_RESOLVER_ADDITIONAL_BUF_CTR=2
CONFIG_DNS_RESOLVER_ADDITIONAL_QUERIES=2
CONFIG_DNS_SERVER_IP_ADDRESSES=y
CONFIG_DNS_SERVER1="8.8.8.8"
CONFIG_DNS_NUM_CONCUR_QUERIES=2

CONFIG_NET_SOCKETS_POLL_MAX=8
```

Would you mind to check those configs, adjust and make another test round?

closed 2024-02-08 by jukkar