zephyrproject-rtos/zephyr #68604: undefined reference to `bt_le_per_adv_set_info_transfer' when trying to build samples/bluetooth/periodic_adv_rsp

TimWei888, 2024-02-06:
**Describe the bug**
I followed the "Getting Stared Guild" https://developer.nordicsemi.com/nRF_Connect_SDK/doc/2.5.2/zephyr/develop/getting_started/index.html and build the Blinky sample successfully and flashed it to my nRF52840dk board, but when I tried to build samples/bluetooth/periodic_adv_rsp with the following command I got an error.

**To Reproduce**
west build -p always -b nrf52840dk_nrf52840  samples/bluetooth/periodic_adv_rsp

**Expected behavior**
build successfully

**Impact**
showstopper

**Logs and console output**
C:\Users\tw\zephyrproject\zephyr>west build -p always -b nrf52840dk_nrf52840  samples/bluetooth/periodic_adv_rsp
-- west build: making build dir C:\Users\tw\zephyrproject\zephyr\build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/Users/tw/zephyrproject/zephyr/samples/bluetooth/periodic_adv_rsp
-- CMake version: 3.28.3
-- Found Python3: C:/Users/tw/AppData/Local/Programs/Python/Python311/python.exe (found suitable version "3.11.7", minim
um required is "3.8") found components: Interpreter
-- Cache files will be written to: C:/Users/tw/zephyrproject/zephyr/.cache
-- Zephyr version: 3.6.0-rc1 (C:/Users/tw/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: nrf52840dk_nrf52840
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.5 (C:/Users/tw/zephyr-sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (C:/Users/tw/zephyr-sdk-0.16.5)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: C:/Users/tw/zephyrproject/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Generated zephyr.dts: C:/Users/tw/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/Users/tw/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generate
d.h
-- Including generated dts.cmake file: C:/Users/tw/zephyrproject/zephyr/build/zephyr/dts.cmake

warning: BT_PER_ADV_SYNC_TRANSFER_SENDER (defined at subsys/bluetooth/Kconfig:168) was assigned the
value 'y' but got the value 'n'. Check these unsatisfied dependencies: (!BT_CTLR ||
BT_CTLR_SYNC_TRANSFER_SENDER_SUPPORT) (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_BT_PER_ADV_SYNC_TRANSFER_SENDER and/or look
up BT_PER_ADV_SYNC_TRANSFER_SENDER in the menuconfig/guiconfig interface. The Application
Development Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of
the manual might be helpful too.

Parsing C:/Users/tw/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Users/tw/zephyrproject/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840_defconfig'
Merged configuration 'C:/Users/tw/zephyrproject/zephyr/samples/bluetooth/periodic_adv_rsp/prj.conf'
Configuration saved to 'C:/Users/tw/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/Users/tw/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: c:/users/tw/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zeph
yr-eabi/bin/ld.bfd.exe (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/tw/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
-- Configuring done (6.8s)
-- Generating done (0.4s)
-- Build files have been written to: C:/Users/tw/zephyrproject/zephyr/build
←[92m-- west build: building application
[1/223] Generating include/generated/version.h
-- Zephyr version: 3.6.0-rc1 (C:/Users/tw/zephyrproject/zephyr), build: v3.6.0-rc1
[217/223] Linking C executable zephyr\zephyr_pre0.elf                                                           e.c.obj
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map C:/Users/tw/zephyrproject/zephyr/build/zephyr/zephyr_pre0.map
C:\WINDOWS\system32\cmd.exe /C "cd . && C:\Users\tw\zephyr-sdk-0.16.5\arm-zephyr-eabi\bin\arm-zephyr-eabi-gcc.exe  -gdwa
rf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr\zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./
arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=C:/Users/tw/zephyrproject
/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch_
_common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex
_m.a  zephyr/arch/arch/arm/core/mpu/libarch__arm__core__mpu.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr
/lib/libc/common/liblib__libc__common.a  zephyr/soc/soc/arm/nordic_nrf/libsoc__arm__nordic_nrf.a  zephyr/subsys/random/l
ibsubsys__random.a  zephyr/subsys/bluetooth/common/libsubsys__bluetooth__common.a  zephyr/subsys/bluetooth/host/libsubsy
s__bluetooth__host.a  zephyr/subsys/bluetooth/controller/libsubsys__bluetooth__controller.a  zephyr/subsys/net/libsubsys
__net.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/
drivers/entropy/libdrivers__entropy.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctr
l.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/hal_nordic/nrfx/libmo
dules__hal_nordic__nrfx.a  modules/segger/libmodules__segger.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"c:
/users/tw/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v7e-m/nofp"  -LC:/Users/tw/zephy
rproject/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfp16-f
ormat=ieee  -mtp=soft  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignmen
t  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no
-pie  -DPICOLIBC_LONG_LONG_PRINTF_SCANF  --specs=picolibc.specs  -lc  -lgcc && C:\WINDOWS\system32\cmd.exe /C "cd /D C:\
Users\tw\zephyrproject\zephyr\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -E true""
c:/users/tw/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.b
fd.exe: app/libapp.a(main.c.obj): in function `main':
C:/Users/tw/zephyrproject/zephyr/samples/bluetooth/periodic_adv_rsp/src/main.c:308: undefined reference to `bt_le_per_ad
v_set_info_transfer'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\Users\tw\zephyrproject\zep
hyr\build'

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Zephyr SDK, ...
 - Zephyr version: 3.6.0-rc1

hermabe, 2024-02-06:
> warning: BT_PER_ADV_SYNC_TRANSFER_SENDER (defined at subsys/bluetooth/Kconfig:168) was assigned the
value 'y' but got the value 'n'. Check these unsatisfied dependencies: (!BT_CTLR ||
BT_CTLR_SYNC_TRANSFER_SENDER_SUPPORT) (=n).

Which controller are you using? The zephyr controller (`BT_LL_SW_SPLIT`) does not support Periodic Advertising with Responses (PAwR). If you are building this in nrfconnect context make sure you are using the nordic controller (`BT_LL_SOFTDEVICE`) that supports PAwR.

Questions about nrfconnect should be asked on devzone and not here in zephyr.

TimWei888, 2024-02-07:
Hi Herman,

Thank you for your reply. Looks like I am using BT_LL_SW_SPLIT. 
When I run the C:\Users\tw\zephyrproject\zephyr\samples\bluetooth\periodic_adv_rsp>west build -b nrf52840dk_nrf52840 -t guiconfig, looks like BT_LL_SW_SPLIT is the only option available.

[image]

hermabe, 2024-02-07:
The nordic proprietary controller is not available in upstream zephyr, which it looks like you are using (`3.6.0-rc1` is not in nrfconnect yet).

Questions about nordic proprietary should be asked on nordics devzone instead, closing this as not an issue with zephyr.

closed 2024-02-07 by hermabe

TimWei888, 2024-02-07:
I downloaded the nrfconnect and built these two projects successfully. Thank you for your help.