zephyrproject-rtos/zephyr #68622: I also noticed that "requires-ulbpr" is not implemented in stm32-qspi-nor driver and is needed for used Microchip SST26VF064B

tt0776, 2024-02-06:
From https://github.com/zephyrproject-rtos/zephyr/discussions/68423#discussioncomment-8379966
@erwango 
^^ @FRASTM

I also noticed that "requires-ulbpr" is not implemented in stm32-qspi-nor driver and is needed for used Microchip SST26VF064B.

[st,stm32-qspi-nor](https://docs.zephyrproject.org/latest/build/dts/api/bindings/flash_controller/st,stm32-qspi-nor.html)

That is pressent in as example https://docs.zephyrproject.org/latest/build/dts/api/bindings/mtd/nordic,qspi-nor.html
That looks to be implemented in ST spi driver and I haven't found in the st qspi.

I am currently need it for the Microchip SST26VF that need this feature implemented :-(
Is there a workaround to send command SPI_NOR_CMD_ULBPR (98h)?

tt0776, 2024-02-06:
I a new title with less detail etc. ;-)

kk3008, 2024-02-15:
@FRASTM 
I work together with tt0776 on the same Zephyr based project.

We have now got much further implementing the interface between a stm32l476 based system and a Microchip SST26VF QSpi flash chip running under Zephyr.

As tt077 mentioned earlier, the standard STM32CubeMx stm32l476xx and stm32l486xx libraries does not support the Global Block Protection Unlock instruction (ULBPR - 98H), which is required for being able to program the Microchip flash. 
"qspi is failing in compile stm32l476xx + stm32l486xx ..." (https://github.com/zephyrproject-rtos/zephyr/discussions/68423)

Therefor we have made a git patch, that adds the definitions of QUADSPI_CR_DFM and QUADSPI_CR_FSEL to stm32l476xx.h and stm32l486xx.h, see (https://github.com/zephyrproject-rtos/zephyr/discussions/68423)

Another challenge we have also discovered, is that Microchip SST26VF QSpi flash chips supports the SPI Quad Page Program instruction (32H), but unlike other chip manufacturers, the instruction expect this instruction as 1-4-4, and not 1-1-4 as normal.

Also here we have implemented a git patch, that updates flash_stm32_qspi.c to fix this issue, controlled by a Kconfig symbol: `USE_MICROCHIP_QSPI_FLASH_WITH_STM32`.

I have attached the patch that fixes the problem for us, in the hope that it can be a help for others facing similar problems.

Kconfig.board:
```
config USE_MICROCHIP_QSPI_FLASH_WITH_STM32 
   bool "Include patch for Microchip qspi flash when running with stm32"
   default y
   depends on BOARD_XXXX

```
Patch:
```
diff --git a/drivers/flash/flash_stm32_qspi.c b/drivers/flash/flash_stm32_qspi.c
index 4090cd41bb..b955fc58d9 100644
--- a/drivers/flash/flash_stm32_qspi.c
+++ b/drivers/flash/flash_stm32_qspi.c
@@ -174,9 +174,18 @@ static inline int qspi_prepare_quad_program(const struct device *dev,
 			dev_data->qspi_write_cmd == SPI_NOR_CMD_PP_1_4_4);
 
 	cmd->Instruction = dev_data->qspi_write_cmd;
+
+#if defined(CONFIG_USE_MICROCHIP_QSPI_FLASH_WITH_STM32)
+	if (cmd->Instruction == SPI_NOR_CMD_PP_1_1_4)
+	{
+		cmd->AddressMode = QSPI_ADDRESS_4_LINES;
+	}
+#else
 	cmd->AddressMode = ((cmd->Instruction == SPI_NOR_CMD_PP_1_1_4)
 				? QSPI_ADDRESS_1_LINE
 				: QSPI_ADDRESS_4_LINES);
+#endif
+
 	cmd->DataMode = QSPI_DATA_4_LINES;
 	cmd->DummyCycles = 0;
 
@@ -247,6 +256,41 @@ static int qspi_read_access(const struct device *dev, QSPI_CommandTypeDef *cmd,
 	return dev_data->cmd_status;
 }
 
+#if defined(CONFIG_USE_MICROCHIP_QSPI_FLASH_WITH_STM32)
+static int flash_unprotect(const struct device *dev)
+{
+	int ret = 0;
+
+	QSPI_CommandTypeDef cmd_write_en = {
+		.Instruction = SPI_NOR_CMD_WREN,
+		.InstructionMode = QSPI_INSTRUCTION_1_LINE,
+	};
+
+	QSPI_CommandTypeDef cmd_unprotect = {
+		.Instruction = SPI_NOR_CMD_ULBPR,
+		.InstructionMode = QSPI_INSTRUCTION_1_LINE,
+	};
+
+	ret = qspi_send_cmd(dev, &cmd_write_en);
+
+	if (ret != 0) {
+		printf("\n%s SPI flash unprotect WREN failed: %d\n", dev->name, ret);
+	}
+
+	ret = qspi_send_cmd(dev, &cmd_unprotect);
+
+	if (ret == 0) {
+		printf("\n%s SPI flash unprotect success.\n", dev->name);
+	}
+	else
+	{
+		printf("\n%s SPI flash unprotect failed - status: %d\n", dev->name, ret);
+	}
+	
+	return ( ret );
+}
+#endif
+
 /*
  * Perform a write access over QSPI bus.
  */
@@ -1337,6 +1381,10 @@ static int flash_stm32_qspi_init(const struct device *dev)
 	}
 #endif /* CONFIG_FLASH_PAGE_LAYOUT */
 
+#if defined(CONFIG_USE_MICROCHIP_QSPI_FLASH_WITH_STM32)
+	flash_unprotect(dev);
+#endif
+
 	return 0;
 }
 

```

erwango, 2024-02-27:
@tt0776 Currenlty STM32 QSPI driver binding includes `jedec,jesd216.yaml` binding. See https://github.com/zephyrproject-rtos/zephyr/blob/94b720845d3578da9a340ab14ed8c6a7510fcd36/dts/bindings/flash_controller/st%2Cstm32-qspi-nor.yaml#L21
`requires-ulbpr` property is actually available in `dts/bindings/mtd/jedec,spi-nor-common.yaml` binding, which is a superset of `jedec,jesd216.yaml`.
As explained in the binding, this bindins provides "Common properties used by nodes describing M25P80-compatible SPI NOR
serial flash devices, regardless of which Zephyr driver is being used.".

So, ideally, by including `jedec,spi-nor-common.yaml` instead of `jedec,jesd216.yaml`, STM32 QSPI driver would then support whatever additional properties that are provided by this binding:
In `dts/bindings/flash_controller/st,stm32-qspi-nor.yaml`
```diff
+ include: ["flash-controller.yaml", "jedec,jesd216.yaml"]
- include: ["flash-controller.yaml", "jedec,spi-nor-common.yaml"]
```
The only tricky part in that operation is to be able to validate this is working as expected.

Would you be so kind to have a try to do this binding inclusion modification, define the property as needed on your board, make some tests and if working, contribute this change ?

tt0776, 2024-02-28:
@erwango I will try to check this out with @kk3008
Lets see if that changes, will also "solve", there was a different way Microchip SST26VF QSpi handling write expect this instruction as 1-4-4, and not 1-1-4 as normal.

kk3008, 2024-02-28:
I have now tested your recommendations, and changing “jedec,jesd216.yaml” to “jedec,spi-nor-common.yaml” fixed the problem that we had to implement and call the flash unprotect function in the drivers/flash/flash_stm32_qspi.c.
But the issue with the Microchip SST26VF QSpi flash chips running the SPI Quad Page Program instruction (32H) as 1-4-4, was still there.
I have updated the required patch:
```
diff --git a/drivers/flash/flash_stm32_qspi.c b/drivers/flash/flash_stm32_qspi.c
index 4090cd41bb..1e1afe2c82 100644
--- a/drivers/flash/flash_stm32_qspi.c
+++ b/drivers/flash/flash_stm32_qspi.c
@@ -174,9 +174,18 @@ static inline int qspi_prepare_quad_program(const struct device *dev,
 			dev_data->qspi_write_cmd == SPI_NOR_CMD_PP_1_4_4);
 
 	cmd->Instruction = dev_data->qspi_write_cmd;
+
+#if defined(CONFIG_USE_MICROCHIP_QSPI_FLASH_WITH_STM32)
+	if (cmd->Instruction == SPI_NOR_CMD_PP_1_1_4)
+	{
+		cmd->AddressMode = QSPI_ADDRESS_4_LINES;
+	}
+#else
 	cmd->AddressMode = ((cmd->Instruction == SPI_NOR_CMD_PP_1_1_4)
 				? QSPI_ADDRESS_1_LINE
 				: QSPI_ADDRESS_4_LINES);
+#endif
+
 	cmd->DataMode = QSPI_DATA_4_LINES;
 	cmd->DummyCycles = 0;
 
diff --git a/dts/bindings/flash_controller/st,stm32-qspi-nor.yaml b/dts/bindings/flash_controller/st,stm32-qspi-nor.yaml
index 6d7bf5105e..49d615fd36 100644
--- a/dts/bindings/flash_controller/st,stm32-qspi-nor.yaml
+++ b/dts/bindings/flash_controller/st,stm32-qspi-nor.yaml
@@ -19,7 +19,7 @@ description: |
 
 compatible: "st,stm32-qspi-nor"
 
-include: ["flash-controller.yaml", "jedec,jesd216.yaml"]
+include: ["flash-controller.yaml", "jedec,spi-nor-common.yaml"]
 
 on-bus: qspi

```

erwango, 2024-02-28:
@kk3008 Thanks for the attempt. Would you mind contributing the binding change ?

Regarding the other patch, since it is Microchip specific and not STM32, I'd like the Kconfig to be flash generic (`drivers/flash/Kconfig`).
@de-nordic any insight on this ?

erwango, 2024-02-29:
Thanks for the update. Ok, my bad. I was under the assumption spi_nor.c was a sor of library which we were already calling. Which is wrong, clearly.

> What would you recommend that I do? I can add support for requires-ulbpr in flash_stm32_qspi.c and keep jedec,spi-nor-common.yaml in the st,stm32-qspi-nor.yaml file, or I can go back to my originally fix in flash_stm32_qspi.c.

First option: Rely on device tree bindings, and implement support in driver. Note that you need to include only the property you need with use of [property-allowlist](https://github.com/zephyrproject-rtos/zephyr/blob/e8be0210376e60ef76d6077757a2af0b468851c7/dts/bindings/clock/st%2Cstm32-clock-mux.yaml#L19)

kk3008, 2024-02-29:
@erwango Thank you so much for your help! 
I have now created a patch that defines a `qspi_write_unprotect` function, that uses `requires-ulbpr` to decide if the `ULBPR` command will be send. This function is always called during startup.

```
diff --git a/drivers/flash/flash_stm32_qspi.c b/drivers/flash/flash_stm32_qspi.c
index 4090cd41bb..34ae667746 100644
--- a/drivers/flash/flash_stm32_qspi.c
+++ b/drivers/flash/flash_stm32_qspi.c
@@ -120,6 +120,8 @@ struct flash_stm32_qspi_data {
 	bool flag_access_32bit: 1;
 };
 
+static int qspi_write_unprotect(const struct device *dev);
+
 static inline void qspi_lock_thread(const struct device *dev)
 {
 	struct flash_stm32_qspi_data *dev_data = dev->data;
@@ -174,9 +176,18 @@ static inline int qspi_prepare_quad_program(const struct device *dev,
 			dev_data->qspi_write_cmd == SPI_NOR_CMD_PP_1_4_4);
 
 	cmd->Instruction = dev_data->qspi_write_cmd;
+
+#if defined(CONFIG_USE_MICROCHIP_QSPI_FLASH_WITH_STM32)
+	if (cmd->Instruction == SPI_NOR_CMD_PP_1_1_4)
+	{
+		cmd->AddressMode = QSPI_ADDRESS_4_LINES;
+	}
+#else
 	cmd->AddressMode = ((cmd->Instruction == SPI_NOR_CMD_PP_1_1_4)
 				? QSPI_ADDRESS_1_LINE
 				: QSPI_ADDRESS_4_LINES);
+#endif
+
 	cmd->DataMode = QSPI_DATA_4_LINES;
 	cmd->DummyCycles = 0;
 
@@ -330,6 +341,35 @@ static int qspi_read_jedec_id(const struct device *dev, uint8_t *id)
 }
 #endif /* CONFIG_FLASH_JESD216_API */
 
+static int qspi_write_unprotect(const struct device *dev)
+{
+	int ret = 0;
+
+	if (IS_ENABLED(DT_INST_PROP(0, requires_ulbpr)))
+	{
+		QSPI_CommandTypeDef cmd_write_en = {
+			.Instruction = SPI_NOR_CMD_WREN,
+			.InstructionMode = QSPI_INSTRUCTION_1_LINE,
+		};
+
+		QSPI_CommandTypeDef cmd_unprotect = {
+			.Instruction = SPI_NOR_CMD_ULBPR,
+			.InstructionMode = QSPI_INSTRUCTION_1_LINE,
+		};
+
+		ret = qspi_send_cmd(dev, &cmd_write_en);
+
+		if (ret != 0)
+		{
+			return ret;
+		}
+
+		ret = qspi_send_cmd(dev, &cmd_unprotect);
+	}
+	
+	return ( ret );
+}
+
 /*
  * Read Serial Flash Discovery Parameter
  */
@@ -1337,6 +1377,13 @@ static int flash_stm32_qspi_init(const struct device *dev)
 	}
 #endif /* CONFIG_FLASH_PAGE_LAYOUT */
 
+	ret = qspi_write_unprotect(dev);
+
+	if (ret != 0) {
+		LOG_ERR("write unprotect failed: %d", ret);
+		return -ENODEV;
+	}
+
 	return 0;
 }
 
diff --git a/dts/bindings/flash_controller/st,stm32-qspi-nor.yaml b/dts/bindings/flash_controller/st,stm32-qspi-nor.yaml
index 6d7bf5105e..49d615fd36 100644
--- a/dts/bindings/flash_controller/st,stm32-qspi-nor.yaml
+++ b/dts/bindings/flash_controller/st,stm32-qspi-nor.yaml
@@ -19,7 +19,7 @@ description: |
 
 compatible: "st,stm32-qspi-nor"
 
-include: ["flash-controller.yaml", "jedec,jesd216.yaml"]
+include: ["flash-controller.yaml", "jedec,spi-nor-common.yaml"]
 
 on-bus: qspi
 

```

kk3008, 2024-03-12:
The problem is not directly related to the ST implementation of the qspi driver.

The problem is that the QSPI flash device, SST26VF032B from Microchip we use, does not follow the de facto standard for how command SQI Quad Page Program (0x32) works. It is implemented as a 1-4-4 command, see datasheet: ( https://ww1.microchip.com/downloads/en/devicedoc/20005218e.pdf ) section  5.21 page 29, and not as a 1-1-4 as most other qspi flash devices works. 

And we cannot select PP_1_4_4 for writeoc, because command 0x38 is not supported by the Microchip device.

I hope this explains why we need the patch. But as I have said earlier, I'm very new in working with Zephyr, so I'm sure there are smarter ways to overcome the problem.

FRASTM, 2024-06-19:
I am preparing a branch for the change you described: https://github.com/FRASTM/zephyr/tree/qspi_ulbpr
that I cannot test (no microchip quad-spi NOR flash).

--> Could the  CONFIG_USE_MICROCHIP_QSPI_FLASH_WITH_STM32 also flag the  call to the qspi_write_unprotect() during init (flash_stm32_qspi_init)  ?

No impact seen when running on the stm32l496g_disco : 
west build -p auto -b stm32l496g_disco samples/drivers/jesd216/ -- -DCONFIG_USE_MICROCHIP_QSPI_FLASH_WITH_STM32=y

FRASTM, 2024-06-19:
@kk3008 ,  I see in your proposal
+
+#if defined(CONFIG_USE_MICROCHIP_QSPI_FLASH_WITH_STM32)
+	if (cmd->Instruction == SPI_NOR_CMD_PP_1_1_4)
+	{
+		cmd->AddressMode = QSPI_ADDRESS_4_LINES;
+	}
+#else
 	cmd->AddressMode = ((cmd->Instruction == SPI_NOR_CMD_PP_1_1_4)
 				? QSPI_ADDRESS_1_LINE
 				: QSPI_ADDRESS_4_LINES);
+#endif

but what should be the else part in #if defined(CONFIG_USE_MICROCHIP_QSPI_FLASH_WITH_STM32) ?
```
+	if (cmd->Instruction == SPI_NOR_CMD_PP_1_1_4)
+	{
+		cmd->AddressMode = QSPI_ADDRESS_4_LINES;
+	} else {
+		cmd->AddressMode = ?
+	}
```

kk3008, 2024-06-19:
If I follow the code, the function this change is made in, is qspi_prepare_quad_program. This function is only called from one other function, flash_stm32_qspi_write, and just before qspi_prepare_quad_program is called the AddressMode has been set to QSPI_ADDRESS_1_LINE. The point in this fix, is that the Microchip qspi flash, does not follow the standard rules. As you know Instruction = SPI_NOR_CMD_PP_1_1_4, stands for command is send in 1 line mode, address is send in 1 line mode and data is send in 4 lines mode. But the Microchip flash expect address to be send in 4 lines mode. Earlier in the discussion I have referred to the qspi flash datasheet. I hope this explains the code. But it's fine, if you set the AddressMode to QSPI_ADDRESS_1_LINE in the else part. There might be static analysis programs that does not allow if statements, not to have an else part.

kk3008, 2024-06-19:
Sorry! My suggestion to set the AddressMode to QSPI_ADDRESS_1_LINE in the else part, is of course wrong. If Instruction is different from SPI_NOR_CMD_PP_1_1_4, the AddressMode shall be set accordingly. It can be SPI_NOR_CMD_PP_1_1_2, SPI_NOR_CMD_PP_1_1_4 or SPI_NOR_CMD_PP_1_4_4.

kk3008, 2024-06-19:
> I am preparing a branch for the change you described: https://github.com/FRASTM/zephyr/tree/qspi_ulbpr that I cannot test (no microchip quad-spi NOR flash).
>
> --> Could the [...]

kk3008: For our sake, it's fine to flag the call to qspi_write_unprotect() during init. But I would expect that it makes sense, in general to call qspi_write_unprotect() during init, to support other qspi flash types, that also startup in protect mode. And I don't think it does any harm on the types that does not require this.

Regarding that you don't see the problem on the stm32l496g_disco, is that the Quad-spi flash on that board is a mx25R6435FM2IL0 from Macronix. And apparently that device follows the standards.

FRASTM, 2024-06-19:
> is no info about quad page program opcodes in the SFDP 
>        tables, hence it has been assumed that NO

Ok, I see cmd->AddressMode is QSPI_ADDRESS_1_LINE before entering the qspi_prepare_quad_program
So that that could just be  
```
	if (cmd->Instruction == SPI_NOR_CMD_PP_1_1_4) {
		cmd->AddressMode = QSPI_ADDRESS_4_LINES;
	}
```
But the case  _cmd->Instruction != SPI_NOR_CMD_PP_1_1_4_  is never considered, anywhere.

kk3008, 2024-06-19:
Basically the changes I have suggested, only handle the special case, where you use the Microchip device, we use, and it takes care of the problem that there is a bug in that family of chips.
You are right, that the cmd->Instruction != SPI_NOR_CMD_PP_1_1_4 is never considered. But to my best knowledge, if the chip you use, follows the standards, the code will support it correct. My change only takes care of the very special case with Microchip.

FRASTM, 2024-06-19:
> Basically the changes I have suggested, only handle the special case, where you use the Microchip device, we use, and it takes care of the problem that there is a bug in that family of chips. You [...]

Let's limit the changes to the `CONFIG_USE_MICROCHIP_QSPI_FLASH_WITH_STM32=y` condition, especially in the context of the V3.7.0

kk3008, 2024-06-19:
Sounds like a good idea!

kk3008, 2024-06-20:
If there is anything I can do to help you test your changes, please let me know. We are of course very interested in that we can use the next Zephyr release right out of the box.

kk3008, 2024-07-08:
Hi FRASTM
Sorry for not replying until now, but I had overseen your mail.
I have looked at the changes you have made, and compared them with my old changes. And it looks fine.
I have not been able to actually test it on our hardware platform, because it requires a lot of updating to the new Zephyr version.
But to my best knowledge, your changes based on my inputs are correct.

Best regards,

Klaus Karkov

Sure, can you have a look at the #74534<https://urldefense.com/v3/__https:/github.com/zephyrproject-rtos/zephyr/pull/74534__;!!OKzgfr8!YPYyJLi6NBJ3lN1IPCot7UOsnlsrQX5GdpGLMG2nfRFksSD89T_7rwcGqkehVfw_c_va-Y30pkSvhgM3fR2jffsSoWM$> and test on your with target with microchip

CONFIDENTIALITY NOTICE: This e-mail, including any attachments and/or linked documents, is intended for the sole use of the intended addressee and may contain information that is privileged, confidential, proprietary, or otherwise protected by law. Any unauthorized review, dissemination, distribution, or copying is prohibited. If you have received this communication in error, please contact the original sender immediately by reply email and destroy all copies of the original message and any attachments. Please note that any views or opinions presented in this e-mail are solely those of the author and do not necessarily represent those of Xylem Inc..

closed 2024-10-01 by aescolar