zephyrproject-rtos/zephyr #68629: Bluetooth: remove IPSP

jori-nordic, 2024-02-06:
## Introduction

The proposal is to remove IPSP (ie `CONFIG_NET_L2_BT`) from the tree.

### Problem description

It has been in a broken state for more than a year, with no one seemingly caring. 
It seems there are no real maintainers for that module.

Linux support is erratic, highly dependent on attached controller, kernel version and config.
That makes it very hard to reproduce bugs reported by the community.

It is the only module that sends so-called network buffer `frags` to Bluetooth.
The whole stack needs to be rewritten to support `frags` properly.
I.e. Drivers can't just do `net_buf_pull()` of `buf->len`, they have to use `frag` functions such as `net_buf_frags_len()` and add extra logic everywhere for pulling from the frag list.

### Proposed change

Remove:
- samples
- kconfig options
- ability to send `frags` in L2CAP

### Related bugs

Most of these have been closed as stale:

https://github.com/zephyrproject-rtos/zephyr/issues/53871
https://github.com/zephyrproject-rtos/zephyr/issues/50598
https://github.com/zephyrproject-rtos/zephyr/issues/54856
https://github.com/zephyrproject-rtos/zephyr/issues/30026
https://github.com/zephyrproject-rtos/zephyr/issues/68627

## Alternatives

Someone steps up as maintainer for IPSP, and adds some testing in the CI. 
It's doable since Bluetooth can be virtualized in babblesim.

jori-nordic, 2024-02-08:
FYI @jukkar

jukkar, 2024-02-08:
+1 from me.

jori-nordic, 2024-02-09:
How about you @rlubos ?
I also plan to send an email to the network WG mailing list.

rlubos, 2024-02-09:
> I also plan to send an email to the network WG mailing list.

Yes, please try to get as much feedback as possible on the topic, maybe even send this to other lists (Dev?).

Anyway, as for me I have no problem with removing IPSP support, if we can't find a maintainer for it and get it to a maintainable state.

jori-nordic, 2024-02-27:
@rlubos and @jukkar Sent an email to net, dev and bt lists with no responses.

Would you know what's the best way to go about it, do I have to deprecate the feature first or do I just remove everything in one PR?

rlubos, 2024-02-27:
I think we should just follow https://docs.zephyrproject.org/latest/develop/api/api_lifecycle.html#deprecated, i. e. deprecate Bluetooth L2 and remove any samples/tests that make use of it.

carlescufi, 2024-02-29:
Bluetooth WG:

- This elicits a wider discussion that affects at least Bluetooth, networking and USB: Should we use linear buffers everywhere or convert everyone to a fragmented model where each net_buf instance is a chain of net_bufs. The implication of using fragments everywhere is substantial: every single user of **data** net_bufs would need to access data via net_buf APIs and not using direct pointer access

carlescufi, 2024-02-29:
> I think we should just follow https://docs.zephyrproject.org/latest/develop/api/api_lifecycle.html#deprecated, i. e. deprecate Bluetooth L2 and remove any samples/tests that make use of it.

In the Bluetooth WG we discussed this. This code has not been functional or maintained for at least two years. This means that deprecating it is essentially meaningless. Instead of this we suggest to remove it via breaking API change:
https://docs.zephyrproject.org/latest/develop/api/api_lifecycle.html#introducing-breaking-api-changes

This will also ensure that we do not carry this code over to LTS.

Users and contributors are free to take the current code and propose to reintegrate it in the future.

closed 2024-04-11 by carlescufi