zephyrproject-rtos/zephyr #68639: [Coverity CID: 334891] Out-of-bounds access in drivers/console/uart_mux.c

zephyrbot, 2024-02-06:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/b8556d0d79770825952c13ddadc4782707163e2f/drivers/console/uart_mux.c

Category: Memory - corruptions
Function: `init_uart_mux`
Component: Drivers
CID: [334891](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=334891)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/b8556d0d79770825952c13ddadc4782707163e2f/drivers/console/uart_mux.c#L902

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=334891

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

dcpleung, 2024-02-06:
Given the Coverity report, it is possible that whichever board is used for Coverity needs more stack space for `uart_mux_stack`. Or that we are really using that much stack space.

dcpleung, 2024-02-06:
Should have been fixed by https://github.com/zephyrproject-rtos/zephyr/pull/68432

closed 2024-02-06 by dcpleung