zephyrproject-rtos/zephyr #68640: [Coverity CID: 344527] Operands don't affect result in subsys/net/lib/lwm2m/lwm2m_senml_cbor_decode.c

zephyrbot, 2024-02-06:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/b8556d0d79770825952c13ddadc4782707163e2f/subsys/net/lib/lwm2m/lwm2m_senml_cbor_decode.c#L92

Category: Integer handling issues
Function: `decode_repeated_record_t`
Component: Networking
CID: [344527](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=344527)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/b8556d0d79770825952c13ddadc4782707163e2f/subsys/net/lib/lwm2m/lwm2m_senml_cbor_decode.c#L92

```
86     {
87       zcbor_log("%s\r\n", __func__);
88     
89       bool tmp_result =
90               ((((zcbor_uint32_expect(state, (6)))) &&
91                 (zcbor_int64_decode(state, (&(*result).record_t))) &&
>>>     CID 344527:  Integer handling issues  (CONSTANT_EXPRESSION_RESULT)
>>>     "result->record_t >= -9223372036854775808LL /* -9223372036854775807LL - 1 */" is always true regardless of the values of its operands. This occurs as the logical first operand of "&&".
92                 ((((*result).record_t >= INT64_MIN) && ((*result).record_t <= INT64_MAX)) ||
93                  (zcbor_error(state, ZCBOR_ERR_WRONG_RANGE), false))));
94     
95       if (!tmp_result) {
96               zcbor_trace_file(state);
97               zcbor_log("%s error: %s\r\n", __func__, zcbor_error_str(zcbor_peek_error(state)));
```

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_CONSTANT_EXPRESSION_RESULT). ([CWE-569](http://cwe.mitre.org/data/definitions/569.html))

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=344527

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

nashif, 2024-02-06:
multiple CIDs report of the same issue

Title: [Coverity CID: 344527] Operands don't affect result in subsys/net/lib/lwm2m/lwm2m_senml_cbor_decode.c
++++++++++++++++++++++++++++++++++++++++++++++++++++++

Title: [Coverity CID: 344523] Operands don't affect result in subsys/net/lib/lwm2m/lwm2m_senml_cbor_encode.c
++++++++++++++++++++++++++++++++++++++++++++++++++++++

Title: [Coverity CID: 344522] Operands don't affect result in subsys/net/lib/lwm2m/lwm2m_senml_cbor_decode.c
++++++++++++++++++++++++++++++++++++++++++++++++++++++

Title: [Coverity CID: 344520] Operands don't affect result in subsys/net/lib/lwm2m/lwm2m_senml_cbor_encode.c
++++++++++++++++++++++++++++++++++++++++++++++++++++++

Title: [Coverity CID: 344518] Operands don't affect result in subsys/net/lib/lwm2m/lwm2m_senml_cbor_decode.c
++++++++++++++++++++++++++++++++++++++++++++++++++++++

Title: [Coverity CID: 344516] Operands don't affect result in subsys/net/lib/lwm2m/lwm2m_senml_cbor_decode.c
++++++++++++++++++++++++++++++++++++++++++++++++++++++

Title: [Coverity CID: 344511] Operands don't affect result in subsys/net/lib/lwm2m/lwm2m_senml_cbor_encode.c
++++++++++++++++++++++++++++++++++++++++++++++++++++++

Title: [Coverity CID: 344508] Operands don't affect result in subsys/net/lib/lwm2m/lwm2m_senml_cbor_encode.c
++++++++++++++++++++++++++++++++++++++++++++++++++++++

Other issues closed in favor of this one, please fix all occurances.

rlubos, 2024-02-06:
This applies to the autogenerated zcbor code, the issue was reported in the upstream zcbor repository (https://github.com/NordicSemiconductor/zcbor/issues/335), yet still not fixed. I don't think there's a need to keep the bug open in Zephyr, given it's practically harmless.

closed 2024-02-06 by rlubos