zephyrproject-rtos/zephyr #68662: ESP32C3: Timeout in HCI leading to fatal error during data transfer over BLE [zephyr-v3.5.0]

abdulhaseebayub, 2024-02-06:
**Describe the bug**
I have a BLE Peripheral based on Espressif `esp32c3` running Zephyr `v3.5.0`. Sometimes, either when data is being transferred from the the Peripheral to the Central, there is a timeout in the HCI which immediately leads to a fatal error. I understand that there was a timeout in the HCI layer with the code `-EAGAIN`, but not sure why the fatal error occurs due to it.

The issue is not easily reproducable, and came up in about 10% of the tests.

**Expected behavior**
The module should not crash due to the fatal error.

**Impact**
Showstopper, since my application gets interrupted during critical operations.

**Logs and console output**
```
ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:331
\u0009command opcode 0x2013 timeout with err -11
[00:00:18.935,000]  os:      a0: 00000003    t0: 00000000
[00:00:18.935,000]  os:      a1: 00000002    t1: 42043710
[00:00:18.935,000]  os:      a2: 00000008    t2: 00000020
[00:00:18.935,000]  os:      a3: 60023008    t3: 00000004
[00:00:18.935,000]  os:      a4: 00001888    t4: 0000004c
[00:00:18.935,000]  os:      a5: 00000009    t5: aaaaaaaa
[00:00:18.935,000]  os:      a6: 3fca3cac    t6: aaaaaaaa
[00:00:18.935,000]  os:      a7: 0000002e
[00:00:18.935,000]  os:      ra: 4204796c
[00:00:18.935,000]  os:    mepc: 42047974
[00:00:18.935,000]  os: mstatus: 00001880
[00:00:18.935,000]  os: 
[00:00:18.935,000]  os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:18.935,000]  os: Current thread: 0x3fc8c7f0 (unknown)
[00:00:19.019,000]  os: Halting system
```

**Environment (please complete the following information):**
- Devcontainer based on image: zephyrprojectrtos/zephyr-build:v0.26.6
- Toolchain: Zephyr SDK 0.16.4
- Zephyr branch: zephyr-v3.5.0
- Target: esp32c3 (DevkitM-1)

henrikbrixandersen, 2024-02-06:
Can you reproduce this on current `main`?

henrikbrixandersen, 2024-02-06:
Perhaps related to #68365?

jori-nordic, 2024-02-08:
Could you provide an HCI trace? `CONFIG_BT_MONITOR` can help.

carlescufi, 2024-02-29:
Closing this due to reporter inactivity. @abdulhaseebayub please reopen if this is still an issue.

closed 2024-02-29 by carlescufi

abdulhaseebayub, 2024-03-01:
> Closing this due to reporter inactivity. @abdulhaseebayub please reopen if this is still an issue.

Hi, it looks like the new Espressif HAL v5.1 has resolved this issue (PR still open: https://github.com/zephyrproject-rtos/zephyr/pull/68760). I've tested thoroughly with it and this bug didn't come up, if it does I'll reopen the issue again.

Thank you!