zephyrproject-rtos/zephyr #68667: HWMv2 Branch fails to build on Windows

danieldegrasse, 2024-02-06:
**Describe the bug**
It appears that the current tip of the `hwm-collab` branch (109f6ae99fc846d626aee4da432faaaf63d9a715) does not build on Windows. I do not encounter similar build failures on Linux based systems.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
   `stm32f103_mini`- however, the issue does not seem specific to this board.
 - What have you tried to diagnose or workaround this issue?
   Updated python dependencies, as well as all Zephyr development tools (via `choco upgrade`)

**To Reproduce**
Steps to reproduce the behavior:
1. checkout `hwm-collab` branch (tested on 109f6ae99fc846d626aee4da432faaaf63d9a715)
2. west build -p always -b stm32f103_mini samples\hello_world
3. See error

**Expected behavior**
Build should not fail

**Impact**
Will slow HWMv2 porting efforts

**Logs and console output**
```
(.venv) C:\Users\daniel\zephyrproject\zephyr>west build -p always -b stm32f103_mini samples\hello_world
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/Users/daniel/zephyrproject/zephyr/samples/hello_world
-- CMake version: 3.27.9
-- Found Python3: C:/Users/daniel/zephyrproject/.venv/Scripts/python.exe (found suitable version "3.10.5", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: C:/Users/daniel/zephyrproject/zephyr/.cache
-- Zephyr version: 3.5.99 (C:/Users/daniel/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: stm32f103_mini, variant: stm32f103xe
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.4 (C:/Users/daniel/zephyr-sdk-0.16.4)
-- Found toolchain: zephyr 0.16.4 (C:/Users/daniel/zephyr-sdk-0.16.4)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: C:/Users/daniel/zephyrproject/zephyr/boards/st/stm32f103_mini/stm32f103_mini.dts
-- Generated zephyr.dts: C:/Users/daniel/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/Users/daniel/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/Users/daniel/zephyrproject/zephyr/build/zephyr/dts.cmake

warning: SPARC_CASA (defined at soc/soc_legacy\sparc\Kconfig:4) defined without a typeParsing C:/Users/daniel/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Users/daniel/zephyrproject/zephyr/boards/st/stm32f103_mini/stm32f103_mini_defconfig'
Merged configuration 'C:/Users/daniel/zephyrproject/zephyr/samples/hello_world/prj.conf'

error: Aborting due to Kconfig warnings

CMake Error at C:/Users/daniel/zephyrproject/zephyr/cmake/modules/kconfig.cmake:389 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  C:/Users/daniel/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  C:/Users/daniel/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  C:/Users/daniel/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' -DWEST_PYTHON=C:/Users/daniel/zephyrproject/.venv/Scripts/python.exe '-BC:\Users\daniel\zephyrproject\zephyr\build' -GNinja -DBOARD=stm32f103_mini '-SC:\Users\daniel\zephyrproject\zephyr\samples\hello_world'
```

**Environment (please complete the following information):**
 - Windows 10, build 22H2. Python 3.10.5, cmake version 3.27.9
 - Toolchain: zephyr SDK 0.16.4
 - 109f6ae99fc846d626aee4da432faaaf63d9a715

erwango, 2024-02-07:
Path seems weird `soc/soc_legacy\sparc\Kconfig`. Could it be that there is a linux path hardcoded somewhere that came with the soc_legacy introduction ?

tejlmand, 2024-02-07:
> Path seems weird `soc/soc_legacy\sparc\Kconfig`

and that's the funny thing, in that specific case Kconfig was able to handle mix path separators, because it could correctly identify line 4 in `soc/soc_legacy\sparc\Kconfig`.

But it was indeed a mix path separator issue but in other Kconfig files which caused Kconfig to have issues, and this is fixed here #68679

carlescufi, 2024-02-07:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/68679

PR #68679 (2024-02-07): cmake: convert path to CMake style before writing Kconfig files
Fixes: #68667

Kconfig is not fond of mixed path separators, therefore ensure CMake style path is used in generated Kconfig files. CMake path style uses `/`, and thus ensures correct behavior in Kconfig.
commit 6f7d434e (2024-02-07): cmake: convert path to CMake style before writing Kconfig files

Fixes: #68667

Kconfig is not fond of mixed path separators, therefore ensure CMake
style path is used in generated Kconfig files. CMake path style uses
`/`, and thus ensures correct behavior in Kconfig.
closed 2024-02-07 by carlescufi