zephyrproject-rtos/zephyr #68682: nsim_em: kernel.common.stack_protection Fail

KamilxPaszkiet, 2024-02-07:
**Describe the bug**
`tests/kernel/fatal/exception/kernel.common.stack_protection` fail.

**To Reproduce**
`twister -p nsim_em -s tests/kernel/fatal/exception/kernel.common.stack_protection`

**Logs and console output**
```
*** Booting Zephyr OS build v3.6.0-rc1-194-g1cadd91b05a1 ***
Running TESTSUITE fatal_exception
===================================================================
START - test_fatal
test alt thread 1: generic CPU exception
E: ***** Exception vector: 0x8, cause code: 0x0, parameter 0x0
E: Address 0xc0
E: EV_SWI
E:  r0: 0x0  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x0  r9: 0x0 r10: 0x0 r11: 0x0
E: r12: 0xb8 r13: 0x0  pc: 0xc0
E:  blink: 0xef0 status32: 0x80186006
E: lp_end: 0x0 lp_start: 0x0 lp_count: 0x0
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x80001824 (unknown)
Caught system error -- reason 0
test alt thread 1: generic CPU exception divide zero
E: ***** Exception vector: 0x8, cause code: 0x0, parameter 0x0
E: Address 0x248
E: EV_SWI
E:  r0: 0x0  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x0  r9: 0x80001b4c r10: 0x8 r11: 0x2326
E: r12: 0x240 r13: 0x0  pc: 0x248
E:  blink: 0xef0 status32: 0x80186006
E: lp_end: 0x0 lp_start: 0x80001824 lp_count: 0x800018cc
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x80001824 (unknown)
Caught system error -- reason 0
test alt thread 2: initiate kernel oops
E:  r0: 0x3  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x0  r9: 0x80001b4c r10: 0x8 r11: 0x2326
E: r12: 0xd4 r13: 0x0  pc: 0xe2
E:  blink: 0xef0 status32: 0x80186006
E: lp_end: 0x0 lp_start: 0x80001824 lp_count: 0x800018cc
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Current thread: 0x80001824 (unknown)
Caught system error -- reason 3
test alt thread 3: initiate kernel panic
E:  r0: 0x4  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x0  r9: 0x80001b4c r10: 0x8 r11: 0x2326
E: r12: 0xe4 r13: 0x0  pc: 0xf2
E:  blink: 0xef0 status32: 0x80186006
E: lp_end: 0x0 lp_start: 0x80001824 lp_count: 0x800018cc
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x80001824 (unknown)
Caught system error -- reason 4
test alt thread 4: fail assertion
ASSERTION FAIL [0] @ WEST_TOPDIR/tests/kernel/fatal/exception/src/main.c:159
	intentionally failed assertion
E:  r0: 0x4  r1: 0x9f  r2: 0x80001824  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x0  r9: 0x80001b4c r10: 0x8 r11: 0x2326
E: r12: 0xffffffff r13: 0xb818  pc: 0xf54
E:  blink: 0x14c status32: 0x80186006
E: lp_end: 0x0 lp_start: 0x80001824 lp_count: 0x800018cc
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x80001824 (unknown)
Caught system error -- reason 4
test alt thread 5: initiate arbitrary SW exception
E:  r0: 0x7fffffff  r1: 0x0  r2: 0x7fffffff  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x0  r9: 0x80001b4c r10: 0x8 r11: 0x2326
E: r12: 0xf4 r13: 0x0  pc: 0x10a
E:  blink: 0xef0 status32: 0x80186006
E: lp_end: 0x0 lp_start: 0x80001824 lp_count: 0x800018cc
E: >>> ZEPHYR FATAL ERROR 2147483647: Unknown error on CPU 0
E: Current thread: 0x80001824 (unknown)
Caught system error -- reason 2147483647
test alt thread 6: initiate arbitrary SW exception negative
E:  r0: 0xfffffffe  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x0  r9: 0x80001b4c r10: 0x8 r11: 0x2326
E: r12: 0x10c r13: 0x0  pc: 0x11a
E:  blink: 0xef0 status32: 0x80186006
E: lp_end: 0x0 lp_start: 0x80001824 lp_count: 0x800018cc
E: >>> ZEPHYR FATAL ERROR -2: Unknown error on CPU 0
E: Current thread: 0x80001824 (unknown)
Caught system error -- reason -2
test stack HW-based overflow - supervisor 1
E: ***** Exception vector: 0x6, cause code: 0x2, parameter 0x2
E: Address 0x80002ff0
E: EV_ProtV
E: Memory write protection violation (stack checking scheme)
E:  r0: 0x0  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0x5
E:  r8: 0x0  r9: 0x80001b4c r10: 0x8 r11: 0x2326
E: r12: 0x2f4 r13: 0x0  pc: 0x7c
E:  blink: 0x90 status32: 0x80186006
E: lp_end: 0x0 lp_start: 0x80001824 lp_count: 0x800018cc
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x80001824 (unknown)
Caught system error -- reason 2
test stack HW-based overflow - supervisor 2
E: ***** Exception vector: 0x6, cause code: 0x2, parameter 0x2
E: Address 0x80002ff0
E: EV_ProtV
E: Memory write protection violation (stack checking scheme)
E:  r0: 0x0  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0x5
E:  r8: 0x0  r9: 0x80001b4c r10: 0x8 r11: 0x2326
E: r12: 0x2f4 r13: 0x0  pc: 0x7c
E:  blink: 0x90 status32: 0x80186006
E: lp_end: 0x0 lp_start: 0x80001824 lp_count: 0x800018cc
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x80001824 (unknown)
Caught system error -- reason 2
test stack HW-based overflow - user 1
E: ***** Exception vector: 0x6, cause code: 0x2, parameter 0x4
E: Address 0x80002ffc
E: EV_ProtV
E: Memory write protection violation (MPU)
E:  r0: 0x0  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0x0
E:  r8: 0x0  r9: 0x0 r10: 0x0 r11: 0x0
E: r12: 0x2f4 r13: 0x0  pc: 0x80
E:  blink: 0x90 status32: 0x80186086
E: lp_end: 0x0 lp_start: 0x80001824 lp_count: 0x800018cc
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x80001824 (unknown)
Caught system error -- reason 0
Wrong crash type got 0 expected 2
PROJECT EXECUTION FAILED
```

**Environment:**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.16.3
 - v3.6.0-rc1-194-g1cadd91b05a1

nashif, 2024-02-07:
@evgeniy-paltsev @abrodkin  @kokas-a

kokas-a, 2024-02-07:
Became broken after b898541809b85292dd5b6928ab34cad0be23da57 commit

ruuddw, 2024-02-07:
That might be a coincidence. Looks like both the MPU check and stack bounds check are/should be triggered at the same time, but only the MPU one is reported by nSIM. Could be a simulator cornercase.

evgeniy-paltsev, 2024-08-19:
The issue is caused by peculiarities of ARC GNU toolchain code generation, see more details here: https://github.com/foss-for-synopsys-dwc-arc-processors/toolchain/issues/643

The ARC MWDT toolchain is not affected.

gbarkadiusz, 2024-11-27:
Observed also on `nsim/nsim_hs/mpuv6`        
`tests/kernel/fatal/exception/kernel.common.stack_protection FAILED Failed (nsim 0.132s)`
_/twister-out/nsim_nsim_hs_mpuv6/tests/kernel/fatal/exception/kernel.common.stack_protection/handler.log_
```
*** Booting Zephyr OS build v4.0.0-rc2-384-ge62e3a85c8ef ***
Running TESTSUITE fatal_exception
===================================================================
START - test_fatal
test alt thread 1: generic CPU exception
E: ***** Exception vector: 0x8, cause code: 0x0, parameter 0x0
E: Address 0xc0
E: EV_SWI
E:  r0: 0x0  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x0  r9: 0x0 r10: 0x0 r11: 0x0
E: r12: 0xb8 r13: 0x0  pc: 0xc0
E:  blink: 0xef4 status32: 0x80186002
E: lp_end: 0x0 lp_start: 0x0 lp_count: 0x0
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x80000884 (unknown)
Caught system error -- reason 0
test alt thread 1: generic CPU exception divide zero
E: ***** Exception vector: 0x8, cause code: 0x0, parameter 0x0
E: Address 0x244
E: EV_SWI
E:  r0: 0x0  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x80000d68  r9: 0x0 r10: 0x0 r11: 0x237a
E: r12: 0x23c r13: 0x0  pc: 0x244
E:  blink: 0xef4 status32: 0x80186002
E: lp_end: 0x0 lp_start: 0x80000884 lp_count: 0x8000092c
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x80000884 (unknown)
Caught system error -- reason 0
test alt thread 2: initiate kernel oops
E:  r0: 0x3  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x80000d68  r9: 0x0 r10: 0x0 r11: 0x237a
E: r12: 0xd4 r13: 0x0  pc: 0xe2
E:  blink: 0xef4 status32: 0x80186002
E: lp_end: 0x0 lp_start: 0x80000884 lp_count: 0x8000092c
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Current thread: 0x80000884 (unknown)
Caught system error -- reason 3
test alt thread 3: initiate kernel panic
E:  r0: 0x4  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x80000d68  r9: 0x0 r10: 0x0 r11: 0x237a
E: r12: 0xe4 r13: 0x0  pc: 0xf2
E:  blink: 0xef4 status32: 0x80186002
E: lp_end: 0x0 lp_start: 0x80000884 lp_count: 0x8000092c
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x80000884 (unknown)
Caught system error -- reason 4
test alt thread 4: fail assertion
ASSERTION FAIL [0] @ WEST_TOPDIR/os.rtos.zephyr.zephyr/tests/kernel/fatal/exception/src/main.c:159
	intentionally failed assertion
E:  r0: 0x4  r1: 0x9f  r2: 0x80000884  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x80000d68  r9: 0x0 r10: 0x0 r11: 0x237a
E: r12: 0xffffffff r13: 0xb320  pc: 0xf30
E:  blink: 0x14c status32: 0x80186002
E: lp_end: 0x0 lp_start: 0x80000884 lp_count: 0x8000092c
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x80000884 (unknown)
Caught system error -- reason 4
test alt thread 5: initiate arbitrary SW exception
E:  r0: 0x7fffffff  r1: 0x0  r2: 0x7fffffff  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x80000d68  r9: 0x0 r10: 0x0 r11: 0x237a
E: r12: 0xf4 r13: 0x0  pc: 0x10a
E:  blink: 0xef4 status32: 0x80186002
E: lp_end: 0x0 lp_start: 0x80000884 lp_count: 0x8000092c
E: >>> ZEPHYR FATAL ERROR 2147483647: Unknown error on CPU 0
E: Current thread: 0x80000884 (unknown)
Caught system error -- reason 2147483647
test alt thread 6: initiate arbitrary SW exception negative
E:  r0: 0xfffffffe  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0xfffffff5
E:  r8: 0x80000d68  r9: 0x0 r10: 0x0 r11: 0x237a
E: r12: 0x10c r13: 0x0  pc: 0x11a
E:  blink: 0xef4 status32: 0x80186002
E: lp_end: 0x0 lp_start: 0x80000884 lp_count: 0x8000092c
E: >>> ZEPHYR FATAL ERROR -2: Unknown error on CPU 0
E: Current thread: 0x80000884 (unknown)
Caught system error -- reason -2
test stack HW-based overflow - supervisor 1
E: ***** Exception vector: 0x6, cause code: 0x2, parameter 0x2
E: Address 0x80001ff0
E: EV_ProtV
E: Memory write protection violation (stack checking scheme)
E:  r0: 0x0  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0x5
E:  r8: 0x80000d68  r9: 0x0 r10: 0x0 r11: 0x237a
E: r12: 0x304 r13: 0xb524  pc: 0x7c
E:  blink: 0x90 status32: 0x80186002
E: lp_end: 0x0 lp_start: 0x80000884 lp_count: 0x8000092c
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x80000884 (unknown)
Caught system error -- reason 2
test stack HW-based overflow - supervisor 2
E: ***** Exception vector: 0x6, cause code: 0x2, parameter 0x2
E: Address 0x80001ff0
E: EV_ProtV
E: Memory write protection violation (stack checking scheme)
E:  r0: 0x0  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0x5
E:  r8: 0x80000d68  r9: 0x0 r10: 0x0 r11: 0x237a
E: r12: 0x304 r13: 0xb524  pc: 0x7c
E:  blink: 0x90 status32: 0x80186002
E: lp_end: 0x0 lp_start: 0x80000884 lp_count: 0x8000092c
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x80000884 (unknown)
Caught system error -- reason 2
test stack HW-based overflow - user 1
E: ***** Exception vector: 0x6, cause code: 0x2, parameter 0x4
E: Address 0x80001ffc
E: EV_ProtV
E: Memory write protection violation (MPU)
E:  r0: 0x0  r1: 0x0  r2: 0x0  r3: 0x0
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0x0
E:  r8: 0x0  r9: 0x0 r10: 0x0 r11: 0x0
E: r12: 0x304 r13: 0xb524  pc: 0x80
E:  blink: 0x90 status32: 0x80186082
E: lp_end: 0x0 lp_start: 0x80000884 lp_count: 0x8000092c
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x80000884 (unknown)
Caught system error -- reason 0
Wrong crash type got 0 expected 2
===================================================================
RunID: a42da608fd0d7565460a6f5a76086fc3
PROJECT EXECUTION FAILED
```

gbarkadiusz, 2025-05-23:
Still occur.

PR #95901 (2025-09-21): tests: kernel: fix issue with stack overflow detection
on arc, the test fails with an MPU error instead of a stack
overflow because the priv stack is merged into the defined
stack.

Fixes #68682
commit e3d2e7d6 (2025-09-21): tests: kernel: fix issue with stack overflow detection

on arc, the test fails with an MPU error instead of a stack
overflow because the priv stack is merged into the defined
stack.

Fixes #68682
closed 2025-09-21 by cfriedt