zephyrproject-rtos/zephyr #68707: Enable acquisition_time for NXP adc_mcux_lpadc.c driver

DerekSnell, 2024-02-07:
Currently, the adc_mcux_lpadc.c driver does not support the[ acquisition_time](https://github.com/zephyrproject-rtos/zephyr/blob/129b73ce266e1eabc518333b38d2971c207685fa/drivers/adc/adc_mcux_lpadc.c#L78) configuration.  The HAL does support configuring the [sampleTimeMode ](https://github.com/zephyrproject-rtos/hal_nxp/blob/d45b14c198d778658b7853b48378d2e132a6c4be/mcux/mcux-sdk/drivers/lpadc/fsl_lpadc.c#L460) in the [lpadc_conv_command_config_t struct](https://github.com/zephyrproject-rtos/hal_nxp/blob/d45b14c198d778658b7853b48378d2e132a6c4be/mcux/mcux-sdk/drivers/lpadc/fsl_lpadc.h#L484).

ZhaoxiangJin, 2024-03-01:
Hi @DerekSnell, could you please assign this issue ticket to me? thanks.

decsny, 2025-01-17:
Fixed in #70636

closed 2025-01-17 by decsny