zephyrproject-rtos/zephyr #68712: LE Audio: Make stream_lang `char[3]` instead of `uint32_t`

Thalley, 2024-02-07:
**Is your enhancement proposal related to a problem? Please describe.**
The stream language is an ISO 639-3 value such as "eng", "fra", "deu", etc. (https://en.wikipedia.org/wiki/ISO_639-3), but is currently handled as a uint32_t value. It makes more sense to treat it as a `char[3]` value.

**Describe the solution you'd like**
Modify the following functions to treat it as a `char[3]` instead of `uint32_t`:
```
bt_audio_codec_cfg_meta_get_stream_lang
bt_audio_codec_cfg_meta_set_stream_lang
bt_audio_codec_cap_meta_get_stream_lang
bt_audio_codec_cap_meta_set_stream_lang
```

**Describe alternatives you've considered**
N/A

**Additional context**
See BT assigned numbers for more information

marc-hb, 2024-02-10:
> It makes more sense to...

This is quite vague, can you be more specific? This seems to allow re-use of generic code. Maybe it should be char[3] at some levels and LE uint32_t at other, lower levels?

Thalley, 2024-02-12:
> > It makes more sense to...
> 
> This is quite vague, can you be more specific? This seems to allow re-use of generic code. Maybe it should be char[3] at some levels and LE uint32_t at other, lower levels?

Since the ISO 639-3 codes are 3-character strings, it generally does not make sense to treat them as unsigned integers at any point, which is why I think `char[3]` makes more sense. Ideally we'd define a `#define BT_AUDIO_ISO_LANG_SIZE 3` (or something like that) and use that instead of the hardcoded `3`.

marc-hb, 2024-02-12:
>  it generally does not make sense to treat them as unsigned integers at any point

You just rephrased without adding more details.

I spent 5 minutes looking at the code and I found one potential reason that I mentioned above.

Thalley, 2024-02-12:
> You just rephrased without adding more details.

I did :) I don't have a better reason than using `char` for string values, instead of `uint32_t` for string values. 

> I spent 5 minutes looking at the code and I found one potential reason that I mentioned above.

Can you please share that reason? :)

closed 2024-06-12 by nashif