zephyrproject-rtos/zephyr #68718: UART Interrupt and Async on Different Peripherals

KyleKotowick, 2024-02-07:
**Describe the bug**
I have a device with two UART peripherals. It's a custom board based on the nRF5340DK (target device is an nRF5340).

`uart0` is for Modbus, and `uart1` is for a custom driver for a ThingMagic RFID reader.

The [Modbus driver](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/modbus/modbus_serial.c) uses the [interrupt-driven UART API](https://docs.zephyrproject.org/latest/hardware/peripherals/uart.html#uart-interrupt-api), and for the RFID reader I'd like to use the [Async UART API](https://docs.zephyrproject.org/latest/hardware/peripherals/uart.html#uart-async-api).

Note that while the [UART documentation page](https://docs.zephyrproject.org/latest/hardware/peripherals/uart.html) has a large warning box stating that the interrupt and async APIs should not be used on the **same hardware peripheral**, I'm using two separate peripherals so this shouldn't be an issue.

So, since I need both, my `proj.conf` includes:

```
CONFIG_UART_ASYNC_API=y
CONFIG_UART_INTERRUPT_DRIVEN=y
```

When `CONFIG_UART_INTERRUPT_DRIVEN` is enabled, however, it causes the `uart_callback_set` function to return `-ENOSYS`. The exact same code, which doesn't even actually _use_ the interrupt API, works fine if `CONFIG_UART_INTERRUPT_DRIVEN` is disabled.

The [Nordic DevAcademy Lesson 5](https://github.com/NordicDeveloperAcademy/ncs-fund/tree/main/v2.x.x/lesson5/fund_less5_exer1_solution) sample is the simplest example of this. Simply add `CONFIG_UART_INTERRUPT_DRIVEN=y` to `proj.conf`, and all of a sudden `uart_callback_set` will start returning `-ENOSYS`.

**Expected behavior**
The async API should function even if `CONFIG_UART_INTERRUPT_DRIVEN=y`, as long as both interrupt and async APIs are not being configured on the same peripheral.

**Impact**
High. We could re-write all of our custom driver code to switch from async to interrupt, so we would only need to include the interrupt API, but this shouldn't be necessary since there's no documented reason that these two APIs can't co-exist, and the async API is the recommended one to use according to Nordic.

**Environment (please complete the following information):**
 - OS: Windows 11
 - Toolchain: nRF Connect v2.5.2 (includes Zephyr 3.4.99 `a768a05e6205e415564226543cee67559d15b736`)

KyleKotowick, 2024-02-07:
Digging through the nRF Connect source code, I found the issue. Their UART driver requires doing the following:
```
# Enable both APIs
CONFIG_UART_ASYNC_API=y
CONFIG_UART_INTERRUPT_DRIVEN=y

# Flag UART0 as not ASYNC, but INTERRUPT
CONFIG_UART_0_ASYNC=n
CONFIG_UART_0_INTERRUPT_DRIVEN=y

# Flag UART1 as not INTERRUPT, but ASYNC
CONFIG_UART_1_INTERRUPT_DRIVEN=n
CONFIG_UART_1_ASYNC=y
```

It would be very helpful if this was included in the UART documentation, but these are Nordic-specific Kconfig options, so I'm not sure how that works.

caioorafael, 2024-02-08:
Nice, I Just had the same issue in the past, but in my case I was using a third modbus library included in my project using async uart and I needed the interrupt driven for modem debug through at cmd. Nice to know that this approachs works! I'll try it.

I don't know how this work beacuse I'm new at commenting in open source projects, but I think the ncs-sdk is open also, and you could open a issue there also. 

Big thanks. :)

henrikbrixandersen, 2024-02-08:
Thank you for reporting this. However - unless you are able to reproduce this issue with upstream Zephyr `main` - please report issues with the nRF Connect SDK (NCS) on the [Nordic Semiconductor DevZone](https://devzone.nordicsemi.com/).

KyleKotowick, 2024-02-09:
Yep, this is a Nordic issue. Closing.

closed 2024-02-09 by KyleKotowick