zephyrproject-rtos/zephyr #68728: rp2040 flash driver hangs infinitely when used with settings nvs backend

lgehreke, 2024-02-08:
**Describe the bug**
When using the settings api with NVS backend a call to `settings_save_one()` will often hang up and never recover.
The cause seems to be the rp2040 flash driver. It is waiting to receive bytes from the flash but they never arrive so
`flash_put_get()` will run in a while loop. This will never recover. The `tx_count` and `rx_skip` are both 0 but
the while loop is still running because `rx_count` remains non 0 and will never decrease.

I am using a custom board with the rp2040 and a W25Q16JV flash attached to the pico.
Unfortunately i did find no workaround for this.

**To Reproduce**
This can be reproduced with the settings sample `samples/subsys/settings`.
I extended the proj.conf:

```
CONFIG_STDOUT_CONSOLE=y
CONFIG_FLASH=y
CONFIG_FLASH_PAGE_LAYOUT=y
CONFIG_FLASH_MAP=y

CONFIG_SETTINGS=y
CONFIG_SETTINGS_RUNTIME=y
CONFIG_NVS=y
CONFIG_SETTINGS_NVS=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_INITIALIZE_AT_BOOT=y
```

And using this device tree

```
/ {
	chosen {
		zephyr,sram = &sram0;
		zephyr,flash = &flash0;
		zephyr,flash-controller = &ssi;
		zephyr,console = &cdc_acm_uart0 ;
		zephyr,shell-uart = &cdc_acm_uart0 ;
		zephyr,code-partition = &slot0_partition;
	};
};

zephyr_udc0: &usbd {
	status = "okay";
	cdc_acm_uart0: cdc_acm_uart0 {
		compatible = "zephyr,cdc-acm-uart";
	};
};

&flash0 {
	reg = <0x10000000 DT_SIZE_M(2)>;

	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		/* Reserved memory for the second stage bootloader */
		second_stage_bootloader: partition@0 {
			label = "second_stage_bootloader";
			reg = <0x00000000 0x100>;
			read-only;
		};

		slot0_partition: partition@100 {
			label = "image-0";
			reg = <0x100 (DT_SIZE_M(1) - 0xF000)>;
		};

		storage_partition: partition@1f9000 {
			label = "storage";
			reg = <0x1F9000 0x00007000>;
		};
	};
};
```

To reproduce build this sample with the extended proj.conf and device tree for a rp2040 based board and flash it.
This will consistently hang up in `flash_put_get()` .

**Expected behavior**
Not hanging up during flash write.

**Impact**
This is a showstopper since the settings nvs may hang up the whole system.

**Logs and console output**
Here is a gdb stack trace of the hangup

```
#0  flash_put_get (tx=0x10008de9 "", tx@entry=0x10008de2 "", rx=0x0, count=count@entry=7, rx_skip=0, rx_skip@entry=4) at /data/projects/nfc/zephyr/drivers/flash/flash_rpi_pico.c:105
#1  0x20000548 in flash_write_partial_internal (addr=addr@entry=2072832, data=data@entry=0x10008de2 "", size=size@entry=7) at /data/projects/nfc/zephyr/drivers/flash/flash_rpi_pico.c:157
#2  0x20000226 in flash_write_partial (flash_offs=flash_offs@entry=2072832, data=data@entry=0x10008de2 "", count=count@entry=7) at /data/projects/nfc/zephyr/drivers/flash/flash_rpi_pico.c:180
#3  0x10007ef4 in flash_rpi_write (dev=<optimized out>, offset=<optimized out>, data=<optimized out>, size=<optimized out>) at /data/projects/nfc/zephyr/drivers/flash/flash_rpi_pico.c:241
#4  0x1000730a in z_impl_flash_write (len=12, data=0x10008ddd, offset=2072827, dev=<optimized out>) at /data/projects/nfc/zephyr/include/zephyr/drivers/flash.h:212
#5  flash_write (len=12, data=0x10008ddd, offset=2072827, dev=<optimized out>) at /data/projects/nfc/zephyr/samples/subsys/settings/build/zephyr/include/generated/syscalls/flash.h:64
#6  nvs_flash_al_wrt (fs=fs@entry=0x20000b48 <default_settings_nvs+8>, addr=<optimized out>, data=data@entry=0x10008ddd, len=len@entry=12) at /data/projects/nfc/zephyr/subsys/fs/nvs/nvs.c:122
#7  0x10007620 in nvs_flash_data_wrt (len=12, data=0x10008ddd, fs=0x20000b48 <default_settings_nvs+8>) at /data/projects/nfc/zephyr/subsys/fs/nvs/nvs.c:182
#8  nvs_flash_wrt_entry (len=12, data=0x10008ddd, id=32776, fs=0x20000b48 <default_settings_nvs+8>) at /data/projects/nfc/zephyr/subsys/fs/nvs/nvs.c:415
#9  nvs_write (fs=fs@entry=0x20000b48 <default_settings_nvs+8>, id=id@entry=32776, data=data@entry=0x10008ddd, len=12) at /data/projects/nfc/zephyr/subsys/fs/nvs/nvs.c:1112
#10 0x10003f38 in settings_nvs_save (cs=0x20000b40 <default_settings_nvs>, name=0x10008ddd "", value=0x20000000 <length_val> "d", val_len=8) at /data/projects/nfc/zephyr/subsys/settings/src/settings_nvs.c:297
#11 0x10003aa0 in settings_save_one (name=0x10008ddd "", value=0x20000000 <length_val>, val_len=8) at /data/projects/nfc/zephyr/subsys/settings/src/settings_store.c:106
#12 0x10001d1e in alpha_handle_export (cb=0x10003a75 <settings_save_one>) at /data/projects/nfc/zephyr/samples/subsys/settings/src/main.c:163
#13 0x10003bac in settings_save () at /data/projects/nfc/zephyr/subsys/settings/src/settings_store.c:147
#14 0x10002204 in example_save_and_load_basic () at /data/projects/nfc/zephyr/samples/subsys/settings/src/main.c:245
#15 main () at /data/projects/nfc/zephyr/samples/subsys/settings/src/main.c:555
(gdb) p tx_count
$1 = 0
(gdb) p rx_count
$2 = 1
(gdb) p rx_skip
$3 = 0
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr sdk 0.16.3
 - Commit: zephyr-v3.5.0

lgehreke, 2024-02-13:
The problem seems to be coming from the settings nvs subsystem itself.

When you write a entry with `settings_save_one()` , for example `settings_save_one("alpha/beta/voltage", (const void *)&val, sizeof(val_s32));`, the key `"alpha/beta/voltage"` will reside in flash memory. By default xip caching is enabled, so there is a chance that this key is cached in ram. If it is not in ram the flash driver will attempt to read the flash while writing. Since we exited xip execution to be able to write to the flash this will cause the hangup. When disabling xip caching the hangup will come every time when we try to write to flash.

To disable xip caching you have to modify `zephyr/soc/arm/rpi_pico/rp2/soc.c`:

```
#include <hardware/structs/xip_ctrl.h>

...

static int rp2040_init(void)
{
	reset_block(~(RESETS_RESET_IO_QSPI_BITS | RESETS_RESET_PADS_QSPI_BITS |
		      RESETS_RESET_PLL_USB_BITS | RESETS_RESET_PLL_SYS_BITS));

	unreset_block_wait(RESETS_RESET_BITS &
			   ~(RESETS_RESET_ADC_BITS | RESETS_RESET_RTC_BITS |
			     RESETS_RESET_SPI0_BITS | RESETS_RESET_SPI1_BITS |
			     RESETS_RESET_UART0_BITS | RESETS_RESET_UART1_BITS |
			     RESETS_RESET_USBCTRL_BITS | RESETS_RESET_PWM_BITS));

	clocks_init();

	unreset_block_wait(RESETS_RESET_BITS);

	xip_ctrl_hw->ctrl &= ~(XIP_CTRL_EN_BITS);

	return 0;
}
```

and `zephyr/drivers/flash/flash_rpi_pico.c`

```
void __no_inline_not_in_flash_func(flash_write_partial)(uint32_t flash_offs, const uint8_t *data,
							size_t count)
{
	rom_connect_internal_flash_fn connect_internal_flash =
		(rom_connect_internal_flash_fn)rom_func_lookup_inline(
			ROM_FUNC_CONNECT_INTERNAL_FLASH);
	rom_flash_exit_xip_fn flash_exit_xip =
		(rom_flash_exit_xip_fn)rom_func_lookup_inline(ROM_FUNC_FLASH_EXIT_XIP);
	rom_flash_flush_cache_fn flash_flush_cache =
		(rom_flash_flush_cache_fn)rom_func_lookup_inline(ROM_FUNC_FLASH_FLUSH_CACHE);

	flash_init_boot2_copyout();

	__compiler_memory_barrier();

	connect_internal_flash();
	flash_exit_xip();
	flash_write_partial_internal(flash_offs, data, count);
	flash_flush_cache();
	flash_enable_xip_via_boot2();
	// disable the caching
	xip_ctrl_hw->ctrl &= ~(XIP_CTRL_EN_BITS);
}

```

We also need to modify `flash_write_partial()` because a call to `flash_flush_cache()` will re-enable xip caching

lgehreke, 2024-02-13:
Maybe the settings subsystem should check if the name lies in flash and if so buffer the name in ram before calling `nvs_write()`

MaureenHelm, 2024-02-13:
@Laczen can you take a look please?

lgehreke, 2024-02-13:
@Laczen The problem is that the read access to the flash happens while we are currently writing bytes over the qspi interface. The flash cannot handle another interrupting command at this time. This is why all flash_write functions in the pico driver are placed in RAM instead of rom. They would not be executable if ``flash_exit_xip()` is called. Have a look at this pull request: https://github.com/zephyrproject-rtos/zephyr/pull/68951 This should solve the problem at driver level

zhang-wenchao, 2025-09-09:
Request to reopen this investigation

PR #68951 (2024-04-05): drivers: flash: flash_rpi_pico: Added buffered flash write for addresses in flash
It is impossible to perform flash reads during a flash write. When the data to write to flash lies in the flash itself, for example when using names in the settings api, it is buffered in ram before writing.

This fixes #68728