zephyrproject-rtos/zephyr #68736: posis: Deadlock when calling read() on eventfd with count 0

ljdevdev, 2024-02-08:
**Describe the bug**

When calling `read()` on an eventFd with count 0, the calling thread deadlocks. Additionally any thread subsequently calling write() on the same fd also deadlocks.

The reason for this is, that `eventfd_rw_op` locks the same mutex, that write() locks and then blocks on the condition variable that is supposed to be signaled by write() without releasing the mutex first.

```c
// From zephyr/lib/os/fdtable.c
ssize_t write(int fd, const void *buf, size_t sz)
{
	ssize_t res;

	if (_check_fd(fd) < 0) {
		return -1;
	}

	(void)k_mutex_lock(&fdtable[fd].lock, K_FOREVER);

	res = fdtable[fd].vtable->write(fdtable[fd].obj, buf, sz);

	k_mutex_unlock(&fdtable[fd].lock);

	return res;
}
```

```c
// From zephyr/lib/posix/options/eventfd.c
/* common to both eventfd_read_op() and eventfd_write_op() */
static ssize_t eventfd_rw_op(void *obj, void *buf, size_t sz,
			     int (*op)(struct eventfd *efd, eventfd_t *value))
{
...
                err = k_mutex_lock(lock, K_FOREVER); // this is also the fdentry.lock that is locked in both read() and write()
...
		/* wait for a write or close */ 
		// HERE THE MUTEX IS STILL LOCKED
		err = k_condvar_wait(cond, lock, K_FOREVER);
... 
```

**To Reproduce**

```bash
# initialize my-workspace for the example-application (main branch)
west init -m https://github.com/zephyrproject-rtos/example-application --mr main my-workspace
# update Zephyr modules
cd my-workspace
west update
```

replace `app/src/main.c`

```c
#include <sys/eventfd.h>
#include <stdio.h>
#include <unistd.h>
#include <stdlib.h>
#include <poll.h>
#include <pthread.h>
#include <assert.h>

void* readFunction(void* arg){
    int eventFd = *(int*)arg;

    uint64_t readValue;
    printf("calling read.\n");
    ssize_t readSize = read(eventFd, &readValue, sizeof(uint64_t));
    printf("returned from read. readSize=%d\n",readSize);

    return NULL;
}

void* writeFunction(void* arg){
    int eventFd = *(int*)arg;

    uint64_t incrementValue = 1;
    printf("calling write.\n");
    ssize_t writtenSize = write(eventFd, &incrementValue, sizeof(uint64_t));
    printf("returned from write. writtenSize=%d\n", writtenSize);

    return NULL;
}

int main(){
    int eventFd = eventfd(0,0);
    printf("eventFd=%d\n", eventFd);
    int err;
    pthread_t readThread;
    err = pthread_create(&readThread, NULL, readFunction, &eventFd);
    if(err != 0){
        printf("error creating thread\n");
        return 1;
    }
    sleep(1);

    pthread_t writeThread; 
    err = pthread_create(&writeThread, NULL, writeFunction, &eventFd);
    if(err != 0){
        printf("error creating thread\n");
        return 1;
    }
    
    printf("join read thread\n");
    pthread_join(readThread, NULL);
    printf("join write thread\n");
    pthread_join(writeThread, NULL);
    printf("exit\n");
    return 0;
}

```

replace `app/prj.conf`:

```
CONFIG_LOG=y
CONFIG_POSIX_API=y
CONFIG_THREAD_STACK_INFO=y
CONFIG_DYNAMIC_THREAD=y
CONFIG_DYNAMIC_THREAD_STACK_SIZE=1024
CONFIG_POLL=y
CONFIG_EVENTFD=y
CONFIG_EVENTFD_MAX=50
CONFIG_MINIMAL_LIBC=y
CONFIG_DYNAMIC_THREAD_POOL_SIZE=10

```

run 
```bash
west build -b native_sim example-application/app
./build/zephyr/zephyr.exe
```
**Actual behavior**
```
*** Booting Zephyr OS build v3.6.0-rc1-65-g7551b85f2287 ***
eventFd=3
calling read.
join read thread
calling write.
^C
Stopped at 5.150s
```
The programm deadlocks

**Expected behavior**
(You can get this if compiling with empty prj.conf, which results in linking against the host libc)
```
*** Booting Zephyr OS build v3.6.0-rc1-65-g7551b85f2287 ***
eventFd=4
calling read.
join read thread
calling write.
returned from write. writtenSize=8
returned from read. readSize=8
join write thread
exit
```

**Impact**
This severely hinders the usage of the posix poll API.

**Environment:**
 - OS: Linux
 - Toolchain Zephyr SDK
 - v3.6.0-rc1-65-g7551b85f2287

cfriedt, 2024-02-08:
Hi @ljdevdev - are you able to create a PR that fixes the issue? Thanks for the bug report.

cfriedt, 2024-02-09:
> after thinking about this a bit I noticed that the problem is quite fundamental and not bound to the eventfd functionality. 

Hi @ljdevdev - that's quite likely. Also, it looks like the problem is not particularly bound to the POSIX API judging by the fact that it has been encountered also by the network subsystem (#42526 and #42553).

The fdtable code that includes read, write, open, and close has been around for quite some time (not originally as part of POSIX) and there is a bit of a rewrite planned for it as part of #51211 (also mentioned in #66132).

For the mean time, it might make sense to use `eventfd_read()` and `eventfd_write()`, although eventually `read()` and `write()` should just be bulletproof as well.

ljdevdev, 2024-02-09:
> For the mean time, it might make sense to use `eventfd_read()` and `eventfd_write()`, although eventually `read()` and `write()` should just be bulletproof as well.

Yeah, that is what I resorted too already. Thanks for the quick answer. I'll be waiting for that rewrite then. I'll look around the connected issues, maybe I can be of some help.

cfriedt, 2024-02-09:
@ljdevdev - would be great to get some additional help on building ZVFS. It's conceptually not that hard, but touches a lot of  areas. Even just extra testing or code review would be awesome 👍

closed 2024-04-28 by github-actions[bot]