zephyrproject-rtos/zephyr #68747: STM32WB55 ZEPHYR FATAL ERROR 3 stm32wb5x_BLE_HCILayer_fw.bin

KenthJohan, 2024-02-08:
**Describe the bug**
ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0 when pressing connect on NRF bluetooth app.
 - target platform: STM32WB55

**To Reproduce**
Upload this firmware from STM32CubeProgrammer: [stm32wb5x_BLE_HCILayer_fw.bin (V1.14.1)](https://github.com/STMicroelectronics/STM32CubeWB/blob/v1.14.1/Projects/STM32WB_Copro_Wireless_Binaries/STM32WB5x/stm32wb5x_BLE_HCILayer_fw.bin)

[image]
### prj.conf
```
CONFIG_BT=y
CONFIG_BT_PERIPHERAL=y
CONFIG_BT_SMP=n
CONFIG_TINYCRYPT=y
CONFIG_BT_DEVICE_NAME="Simple App1"
CONFIG_BT_DIS=y
CONFIG_BT_DIS_PNP=n
CONFIG_BT_BAS=y
CONFIG_BT_DEBUG_LOG=y
CONFIG_BT_GATT_CLIENT=y
CONFIG_BT_AUTO_PHY_UPDATE=n
CONFIG_LOG=y
CONFIG_LOG_BUFFER_SIZE=2048
```
### main.c
```c
#include <stdbool.h>
#include <zephyr/types.h>
#include <stddef.h>
#include <string.h>
#include <errno.h>
#include <zephyr/sys/printk.h>
#include <zephyr/sys/byteorder.h>
#include <zephyr/kernel.h>

#include <zephyr/bluetooth/bluetooth.h>
#include <zephyr/bluetooth/hci.h>
#include <zephyr/bluetooth/conn.h>
#include <zephyr/bluetooth/uuid.h>
#include <zephyr/bluetooth/gatt.h>
#include <zephyr/bluetooth/services/bas.h>

BT_GATT_SERVICE_DEFINE(ess_svc,
	BT_GATT_PRIMARY_SERVICE(BT_UUID_ESS),
);

static const struct bt_data ad[] = {
	BT_DATA_BYTES(BT_DATA_FLAGS, (BT_LE_AD_GENERAL | BT_LE_AD_NO_BREDR)),
	BT_DATA_BYTES(BT_DATA_GAP_APPEARANCE, 0x00, 0x03),
};

static void connected(struct bt_conn *conn, uint8_t err)
{
	if (err) {
		printk("Connection failed (err 0x%02x)\n", err);
	} else {
		printk("Connected\n");
	}
}

static void disconnected(struct bt_conn *conn, uint8_t reason)
{
	printk("Disconnected (reason 0x%02x)\n", reason);
}

BT_CONN_CB_DEFINE(conn_callbacks) = {
	.connected = connected,
	.disconnected = disconnected,
};

static void bt_ready(void)
{
	int err;
	printk("Bluetooth initialized\n");
	err = bt_le_adv_start(BT_LE_ADV_CONN_NAME, ad, ARRAY_SIZE(ad), NULL, 0);
	if (err) {
		printk("Advertising failed to start (err %d)\n", err);
		return;
	}
	printk("Advertising successfully started\n");
}

void main(void)
{
	int err;
	err = bt_enable(NULL);
	if (err)
	{
		printk("Bluetooth init failed (err %d)\n", err);
		return;
	}
	bt_ready();
	while (1)
	{
		k_sleep(K_SECONDS(1));
	}
}

```

**Logs and console output**
```
*** Booting Zephyr OS build v3.6.0-rc1-65-g7551b85f2287 ***
ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:331
	command opcode 0xfc0c timeout with err -11
[00:00:10.001,000] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000000  r2/a3:  0x00000002
[00:00:10.001,000] <err> os: r3/a4:  0x20000690 r12/ip:  0x00002711 r14/lr:  0x0800458b
[00:00:10.001,000] <err> os:  xpsr:  0x41000000
[00:00:10.001,000] <err> os: Faulting instruction address (r15/pc): 0x08004596
[00:00:10.001,000] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:10.002,000] <err> os: Current thread: 0x20001628 (unknown)
[00:00:10.052,000] <err> os: Halting system
```

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain Zephyr SDK
 - Zephyr OS build v3.6.0-rc1-65-g7551b85f2287

KenthJohan, 2024-02-08:
I think I found the correct firmware from https://github.com/STMicroelectronics/STM32CubeWB
I think stm32wb5x_BLE_HCILayer_extended_fw.bin is the only one that is working. I got no error yet.
[image]

This is firmwares that is available:
```
stm32wb5x_BLE_HCILayer_extended_fw.bin
stm32wb5x_BLE_HCILayer_fw.bin
stm32wb5x_BLE_HCI_AdvScan_fw.bin
stm32wb5x_BLE_LLD_fw.bin
stm32wb5x_BLE_Mac_802_15_4_fw.bin
stm32wb5x_BLE_Stack_full_extended_fw.bin
stm32wb5x_BLE_Stack_full_fw.bin
stm32wb5x_BLE_Stack_light_fw.bin
stm32wb5x_BLE_Thread_dynamic_fw.bin
stm32wb5x_BLE_Thread_static_fw.bin
stm32wb5x_BLE_Zigbee_FFD_dynamic_fw.bin
stm32wb5x_BLE_Zigbee_FFD_static_fw.bin
stm32wb5x_BLE_Zigbee_RFD_dynamic_fw.bin
stm32wb5x_BLE_Zigbee_RFD_static_fw.bin
stm32wb5x_FUS_fw.bin
stm32wb5x_FUS_fw_for_fus_0_5_3.bin
stm32wb5x_Mac_802_15_4_fw.bin
stm32wb5x_Phy_802_15_4_fw.bin
stm32wb5x_Safeboot_fw.bin
stm32wb5x_Thread_FTD_fw.bin
stm32wb5x_Thread_MTD_fw.bin
stm32wb5x_Thread_RCP_fw.bin
stm32wb5x_Zigbee_FFD_fw.bin
stm32wb5x_Zigbee_RFD_fw.bin
```

nordicjm, 2024-02-08:
As per https://docs.zephyrproject.org/latest/contribute/bin_blobs.html#support-and-maintenance support is not provided for binary blobs

erwango, 2024-02-08:
Might be a simple issue of flash partitionning. I'll have a check

r2r0, 2024-02-08:
> As per https://docs.zephyrproject.org/latest/contribute/bin_blobs.html#support-and-maintenance support is not provided for binary blobs

@nordicjm But it is a way how BLE works on STM32WB MCU - do you suggest that Zephyr will not support BLE on this chip?

nordicjm, 2024-02-08:
> > As per https://docs.zephyrproject.org/latest/contribute/bin_blobs.html#support-and-maintenance support is not provided for binary blobs
> 
> @nordicjm But it is a way how BLE works on STM32WB MCU - do you suggest that Zephyr will not support BLE on this chip?

Absolutely, this is no different than the linux kernel, you use blobs at your own risk with no support provided from zephyr

erwango, 2024-02-08:
Here is an abstract of the link above:

>The Zephyr Project is not expected to be responsible for the maintenance and support of contributed binary blobs. As a consequence, **at the discretion of the Zephyr Project release team, and on a case-by-case basis**:

>GitHub issues reported on the zephyr repository tracker that require use of blobs to reproduce **may not** be treated as bugs

>Such issues **may** be closed as out of scope of the Zephyr project

>**This does not imply that issues which require blobs to reproduce will be closed without investigation. For example, the issue may be exposing a bug in a Zephyr code path that is difficult or impossible to trigger without a blob. Project maintainers may accept and attempt to resolve such issues.**

So, as maintainer, this is my call, and I propose to keep it open for investigation.
If analysis shows that issue is located in binary, this issue will be closed.

KenthJohan, 2024-02-08:
To clarify the v1.18.0 firmware seems to work so far. It looks like Zephyr dropped support for v1.14.1.
- stm32wb5x_BLE_HCILayer_fw.bin v1.14.1 ❌ZEPHYR FATAL ERROR 3
- stm32wb5x_BLE_HCILayer_fw.bin v1.18.0 ✔️So far is working

Can somebody link me the documentation that says which firmware to use with latest Zephyr if it exist.

ajarmouni-st, 2024-02-08:
> Can somebody link me the documentation that says which firmware to use with latest Zephyr if it exist.

https://github.com/zephyrproject-rtos/zephyr/blob/d2001a6fab85af0bde533e5613f12f98016d11ec/boards/arm/nucleo_wb55rg/doc/nucleo_wb55rg.rst?plain=1#L192-L194

erwango, 2024-02-08:
> Can somebody link me the documentation that says which firmware to use with latest Zephyr if it exist.

As stated [here](https://docs.zephyrproject.org/latest/boards/arm/nucleo_wb55rg/doc/nucleo_wb55rg.html#bluetooth-and-compatibility-with-stm32wb-copro-wireless-binaries), we regularly update STM32 support with latest Cube packages releases which provide BLE binaries (once Cube package update at each release). Hence we are ensuring latest Zephyr is compatible with recent STM32WB releases.
Version information is updated here: https://github.com/zephyrproject-rtos/hal_stm32/blob/main/lib/stm32wb/hci/README.

erwango, 2024-02-09:
@KenthJohan Can we close this issue now?