zephyrproject-rtos/zephyr #68817: usb: udp: frames of exact multiples of 64 Byte not transmitted correctly

Aluleichtbau, 2024-02-09:
Hi there,

**Describe the bug**

If a UDP package is sent out using `zsock_send` and the length of the frame on the wire is a multiple of 64 Byte (128, 192, or 256, and so on) the package is not transmitted. Instead the package seems to be buffered. Next calls of `zsock_send` seem also to be buffered, until the sum of all the buffered data is no multiple of 64 Bytes. Then the first package is sent out and all the other data appears in the ethernet trailer. At least it looks like that in Wireshark.

**To Reproduce**

Sniff with wireshark on USB. Do something like the following code on the MCU.

```c
static char my_63_data[21];
static char my_64_data[22/*+64*/];
static struct sockaddr dst_addr;
static int sock_fd;

void some_init()
{
	/* set some dst_addr */
	/*...*/

	zsock_socket(AF_INET, SOCK_DGRAM, IPPROTO_UDP);
	zsock_connect(sock_fd, dst_addr, sizeof(struct sockaddr_in));
	int flags = zsock_fcntl(sock_fd, F_GETFL, 0);
	zsock_fcntl(sock_fd, F_SETFL, flags | O_NONBLOCK);

	snprintf(&my_63_data, sizeof(my_63_data), "63B 63B 63B 63B 63B ");
	snprintf(&my_64_data, sizeof(my_64_data), "64B 64B 64B 64B 64B 4");
}

void send_64_byte(void)
{
	zsock_send(sock_fd, &my_64_data, sizeof(my_64_data), 0);
}

void send_63_byte(void)
{
	zsock_send(sock_fd, &my_63_data, sizeof(my_63_data), 0);
}
```

Call the functions `send_63_byte` and `send_64_byte` several times in some random order.

**Expected behavior**
For each zsock_send call a frame in wireshark.

**Impact**
Some packages get dropped, if the application waits for a response it will timeout eventually or wait forever.

**Logs and console output**
[image]
The "+64" in the example has been uncommented, that's why the payload longer and filled with more zeros. Then `send_63_byte`, `send_64_byte`, and `send_63_byte` have been called in this order.

**Environment**

I use Zephyr OS @ a8a0eb52439caccf51c66f845b2c29ceef0da39e on an STM32H730VB MCU with a gcc-arm-none-eabi-10.3-2021.07 toolchain.

Relevant config:
```
CONFIG_NETWORKING=y
CONFIG_NET_IPV6=n
CONFIG_NET_IPV4=y
CONFIG_NET_UDP=y

CONFIG_USB_DEVICE_STACK_NEXT=y
CONFIG_NET_L2_ETHERNET=y
CONFIG_USBD_CDC_ECM_CLASS=y

CONFIG_NET_SOCKETS=y
```

**Additional context**
N/A

Aluleichtbau, 2024-02-19:
@jfischer-no 
This bug breaks the communication in my case. I am using CoAP on top of it. Simple requests are very likely to have this problem. I have a lot of 64 byte messages. Only the first of the sequence of 64-byte messages on the wire gets an response and there are messages that never get a response because of this.

jfischer-no, 2024-02-19:
Please try an in tree sample like `samples/net/zperf` and on the host side

```bash
#!/bin/bash

IP=$1

for (( n=0; n<1; n++ ))
do
        for (( packetsize=0; packetsize<1025; packetsize=$packetsize+1 ))
        do
                ping -i 0.002 -c 10 -s $packetsize -q $IP
        done
done
```

Aluleichtbau, 2024-02-21:
> Please try an in-tree sample like `samples/net/zperf` and on the host side

@jfischer-no 

I had to perform some extra steps to make zperf work with the new usb device stack (configured device descriptor).

```
ping -i 500 -c 1 -s 21 -q $IP
ping -i 500 -c 1 -s 22 -q $IP
ping -i 500 -c 1 -s 23 -q $IP
```

... resulted in this:

[image]

The monitoring has been done on USBcap interface.

The problem can be observed when the device replies to the ICMP request with packetsize 22, 86 (22+64), ..., (22+64*n). Some replies fail, some are successful, but all of the replies contain non-empty Ethernet trailer.

With a sample sending UDP packages with incrementing lengths the same problem is observed (also starts with 22 UDP data length, step is also 64).

jfischer-no, 2024-02-21:
> > Please try an in-tree sample like `samples/net/zperf` and on the host side
> 
> @jfischer-no
> 
> I had to perform some extra steps to make zperf work with the new usb device stack (configured device descriptor).

Shell commands can be used for this.

> With a sample sending UDP packages with incrementing lengths the same problem is observed (also starts with 22 UDP data length, step is also 64).

Looks like a bug in the new experimental STM32 shim driver, IIRC ZLP flag handling is missing. I have a patch somewhere. Will update this comment asap.

jfischer-no, 2024-02-26:
> Looks like a bug in the new experimental STM32 shim driver, IIRC ZLP flag handling is missing. I have a patch somewhere. Will update this comment asap.

Actually I mixed it with another problem I found in the driver testing high speed, but it is missing something like this:

```diff
diff --git a/drivers/usb/udc/udc_stm32.c b/drivers/usb/udc/udc_stm32.c
index 5780150a0fe..43b6e08e742 100644
--- a/drivers/usb/udc/udc_stm32.c
+++ b/drivers/usb/udc/udc_stm32.c
@@ -322,6 +322,13 @@ void HAL_PCD_DataInStageCallback(PCD_HandleTypeDef *hpcd, uint8_t epnum)
                return;
        }
 
+       if (udc_ep_buf_has_zlp(buf) && ep != USB_CONTROL_EP_IN) {
+               udc_ep_buf_clear_zlp(buf);
+               HAL_PCD_EP_Transmit(&priv->pcd, ep, buf->data, 0);
+
+               return;
+       }
+
        udc_buf_get(dev, ep);
 
        if (ep == USB_CONTROL_EP_IN) {
```

Not sure if it is 100% correct, do not have time to look at details yet.

Aluleichtbau, 2024-02-27:
> Not sure if it is 100% correct, do not have time to look at details yet (Quickly tested, seems to work).

@jfischer-no

Having these lines applied, I tested my setups and I can confirm that it seems to work.

closed 2024-07-01 by nashif