zephyrproject-rtos/zephyr #68831: SCA: CodeChecker: Failed to build

yashi, 2024-02-10:
**Describe the bug**

CodeChecker SCA as [documented](https://docs.zephyrproject.org/latest/develop/sca/codechecker.html) doesn't work.

**To Reproduce**

```
❯ west build -b mimxrt1064_evk samples/basic/blinky -- -DZEPHYR_SCA_VARIANT=codechecker
  :
  :
[ERROR 2024-02-10 22:18] - 
372 warnings and 1 error generated.
Error while processing /home/yashi/src/zephyr-project/zephyr/kernel/mutex.c.
Suppressed 1 warnings (1 in non-user code).
Use -header-filter=.* to display errors from all non-system headers. Use -system-headers to display errors from system headers as well.
Found compiler error(s).

[INFO 2024-02-10 22:18] - ----==== Summary ====----
[INFO 2024-02-10 22:18] - Successfully analyzed
[INFO 2024-02-10 22:18] -   cppcheck: 90
[INFO 2024-02-10 22:18] - Failed to analyze
[INFO 2024-02-10 22:18] -   clangsa: 90
[INFO 2024-02-10 22:18] -   clang-tidy: 90
[INFO 2024-02-10 22:18] - Total analyzed compilation commands: 90
[INFO 2024-02-10 22:18] - ----=================----
[INFO 2024-02-10 22:18] - Analysis finished.
[INFO 2024-02-10 22:18] - To view results in the terminal use the "CodeChecker parse" command.
[INFO 2024-02-10 22:18] - To store results use the "CodeChecker store" command.
[INFO 2024-02-10 22:18] - See --help and the user guide for further options about parsing and storing the reports.
[INFO 2024-02-10 22:18] - ----=================----
[INFO 2024-02-10 22:18] - Analysis length: 9.191490411758423 sec.
[129/131] cd /home/yashi/src/zephyr-project/zephyr/build/zephyr ...le=/home/yashi/src/zephyr-project/zephyr/build/zephyr/zephyr.elf
FAILED: CMakeFiles/codechecker sca/codechecker/codechecker.plist /home/yashi/src/zephyr-project/zephyr/build/CMakeFiles/codechecker /home/yashi/src/zephyr-project/zephyr/build/sca/codechecker/codechecker.plist 
cd /home/yashi/src/zephyr-project/zephyr/build && /home/yashi/src/zephyr-project/.venv/bin/CodeChecker analyze --keep-gcc-include-fixed --keep-gcc-intrin --output /home/yashi/src/zephyr-project/zephyr/build/sca/codechecker/codechecker.plist --name zephyr /home/yashi/src/zephyr-project/zephyr/build/compile_commands.json
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/yashi/src/zephyr-project/zephyr/build
```

```
❯ CodeChecker version
[INFO 2024-02-10 22:19] - CodeChecker analyzer version:
---------------------------------------------------------------
Kind                 | Version                                 
---------------------------------------------------------------
Base package version | 6.22.2                                  
Package build date   | 2023-10-30T11:55                        
Git commit ID (hash) | a3d35cfd509b995187288cd8631446415a334257
Git tag information  | 6.22.2                                  
---------------------------------------------------------------

[INFO 2024-02-10 22:19] - CodeChecker web version:
------------------------------------------------------------------------------
Kind                                | Version                                 
------------------------------------------------------------------------------
Base package version                | 6.22.2                                  
Package build date                  | 2023-10-30T11:55                        
Git commit ID (hash)                | a3d35cfd509b995187288cd8631446415a334257
Git tag information                 | 6.22.2                                  
Server supported Thrift API version | 6.53                                    
Client Thrift API version           | 6.53                                    
------------------------------------------------------------------------------
```

**Expected behavior**

Exit without error. 

**Environment (please complete the following information):**
 - OS: Debian GNU/Linux trixie/sid
 - Toolchain: Zephyr SDK 0.16.5
 - Commit SHA: v3.6.0-rc1-111-ga0a2e9536b1

**Additional context**

```
❯ rm -rf build && west build -b mimxrt1064_evk samples/basic/blinky --cmake-only
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/yashi/src/zephyr-project/zephyr/samples/basic/blinky
-- CMake version: 3.28.3
-- Found Python3: /home/yashi/src/zephyr-project/.venv/bin/python (found suitable version "3.11.7", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/yashi/.cache/zephyr
-- Zephyr version: 3.6.0-rc1 (/home/yashi/src/zephyr-project/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: mimxrt1064_evk
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.5 (/home/yashi/zephyr-sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (/home/yashi/zephyr-sdk-0.16.5)
-- Found Dtc: /usr/bin/dtc (found suitable version "1.7.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/yashi/src/zephyr-project/zephyr/boards/arm/mimxrt1064_evk/mimxrt1064_evk.dts
-- Generated zephyr.dts: /home/yashi/src/zephyr-project/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/yashi/src/zephyr-project/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/yashi/src/zephyr-project/zephyr/build/zephyr/dts.cmake
Parsing /home/yashi/src/zephyr-project/zephyr/Kconfig
Loaded configuration '/home/yashi/src/zephyr-project/zephyr/boards/arm/mimxrt1064_evk/mimxrt1064_evk_defconfig'
Merged configuration '/home/yashi/src/zephyr-project/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to '/home/yashi/src/zephyr-project/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/yashi/src/zephyr-project/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /home/yashi/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/yashi/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
Load components for MIMXRT1064:
driver_common component is included.
driver_reset component is included.
device_CMSIS component is included.
CMSIS_Include_core_cm component is included.
device_system component is included.
driver_lpuart component is included.
driver_igpio component is included.
driver_cache_armv7_m7 component is included.
-- Using ccache: /usr/bin/ccache
-- Configuring done (2.2s)
-- Generating done (0.0s)
-- Build files have been written to: /home/yashi/src/zephyr-project/zephyr/build
```

yashi, 2024-02-10:
Related issues:
- #64236
- #62278
- #61666

henrikbrixandersen, 2024-02-10:
> 372 warnings and 1 error generated.
> Error while processing /home/yashi/src/zephyr-project/zephyr/kernel/mutex.c.

So CodeChecker finds an error - can you please post information about the error it found?

marc-hb, 2024-02-10:
> ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: > /usr/bin/cmake --build /home/yashi/src/zephyr-project/zephyr/build

This does not look like a problem with codechecker or SCA yet.

yashi, 2024-02-11:
Ah, nevermind.  I didn't know `west build` fails when SCA finds a problem.  Sorry for the noise.  I'm closing this.

BTW, by default `west build` succeeds building the image when Parasoft C++test is specified.  It can of course return non zero when it finds a problem for CI or etc.  I need to document that.

closed 2024-02-11 by yashi

marc-hb, 2024-02-12:
`west build` is only a thin wrapper, it returns whatever `cmake` / `ninja` returns.

I'm less confident about the SCA code in zephyr/cmake but I suspect it's also a "wrapper" as far as exit codes are concerned. So I suspect it will just return whatever codecheck returns?

yashi, 2024-02-12:
Yes.

A thing was that, when `west build` failed, it felt like the actual build failed.  Sure I was explicitly setting `-DZEPHYR_SCA_VARIANT=codechecker` so I should have prepare for failure. But... :grin: 

I'm not claming one is better than the other at all, BTW.

Many SCA tools, if not all, have various _modes_ or stages, ex) track actual build commands, and analyze the souces.  So, I'm sure we can do both way.

*case A*
1. `west build`, then
2. analyze with SCA tools

*case B*
1. `west build` and analyze

It's just that CodeChecker is case B and Parasoft C++test took the case A.  While developing, **B** is simpler.  **A** is more explicit and might be helpful for CI.

And I felt that it might be better for our users if all SCAs have a unified behaviour for their UX.

marc-hb, 2024-02-12:
No matter how each checker works, the top level build interface should always offer: 

(1) SINGLE `west build` command to build normally (no checker)
(2) SINGLE command to run a checker from scratch, including all the necessary steps. `west build --DZEPHYR_SCA_VARIANT=...` does that for some (all?) checkers.
(3) A non-zero exit code for ANY sort of failure. Static analysis or not.
(4) A very clear **error message** for ANY sort of failure[*] 
(5) Incremental build **by default**, skipping as many steps as possible. For tricky situations ask users to delete the build directory or to use some `--pristine` option.

In normal use, the Zephyr build system should hide differences between "checker case A" and "checker case B" as often as possible.

The ability to run  steps separately is useful for experts troubleshooting build issues but no one else wants to have to run more than a single command every time. Expect when debugging the build itself, no one wants to know about the difference between CMake configuration, actual build, static analysis, etc. That's why every project has some sort of wrapper script on top of CMake.

BTW compiler warnings are also a form of "static analysis" / "codechecking". I mean there is no clear line between a "compiler" and a "static analyser". What we call "static analysis" is just more elaborate warnings that take longer to run and have more false positives (cc: #67360). So the differences between "case A" and "case B" are just implementation details that are better hidden in normal use.

CI runs commands (1) and (2) in separate CI jobs. Looking at the pass/fail status of (1) and (2) anyone can quickly tell whether the problem is checker-specific or not without even looking at the error messages (yet good error messages are still very important).

cc: @tejlmand , @kartben 

[*] concurrent builds tend to bury the relevant error message. I often use this trick in CI: `build stuff || build stuff -j1`, it's dead simple and works great.