zephyrproject-rtos/zephyr #68911: Adv and Sync fails for start with ESP32-C3

bsrinath-hp, 2024-02-13:
I build the following two samples to sync with each other:
- west build -p always -b esp32c3_devkitm samples/bluetooth/periodic_sync_rsp
- west build -p always -b esp32c3_devkitm samples/bluetooth/periodic_adv_rsp

When I run the samples I hit the following errors:
- periodic_sync_rsp: "PAST subscribe failed (err -134)"
- periodic_adv_rsp:  "Scanning failed to start (err -5)"

I am just looking for a working sample for PAwR with ESP32-C3 which supports BT 5.x

hermabe, 2024-02-13:
The PAwR samples require a controller that supports the Bluetooth 5.4 PAwR features. The esp controller is not in tree so you should probably contact the vendor directly.

bsrinath-hp, 2024-02-14:
> The PAwR samples require a controller that supports the Bluetooth 5.4 PAwR features. The esp controller is not in tree so you should probably contact the vendor directly.

The particular controller that I am using (esp32-c3) claims to support BT-5 and I am of the understanding that 5.4 specifically doesn't require any specific HW. So as long as the device supports BT-5, I think 5.4 support is a matter of adding the necessary FW/SW pieces.

hermabe, 2024-02-14:
> The particular controller that I am using (esp32-c3) claims to support BT-5 and I am of the understanding that 5.4 specifically doesn't require any specific HW. So as long as the device supports [...]

Yes, there are no new hardware/radio requirements in 5.4 compared to 5.0 so it is pure sw. However, that software is in the controller part of the stack. The ESP controller is not in tree and such this is not an issue with zephyr.

The in-tree controller also does not support PAwR so you need to check if your hw vendor provides a controller that does support it out of tree.

bsrinath-hp, 2024-02-15:
Thanks for the explanation! Could you please help me understand.. what do you mean by "The ESP controller is not in tree"? What are you looking for in the device tree? Sorry for the ignorance here! I am new to Zephyr. But trying to learn and contribute back to the community.

hermabe, 2024-02-16:
> what do you mean by "The ESP controller is not in tree"?

The "in-tree" controllers are the ones implemented in the zephyr source-tree (https://github.com/zephyrproject-rtos/zephyr/tree/main/subsys/bluetooth/controller). Currently this is nordic and openisa.

There is also support for "out-of-tree" controllers where vendors can implement their controllers in a separate module. This can either be done by implementing a "lower link layer" as part of the zephyr controller implementation, or more typically through an hci driver like the ones in https://github.com/zephyrproject-rtos/zephyr/tree/main/drivers/bluetooth/hci.

The zephyr issue list is only relevant for the in-tree controllers. Issues with out-of-tree components have to be taken with the vendors directly.

sylvioalves, 2024-02-16:
@bsrinath-hp Hi, as mentioned before, PAwR feature is available in Bluetooth 5.4 spec. There is no implementation for it in ESP32-C3.

closed 2024-02-16 by hermabe

bsrinath-hp, 2024-02-22:
@sylvioalves: Since you represent Espressif here, would you happen to know when will BT 5.4 spec be implemented for ESP32-C3> Or which ESP32 device has support for BT 5.4 spec such that I can run all of Zephyr samples on it?