zephyrproject-rtos/zephyr #68914: Expand generated Kconfigs for board revisions

tautologyclub, 2024-02-13:
When you create board revisions in your board files (revision.cmake), a single Kconfig gets generated and set. I think it's `CONFIG_BOARD_REVISION="1.1"` or something similar.

This is great, but makes it not so easy to work with using the preprocessor. The way Zephyr does it with boards is instead:

```
CONFIG_BOARD="plank"
CONFIG_BOARD_PLANK=y
```

Which enables easy ifdef wrapping etc. I propose he same should be done with board revisions:

```
CONFIG_BOARD_REVISION="1.1"
CONFIG_BOARD_REVISION_1_1=y
```

nordicjm, 2024-02-13:
Why not just do a normal string comparison using `strcmp`? The compiler will optimise it and either leave the code in or remove it.
```
if (strcmp(CONFIG_BOARD_REVISION, "0.7.0") == 0){
    some code here
}
```

tautologyclub, 2024-02-13:
I'm guessing same reasons it's done the way it is with `BOARD` etc - there could be code that doesn't even compile if the right version isn't set, there could be static arrays that you want to #ifdef wrap, etc.

Besides, there's the question of consistency in API design - if we expose the board name as a bool, it makes sense to expose the board revision in the same way.

tautologyclub, 2024-02-16:
Thinking about this some more though, it seems this could be a huge breach of standard procedure - boards are declared way prior to building, which means generating board-specific Kconfig options is not a problem.

But with revision overlays... the build system has already been started when the overlay is picked up. It may not be problematic since the generated Kconfig option will be 100% deterministic, but it still feels sort of iffy that the build system itself auto-generates build configurations, does it not?

tejlmand, 2024-03-14:
> but it still feels sort of iffy that the build system itself auto-generates build configurations, does it not?

@tautologyclub  think more of it as mapping CMake variables to a Kconfig equivalent, like board setting.

Of course a Kconfig can do:
```
config FOO
    bool
    default y if $(BOARD) == "plank"
```

the problem with such approach is that this is error prone to typos like `plakn` or board renames.

Whereas
```
config FOO
    bool
    default y if BOARD_PLANK
```
because a typo like BOARD_PLAKN or board rename results in a undefined Kconfig symbol in compliance check.
Something not possible with string comparison.

tautologyclub, 2024-03-14:
It's not the same for board revisions tho. Boards are 100% known a priori but board revisions aren't. If you did kconfig bools like my original suggestion, you'd have to auto generate kconfig options on the fly, inside the build system. That, or predefine every allowable board revision kconfig, which is basically infinitely many.

That's what's iffy about it.

tautologyclub, 2024-03-14:
Device tree props seems like a much saner option, but I'm not sure if auto generating dt props are considered kosher either...

tejlmand, 2024-03-14:
> It's not the same for board revisions tho. Boards are 100% known a priori but board revisions aren't.

Not sure I fully follow here, revisions are also passed into the build system by the user (in cases with default revision, the build system will internally apply revision before dts and Kconfig).

Technically it would be easy to extend this to cover revision as well:
https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/Kconfig.v2

However, the tricky part arises with the custom revisions.
We can handle known valid revisions from board.yml, but custom revisions may not be present in board.yml
 and thus hard to handle in check compliance which may result in `if BOARD_REVISION_FOOBAR` as a false undefined warning.