zephyrproject-rtos/zephyr #68938: Update stm32f4 RCC driver to support MIPI DSI host/display on stm32f469i_disco

devin-hale, 2024-02-13:
Hey! I am attempting to add soc support for the mipi dsi driver, and ultimately display of the stm32f469i_disco. I've gotten as far as trying to define and map clock registers to the dts/soc, and consulting on the discord before being told that the RCC driver will need to be updated.

If at all possible, it would be great to update this driver so that I can continue adding support for the mipi host and display of this board. The following are some resources I've used so far so that hopefully nobody has to retread any ground:

stm32f469i_disco user manual:
https://www.st.com/resource/en/user_manual/um1932-discovery-kit-with-stm32f469ni-mcu-stmicroelectronics.pdf

stm32f469xx reference manual:
https://www.st.com/resource/en/reference_manual/rm0386-stm32f469xx-and-stm32f479xx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf

default stm32f469i_disco clock config from stm32cubemx:
[image]

ajarmouni-st, 2024-02-28:
Hi @devin-hale, thank you for working on this.
> before being told that the RCC driver will need to be updated.

Can you provide more details about this?

erwango, 2024-02-28:
> Can you provide more details about this?

I'll give you the inputs

jancotipsarevic, 2024-06-12:
Hi, is there any progress on that?, I am very interested!
Thanks to all of you.