zephyrproject-rtos/zephyr #68946: CMake: SCA: Cppcheck finds no defects when including <zephyr/kernel.h>

alfonsoespinosa, 2024-02-13:
**Bug description**

When running static analysis through CodeChecker support in the build system (`-DZEPHYR_SCA_VARIANT=codechecker`), [Cppcheck](https://cppcheck.sourceforge.io/) fails to analyze the code.

Found on 3.4, but also verified to fail in 3.5.

**To Reproduce**
Steps to reproduce the behavior:
1. Introduce some undefined behaviour construct (for instance, zero-division) in for example `zephyr/samples/hello_world/src/main.c`
2. `west build -b qemu_x86 -- -DZEPHYR_SCA_VARIANT=codechecker`
3. The defect is detected
4. Include <zephyr/kernel.h> at the top of the source file
5. Build again (`west build`), and no defect is found for the source file containing the issue

**Expected behavior**
Native integration of static code analysis should work when including Zephyr headers. Otherwise, it offers no functionality for real projects

**Impact**
I can temporarily get static code analysis with a workaround using a script that removes `zephyr/include` from the list of include paths (`-I` parameter) for the call to `cppcheck` executable

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain (Zephyr SDK, tested with both `arm-zephyr-eabi` and `x86_64-zephyr-elf`)
 - v3.5.0 (`a6eef0ba3755f2530c5ce93524e5ac4f5be30194`)

**Additional context**
I have tried running `CodeChecker analyze` with the same arguments set by the build system. Then, when I omit the Cppcheck option `--suppress=preprocessorErrorDirective`, error `#error "Invalid/unknown toolchain configuration"` is raising in `zephyr/include/zephyr/toolchain.h`. When `<zephyr/kernel.h>` is not included in sample source file, probably `zephyr/toolchain.h` gets never included, because even removing `--suppress=preprocessorErrorDirective`, this `#error` is not thrown, performing the static code analysis as expected.

Ccing @pdgendt

alfonsoespinosa, 2024-02-20:
For the Cortex-M I am dealing with, in order to prevent `cppcheck` from triggering the `#error` directives that raise in [zephyr/toolchain.h](https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v3.4.0/include/zephyr/toolchain.h), [zephyr/toolchain/gcc.h](https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v3.4.0/include/zephyr/toolchain/gcc.h), [zephyr/sys/util.h](https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v3.4.0/include/zephyr/sys/util.h), [zephyr/arch/arm/aarch32/error.h](https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v3.4.0/include/zephyr/arch/arm/aarch32/error.h), and [zephyr/kernel.h](https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v3.4.0/include/zephyr/kernel.h) these are the definitions I need to provide as arguments to the analyzer.

```
-D__GNUC__ -D__BYTE_ORDER__=1234 -DCONFIG_ARM -D__CHAR_BIT__=8 -D__SIZEOF_LONG__=4 -D__SIZEOF_LONG_LONG__=8 -DCONFIG_ARMV7_M_ARMV8_M_MAINLINE -DCONFIG_NUM_COOP_PRIORITIES=16 -DCONFIG_NUM_PREEMPT_PRIORITIES=15
```

They can be placed in a file to let `CodeChecker analyze` forward them to `cppcheck` with option `--cppcheckargs`. That can be added to the input arguments in https://github.com/zephyrproject-rtos/zephyr/blob/a6eef0ba3755f2530c5ce93524e5ac4f5be30194/cmake/sca/codechecker/sca.cmake#L22 so I will probably try to figure out how to generate automatically the file with the definitions, depending some of them from the compiler type, architecture or kernel priorities configuration

pdgendt, 2024-02-20:
> ```
> -D__GNUC__ -D__BYTE_ORDER__=1234 -DCONFIG_ARM -D__CHAR_BIT__=8 -D__SIZEOF_LONG__=4 -D__SIZEOF_LONG_LONG__=8 -DCONFIG_ARMV7_M_ARMV8_M_MAINLINE -DCONFIG_NUM_COOP_PRIORITIES=16 [...]

Thanks @alfonsoespinosa for the investigation, it does look like the generated `Kconfig` symbols are not taken into account? Is it possible that `cppcheck` isn't including the generated `autoconf.h` file?

alfonsoespinosa, 2024-02-20:
> 
> Thanks @alfonsoespinosa for the investigation, it does look like the generated `Kconfig` symbols are not taken into account? Is it possible that `cppcheck` isn't including the generated `autoconf.h` file?

It's not being added to the command line arguments by `CodeChecker analyze`, but indeed that is the purpose of `cppcheck` option `--include`:

```
--include=<file>
                         Force inclusion of a file before the checked file. Can
                         be used for example when checking the Linux kernel,
                         where autoconf.h needs to be included for every file
                         compiled. Works the same way as the GCC -include
                         option.
```

So it could be also added from `CodeChecker analyze` in the file passed with `--cppcheckargs`, and at least the 3 `Kconfig`-related missing defines would be solved.

Bad news is that doing this causes a really weird collateral damage:

```
zephyr/include/zephyr/devicetree.h:4231:0: error: failed to expand '_EXC_IRQ_DEFAULT_PRIO', Invalid ## usage when expanding 'DT_CAT3'.
```

alfonsoespinosa, 2024-04-25:
We have bumped to Zephyr 3.5 and the behaviour changes slightly. The workaround in [my previous comment](https://github.com/zephyrproject-rtos/zephyr/issues/68946#issuecomment-1953589745) is not working anymore. Apart from now having to define manually also `CONFIG_CPU_CORTEX_M` (otherwise preprocessor fails [here](https://github.com/zephyrproject-rtos/zephyr/blob/a6eef0ba3755f2530c5ce93524e5ac4f5be30194/include/zephyr/arch/arm/exc.h#L23)), there are a couple of `__ASSERT` that do not work in the Memory Management Internal APIs header file (in this case probably an issue with Nordic's porting).

I will try to catch up to 3.5 and then later try to work out something for newer versions

alfonsoespinosa, 2024-05-06:
Getting back to this on [v3.5.0](https://github.com/zephyrproject-rtos/zephyr/tree/zephyr-v3.5.0), `cppcheck` is giving 

`error: failed to expand '__ASSERT', it is invalid to use a preprocessor directive as macro parameter [preprocessorErrorDirective]
#if CONFIG_SRAM_BASE_ADDRESS != 0`

[here](https://github.com/zephyrproject-rtos/zephyr/blob/a6eef0ba3755f2530c5ce93524e5ac4f5be30194/include/zephyr/sys/mem_manage.h#L161) and [here](https://github.com/zephyrproject-rtos/zephyr/blob/a6eef0ba3755f2530c5ce93524e5ac4f5be30194/include/zephyr/sys/mem_manage.h#L181) in `include/zephyr/sys/mem_manage.h`. One possible solution is to put the preprocessor directive outside, conditionally defining some macro that later would be used in the `__ASSERT()`, like [this diff](https://diffy.org/diff/608c65fed1a57) does.

There is later an `error: failed to expand '_EXC_IRQ_DEFAULT_PRIO', Invalid ## usage when expanding 'DT_CAT3': Unexpected token ')' [preprocessorErrorDirective]` [here](https://github.com/zephyrproject-rtos/zephyr/blob/a6eef0ba3755f2530c5ce93524e5ac4f5be30194/include/zephyr/arch/arm/cortex_m/exc.h#L53C42-L53C58) in `include/zephyr/arch/arm/cortex_m/exc.h`, that seems to be a `cppcheck` bug about failing to handle ## in macro definitions: https://github.com/danmar/simplecpp/issues/228

closed 2024-07-20 by github-actions[bot]

GilDev, 2024-10-14:
No news about this? Having trouble checking my projects.

danmar, 2026-02-05:
If you check Zephyr source code.. please ensure you are using recent code. Old code is not compatible with cppcheck due to UB in zephyr code, I fixed that UB here:
[eb805306186882d9bfa3b8c8af3221e6e89d0813](https://github.com/zephyrproject-rtos/zephyr/commit/eb805306186882d9bfa3b8c8af3221e6e89d0813)

lmapii, 2026-02-05:
> I have only scanned sample projects. But here are some instructions: https://docs.google.com/document/d/1j7zKeTvbvsRhSo3DUhCmRJ_MusHDk4nkRpDKMTCVInw/edit?usp=sharing It might not work in all situations.. and I am interested in feedback.

With these instructions and using the following command (in a target) to generate a header file that is used as input by `cppcheck` the analysis works for me.

`${CMAKE_C_COMPILER} ${TOOLCHAIN_C_FLAGS} -E -dM ${output_dir}/empty.c > ${output_dir}/__compiler.h`

I'm still trying to figure out how to integrate this cleanly with `CodeChecker` but some initial efforts look promising.