zephyrproject-rtos/zephyr #68964: Shell does not work with PM=y on RT11XX

decsny, 2024-02-13:
**Describe the bug**
The shell module very much does not work when power management is enabled on RT11xx.

**To Reproduce**
west build -b mimxrt1170_evk_cm7 -p always samples/subsys/shell/shell_module/

decsny, 2024-02-29:
> tried following:
> `west build -b mimxrt1060_evk -p always zephyr/samples/subsys/shell/shell_module -- -DCONFIG_PM=y`
> Shell works without any issues

The bug report is for 1170 not 1060

danieldegrasse, 2024-11-05:
Moving this to an enhancement- the issue is caused by the fact that the lowest power state setpoint for the RT1170 gates the UART clock, so the peripheral cannot RX or TX. This would have been broken since the point PM was enabled on this platform.