zephyrproject-rtos/zephyr #68978: doc: Documentation issue in 'services/logging/index' k_cycle_get

clamattia, 2024-02-14:
**Describe the bug**

The page speaks of "If not provided, k_cycle_get or k_cycle_get_32 is used for timestamping" but `k_cycle_get` is not found in the tree.
The only other place I found it was here https://github.com/zephyrproject-rtos/zephyr/blob/019b81a184257a9c190ed9307ab662b16efbdc82/tests/kernel/common/src/clock.c#L170 :
```
	zassert_true(d64 >= d32,
		"k_cycle_get() (64-bit): d64: %" PRIu64 " < d32: %u", d64, d32);

	zassert_true(d64 < (d32 << 1),
		"k_cycle_get() (64-bit): d64: %" PRIu64 " >= 2 * d32: %u",
		d64, (d32 << 1));
```

Probably should be `k_cycle_get_64`?

**Environment**

* Page: `services/logging/index`
* Version: main
* SHA-1: 7d63942c31709739578797ce91b69818308bd9e1

henrikbrixandersen, 2024-02-14:
Looks like it is using `sys_clock_tick_get()` for 64-bit timestamps: https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/logging/log_core.c#L224

clamattia, 2024-02-14:
> Looks like it is using `sys_clock_tick_get()` for 64-bit timestamps: https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/logging/log_core.c#L224

Thanks for looking it up.

Out of cursiosity, why does then the 32-bit not use `sys_clock_tick_get_32`

clamattia, 2024-04-15:
> This issue has been marked as stale because it has been open (more than) 60 days with no activity. Remove the stale label or add a comment saying that you would like to have the label removed [...]

Are you interested in adjusting this slight inaccuracy or shall I close the issue?

closed 2024-06-30 by github-actions[bot]