zephyrproject-rtos/zephyr #68999: ARCH unavailable in toolchain early cmake files in HWMv2

fabiobaltieri, 2024-02-14:
**Describe the bug**
Hi,

hardware model v2 introduces a change where the `ARCH` varaible in CMake is derived from Kconfig:

```
if(HWMv2)
  # HWMv2 obtains arch from Kconfig for the given Board / SoC variant because
  # the Board / SoC path is no longer sufficient for determine the arch
  # (read: multi-core and multi-arch SoC).
  set(ARCH ${CONFIG_ARCH})
```
from `cmake/modules/arch_v2.cmake`

The problem is that the variable right now is used in some toolchain configuration file to set other stuff, internally we were using it in our LLVM `generic.cmake` file to determine if newlib is available or not (apparently we only have it for some architectures), which creates a circular dependency of some sort.

I can fix our use case somehow (will probably always enable it and add an extra Kconfig costraint) but I noticed this is used in `xtools/generic.cmake` as well upstrea, which I think is going to break with hardware model v2.

I this something you folks could look into?

**To Reproduce**

```console
$ git diff
diff --git a/cmake/toolchain/llvm/generic.cmake b/cmake/toolchain/llvm/generic.cmake
index 0b138c2868..afcbad57a9 100644
--- a/cmake/toolchain/llvm/generic.cmake
+++ b/cmake/toolchain/llvm/generic.cmake
@@ -20,3 +20,5 @@ set(BINTOOLS llvm)
 set(TOOLCHAIN_HAS_NEWLIB OFF CACHE BOOL "True if toolchain supports newlib")
 
 message(STATUS "Found toolchain: host (clang/ld)")
+
+message("****** ARCH is ${ARCH}")
$ export ZEPHYR_TOOLCHAIN_VARIANT=llvm
$ west build -p -b stm32_min_dev_blue samples/hello_world
...
****** ARCH is arm
...
$ git checkout remotes/origin/collab-hwm
$ west build -p -b stm32_min_dev@blue samples/hello_world
...
****** ARCH is 
```

**Impact**
Those toolchains would probably break, at least in some conditions.

nordicjm, 2024-02-15:
This is the reason bsim/posix is broken if I remember, because it now picks up the arm toolchain before the host one

golowanow, 2024-02-15:
that was a problem with SOF in #68412

tejlmand, 2024-02-21:
> The problem is that the variable right now is used in some toolchain configuration file to set other stuff, internally we were using it in our LLVM generic.cmake file to determine if newlib is [...]

this should not be done in generic.cmake, but instead in the target.cmake.

The difference between generic.cmake and target.cmake is that the generic one is exactly that, generic, hence no target specific information at this point. Generic is used to find a C compiler which can be used for preprocessing, not for compilation.

The toolchain / C compiler to be used for compilation and linking should be defined in target.cmake.

tejlmand, 2024-02-21:
> I can fix our use case somehow (will probably always enable it and add an extra Kconfig costraint) but I noticed this is used in xtools/generic.cmake as well upstrea, which I think is going to break with hardware model v2.

will take a look.

closed 2025-03-09 by github-actions[bot]