zephyrproject-rtos/zephyr #69065: hwmv2: Board documentation that references files that do not exist

nordicjm, 2024-02-16:
A list of files that have been updated to boards v2 in the collab-hwm branch which reference files (using `:zephyr_file:` or \`\```boards...``\`\`) which do not exist:

 - [ ] boards/96boards/96b_carbon/doc/stm32f401xe.rst
   * [ ] boards/96boards/96b_carbon/96b_carbon_defconfig

 - [ ] boards/altera/altera_max10/doc/index.rst
   * [ ] soc/nios2/nios2f-zephyr/cpu
   * [ ] soc/nios2/nios2f-zephyr/cpu/ghrd_10m50da.sof

 - [ ] boards/circuit_dojo/circuitdojo_feather/doc/index.rst
   * [ ] boards/circuit_dojo/circuitdojo_feather_nrf9160/circuitdojo_feather_nrf9160_common.dtsi

 - [ ] boards/holyiot/holyiot_yj16019/doc/index.rst
   * [ ] boards/holyiod/holyiot_yj16019/holyiot_yj16019.dts

 - [ ] boards/others/black_f407ve/doc/index.rst
   * [ ] boards/others/black_f407_generic/black_f407ve_defconfig

 - [ ] boards/phytec/phyboard_lyra_am62x/doc/index.rst
   * [ ] boards/phytec/phyboard_lyra_am63x/phyboard_lyra_am62x_am6234_m4_defconfig

 - [ ] boards/raspberry_pi/rpi_4b/doc/index.rst
   * [ ] boards/raspberrypi/rpi_4b/rpi_4b_defconfig

 - [ ] boards/seeed_studio/xiao_ble/doc/index.rst
   * [ ] boards/xiao_ble/xiao_ble_common.dtsi

 - [ ] boards/sparkfun/sparkfun_thing_plus/doc/index.rst
   * [ ] boards/sparkfun/sparkfun_thing_plus_nrf9160/sparkfun_thing_plus_nrf9160_common.dtsi

 - [ ] boards/st/stm32l562e_dk/doc/index.rst
   * [ ] boards/arm/stm32l562e_dk/stm32l562e_dk.dts
   * [ ] boards/st/stm32l562e_dk/stm32l562e_dk_ns_defconfig
   * [ ] boards/st/stm32l562e_dk/stm32l562e_dk_ns.dts

 - [ ] boards/ti/cc3220sf_launchxl/doc/index.rst
   * [ ] boards/ti/cc3220sf_launchxl/pinmux.c

 - [ ] boards/ti/cc3235sf_launchxl/doc/index.rst
   * [ ] boards/ti/cc3235sf_launchxl/pinmux.c

 - [ ] boards/ublox/ubx_bmd345eval/doc/index.rst
   * [ ] boards/ublox//ubx_bmd340eval/ubx_bmd345eval_nrf52840.dts

 - [ ] boards/ublox/ubx_evkninab3/doc/index.rst
   * [ ] boards/ublox/ubx_evkninab3/ubx_ninab3_nrf52840.dts

 - [ ] boards/weact/blackpill_f401cc/doc/index.rst
   * [ ] boards/weact/blackpill_f401cc/blackpill_f401ce_defconfig

 - [ ] boards/weact/blackpill_f411ce/doc/index.rst
   * [ ] boards/weact/stm32_blackpill_v2/stm32_blackpill_v2_defconfig

 - [ ] boards/ite/it82xx2_evb/doc/index.rst
   * [ ] boards/riscv/it82xx2_evb/it82xx2_evb_defconfig

 - [ ] boards/ite/it8xxx2_evb/doc/index.rst
   * [ ] boards/riscv/it8xxx2_evb/it8xxx2_evb_defconfig

 - [ ] boards/google/google_twinkie_v2/doc/index.rst
   * [ ] boards/arm/google_twinkie_v2/google_twinkie_v2_defconfig

 - [ ] boards/nuvoton/npcx4m8f_evb/doc/index.rst
   * [ ] boards/arm/npcx4m8f_evb/npcx4m8f_evb_defconfig

 - [ ] boards/nuvoton/npcx7m6fb_evb/doc/index.rst
   * [ ] boards/arm/npcx7m6fb_evb/npcx7m6fb_evb_defconfig

 - [ ] boards/nuvoton/npcx9m6f_evb/doc/index.rst
   * [ ] boards/arm/npcx9m6f_evb/npcx9m6f_evb_defconfig

 - [ ] boards/st/b_g474e_dpow1/doc/index.rst
   * [ ] boards/arm/b_g474e_dpow1/b_g474e_dpow1_defconfig

 - [ ] boards/st/nucleo_g031k8/doc/index.rst
   * [ ] boards/arm/nucleo_g031k8/nucleo_g031k8_defconfig

 - [ ] boards/st/nucleo_g070rb/doc/index.rst
   * [ ] boards/arm/nucleo_g070rb/nucleo_g070rb_defconfig

 - [ ] boards/st/nucleo_g071rb/doc/index.rst
   * [ ] boards/arm/nucleo_g071rb/nucleo_g071rb_defconfig

 - [ ] boards/st/nucleo_g0b1re/doc/index.rst
   * [ ] boards/arm/nucleo_g0b1re/nucleo_g0b1re_defconfig

 - [ ] boards/st/nucleo_g431rb/doc/index.rst
   * [ ] boards/arm/nucleo_g431rb/nucleo_g431rb_defconfig

 - [ ] boards/st/nucleo_g474re/doc/index.rst
   * [ ] boards/arm/nucleo_g474re/nucleo_g474re_defconfig

 - [ ] boards/st/stm32g0316_disco/doc/index.rst
   * [ ] boards/arm/stm32g0316_disco/stm32g0316_disco_defconfig

 - [ ] boards/st/stm32g071b_disco/doc/index.rst
   * [ ] boards/arm/stm32g071b_disco/stm32g071b_disco_defconfig

 - [ ] boards/st/stm32g081b_eval/doc/index.rst
   * [ ] boards/arm/stm32g081b_eval/stm32g081b_eval_defconfig

 - [ ] boards/weact/weact_stm32g431_core/doc/index.rst
   * [ ] boards/arm/weact_stm32g431_core/weact_stm32g431_core_defconfig

 - [ ] boards/xen/xenvm/doc/index.rst
   * [ ] boards/arm64/xenvm/Kconfig.defconfig
   * [ ] boards/arm64/xenvm/xenvm_defconfig

nordicjm, 2024-02-16:
@erwango @fabiobaltieri @nordicjm please check above and submit fix PRs

nordicjm, 2024-02-16:
@golowanow

golowanow, 2024-02-16:
@nordicjm no docs with broken references yet in my scope

golowanow, 2024-02-16:
> > @nordicjm no docs with broken references yet in my scope
> 
> Right you are, I ported nios2

wait i see there is this one:
https://github.com/zephyrproject-rtos/zephyr/actions/runs/7928256606/job/21646167791?pr=68178
```
/runner/_work/zephyr/zephyr/doc/_build/src/boards/boards_legacy/x86/index.rst:6: WARNING: toctree glob pattern '[!common]*/**/*' didn't match any documents
```

fabiobaltieri, 2024-02-16:
hey I'll just send another followup and do them all, no sense sending 1000 PRs for this

fabiobaltieri, 2024-02-16:
https://github.com/zephyrproject-rtos/zephyr/pull/69074

golowanow, 2024-02-16:
#69075

nordicjm, 2024-02-19:
@erwango updated list above

erwango, 2024-02-19:
> @erwango updated list above

Thanks. Will fix

erwango, 2024-02-19:
@nordicjm https://github.com/zephyrproject-rtos/zephyr/pull/69205

closed 2024-02-27 by nordicjm