zephyrproject-rtos/zephyr #69088: nxp_rt1010: Flash driver does not work

nguyenmthien-gin, 2024-02-16:
**Describe the bug**
The flash driver no longer working with the NXP iMXRT1170 development board. All firmwares built with `CONFIG_FLASH=y` cannot boot pass the `POST_KERNEL` system init stage.

We are able to confirm this is found from [this commit](https://github.com/zephyrproject-rtos/zephyr/commit/f81113e94811e2e7381f9a6c78dc5f0e29bc6bd7) onwards.

**To Reproduce**
Here we use `mimxrt1170_evk_cm7` with the Flash Shell example, but please notice this behavior is seen across the board with all kind of applications:
1. `cd zephyr`
1. `west build --pristine always --board mimxrt1170_evk_cm7 samples/drivers/flash_shell`
1. `west flash`
1. monitor the UART console 

**Expected behavior**
There should be the Zephyr boot message printed to the console, followed by the shell prompt.

In this case, nothing is printed to the console.

**Impact**
We are unable to confirm if this only happens to other boards in the series or not.

nguyenmthien-gin, 2024-02-16:
@danieldegrasse Do you have any thoughts on this?

danieldegrasse, 2024-02-16:
Updating the title, please let me know if this is incorrect. I'll see if I can reproduce the issue on my hardware. I assume you are using the RT1170 EVK revision A?

danieldegrasse, 2024-02-16:
Confirmed I can reproduce this issue on `2f8c58120e`

danieldegrasse, 2024-02-16:
@nguyenmthien-gin can you test #69102 to verify it fixes the issue on your board? For me, this fixes the issue on `2f8c58120e`. It looks like the clock frequency for the flash was being set higher than desired, leading to the crash.

closed 2024-02-17 by henrikbrixandersen

nguyenmthien-gin, 2024-02-19:
> @nguyenmthien-gin can you test #69102 to verify it fixes the issue on your board? For me, this fixes the issue on `2f8c58120e`. It looks like the clock frequency for the flash was being set higher than desired, leading to the crash.

We verify that this PR fixes our boards. Thank you all for your blazing fast PR and the merging process :smile:

PR #69102 (2024-02-17): soc: arm: nxp_imx: fix flexspi frequency setting for iMXRT11xx SOC
soc: arm: nxp_imx: fix flexspi frequency setting for iMXRT11xx SOC

Commit a10fee2d5e9 (drivers: clock_control: ccm_rev2: add support for
reclocking FlexSPI) introduced the ability to set the FlexSPI
clock frequency at runtime on RT11xx series SOCs. However, this
implementation resulted in the clock frequency being rounded up, not
down. This can result in flash clock frequency violations on some
flash parts, causing the system to crash when running in XIP mode.

Fixes #69088
commit 9a96a0c5 (2024-02-17): soc: arm: nxp_imx: fix flexspi frequency setting for iMXRT11xx SOC

Commit a10fee2d5e9 (drivers: clock_control: ccm_rev2: add support for
reclocking FlexSPI) introduced the ability to set the FlexSPI
clock frequency at runtime on RT11xx series SOCs. However, this
implementation resulted in the clock frequency being rounded up, not
down. This can result in flash clock frequency violations on some
flash parts, causing the system to crash when running in XIP mode.

Fixes #69088