zephyrproject-rtos/zephyr #69109: [Coverity CID: 316544] Use of 32-bit time_t in include/zephyr/sys/timeutil.h

zephyrbot, 2024-02-17:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/adeaa63bb71aba5c1732816df21f40d654e26cd4/include/zephyr/sys/timeutil.h

Category: High impact quality
Function: ``
Component: Other
CID: [316544](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=316544)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/adeaa63bb71aba5c1732816df21f40d654e26cd4/include/zephyr/sys/timeutil.h#L62

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=316544

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_Y2K38_SAFETY). ([CWE-197](http://cwe.mitre.org/data/definitions/197.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

nashif, 2024-02-20:
and old issue (first reported in may 2023) see https://github.com/zephyrproject-rtos/zephyr/issues/58710.

andyross, 2024-02-20:
Not sure what to say about this one.  "time_t" is, indeed, a dangerous 32 bit type, and this call uses it.  But it's library code, not app code, and Zephyr itself doesn't actually use this (largely unmaintained) subsystem layer for anything significant (I see a few counters that want it when IMHO they should be using uptime, and one spot in the shell that exposes it).

Maybe we should just declare it a dumb API we shouldn't have exposed, deprecate it, and port the existing users over to the otherwise identical timeutil_timegm64()?  Sort of above my pay grade; I try to stay out of API bikesheds. :)

henrikbrixandersen, 2024-02-21:
> Maybe we should just declare it a dumb API we shouldn't have exposed, deprecate it, and port the existing users over to the otherwise identical timeutil_timegm64()?

Sounds fair to me.