zephyrproject-rtos/zephyr #69115: [Coverity CID: 347204] Out-of-bounds access in drivers/modem/modem_cellular.c

zephyrbot, 2024-02-17:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/adeaa63bb71aba5c1732816df21f40d654e26cd4/drivers/modem/modem_cellular.c

Category: Memory - corruptions
Function: `modem_cellular_init`
Component: Drivers
CID: [347204](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=347204)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/adeaa63bb71aba5c1732816df21f40d654e26cd4/drivers/modem/modem_cellular.c#L1485

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=347204

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

henrikbrixandersen, 2024-02-17:
I am tempted to close this as a false positive. How could the size of a `struct modem_cmux` be different when passed to `memset()` using `sizeof(*cmux)`?

[image]

rerickson1, 2024-02-19:
maybe it should be `sizeof(struct modem_cmux)` instead. Dereferencing to `*cmux` may be getting the sizeof the first element in that struct?

henrikbrixandersen, 2024-02-19:
> maybe it should be `sizeof(struct modem_cmux)` instead. Dereferencing to `*cmux` may be getting the sizeof the first element in that struct?

That's now how `sizeof()` should work. There are plenty of examples of using `sizeof(*foo)` in tree. Besides, the first element of `struct modem_cmux` is a pointer type.

aescolar, 2024-02-19:
@henrikbrixandersen it seems that for some reason coverity thinks the size of sizeof(struct modem_cmux) in  modem_cellular.c (288) is different than in modem_cmux.c (292).
I have the impression coverity has a tendency to mix builds from different targets. So maybe in one target that struct size is 288 and in another 292. (I did a quick check for native_sim, qemu_x86 and the nrf52833dk and the size was consistently 288, 304 and 320 bytes respectively in both files)

henrikbrixandersen, 2024-02-19:
@aescolar Thanks. So you agree this looks like a false positive?

aescolar, 2024-02-20:
> @aescolar Thanks. So you agree this looks like a false positive?

I think so.

henrikbrixandersen, 2024-02-21:
Marked as false positive in Coverity.

closed 2024-02-21 by henrikbrixandersen