zephyrproject-rtos/zephyr #69119: [Coverity CID: 347188] Logically dead code in drivers/bbram/bbram_it8xxx2.c

zephyrbot, 2024-02-17:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/adeaa63bb71aba5c1732816df21f40d654e26cd4/drivers/bbram/bbram_it8xxx2.c

Category: Control flow issues
Function: `bbram_it8xxx2_init`
Component: Drivers
CID: [347188](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=347188)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/adeaa63bb71aba5c1732816df21f40d654e26cd4/drivers/bbram/bbram_it8xxx2.c#L94

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=347188

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_DEADCODE). ([CWE-561](http://cwe.mitre.org/data/definitions/561.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

RuibinChang, 2024-02-19:
Hi @keith-zephyr and @fabiobaltieri,

Where to know line 94 ` *(base_addr + i) = 0;` error in details?
I guess the details are in the link https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=347188, but I login by my GitHub account, then it shows "401: Unauthorized Sorry, your credentials are not valid for this resource."

str4t0m, 2024-02-19:
[image]

RuibinChang, 2024-02-23:
I have a draft change for this issue in https://github.com/RuibinChang/zephyr/commit/14aa4b5df67f7bfdcd459107ede7f80dc6982cd5

Test the change by running:
west build -p always -b it8xxx2_evb tests/drivers/bbram/
[image]
The test result is pass, but it doesn't run any test item (because tests/drivers/bbram/src/main.c  main function is empty).

So I build tests/drivers/bbram/generic and tests/drivers/bbram/emul on main branch (without my changes), but there are build errors.
(emul test)
[image]

(generic test)
[image]
[image]
[image]

Hi @yperess, there are commits about bbram test in https://github.com/zephyrproject-rtos/zephyr/pull/67191
I would like to know if the bbram tests are ready for running. 
If ready, then how to run the BBRAM test on it8xxx2_evb board?

yperess, 2024-02-23:
I'm OK with removing the loop, but we should check with ITE. To run the tests you can run `west twister -cviG -T tests/drivers/bbram/`

PR #69495 (2024-02-29): bbram: fix bbram dead code issue
Clear the whole 192 bytes of bbram, before writing the magic value to it. test pass "west twister -cviG -T tests/drivers/bbram/".

Fixes #69119
commit 19a23d51 (2024-02-29): bbram: fix bbram dead code issue

Clear the whole 192 bytes of bbram, before writing the magic value to it.
test pass "west twister -cviG -T tests/drivers/bbram/".

Fixes #69119
closed 2024-02-29 by fabiobaltieri