zephyrproject-rtos/zephyr #69148: Kernel function z_impl_k_wakeup is not interrupt safe, can result in invalid thread state and assert

helpfulchicken, 2024-02-17:
I've been having issues with using k_sleep in the main thread, and using k_wakeup in other threads and in interrupts to wake the main thread prematurely. Usually it is fine, but rarely I end up with the ASSERT in z_tick_sleep tripping because the main thread resumes from sleep while its state is still marked suspended.

I believe I've narrowed down the problem to this step in z_impl_k_wakeup, where it performs a bitwise clearing of the suspended bit from the state variable. I don't believe that is an atomic action, at least not on all platforms (I'm developing in ARMv7-M).
https://github.com/zephyrproject-rtos/zephyr/blob/7c9dbde659d03ba4733157c493c07d5fd32988a9/kernel/sched.c#L1656
The function k_wakeup includes a bitwise clearing of the suspended bit from the state variable, but it isn’t protected from interrupts.

I believe what's happening is the following:

1. Initially Thread B (main thread) is sleeping, after a k_sleep call.
2. Thread A: k_wakeup starts, trying to wake up Thread B, thread state of Thread B is currently just suspended.
3. Thread A: k_wakeup prepares to clear the “suspended” bit. It calculates that the new Thread B state will be 0x0.
    a. An interrupt occurs. In the interrupt handler, k_wakeup runs, trying to wakeup Thread B. It clears the suspended state, queues Thread B, and the thread state of Thread B has queued bit set. 
    b. Interrupt finishes. Thread A resumes.
4. Thread A: resuming work in the middle of the earlier k_wakeup call
5. Thread A: The thread state of Thread B is set to the prepared new value, 0x0. This wipes the queued bit that was set in the interrupt. Thread B is now in the ready queue, but doesn't have its queued bit set.
6. Thread A: Finished the k_wakeup function, Thread B is queued (second time), and the thread state of Thread B has queued bit set.
At this point, Thread B is in the kernel scheduler ready queue twice
7. Later, scheduler runs Thread B, wiping the queued bit. Now Thread B is in the scheduler ready queue once, without queued state bit.
8. Thread B: sleeps again, setting its thread state suspended bit, and the scheduler runs the next thread in the ready queue
9. Thread B: immediately wakes because it was the next in the scheduler ready queue
10. Thread B: resuming in z_tick_sleep, the thread state of Thread B is still suspended, because it was not supposed to be in the ready queue and so it didn't clear its suspended state before resuming.
11. Thread B: The program sees that the thread woke up while still suspended, so the program asserts in z_tick_sleep and crashes.

I have a suggested fix to solve this bug. In z_impl_k_wakeup, wrap the call to z_mark_thread_as_not_suspended in a spin lock.
```
	K_SPINLOCK(&sched_spinlock) {
		z_mark_thread_as_not_suspended(thread);
	}
```

I'm developing with nRF Connect SDK v2.5.2, which appears to be using Zephyr 3.4.99. If that doesn't mean anything to you, then their release notes say they're using all commits from Zephyr up to and including a768a05e6205e415564226543cee67559d15b736

henrikbrixandersen, 2024-02-17:
Thank you for reporting this. However - unless you are able to reproduce this issue with upstream Zephyr `main` - please report issues with the nRF Connect SDK (NCS) on the [Nordic Semiconductor DevZone](https://devzone.nordicsemi.com/).

helpfulchicken, 2024-02-18:
I believe I can reproduce this on Zephyr `main`, but it will be a few days at least before I have time to work on that. I don't have an existing pure Zephyr build environment ready.

The key point of this report is that the line 
https://github.com/zephyrproject-rtos/zephyr/blob/a6e9f12ca85bcb2f3a8e264dd5b0e0e4b7fc1f38/kernel/include/ksched.h#L152
in `z_mark_thread_as_not_suspended` is being used in `z_impl_k_wakeup` without wrapping in a spin lock, so it is possible (but rare) that an interrupt can occur during that instruction, resulting in an invalid thread state if the thread state is modified in the interrupt. The steps I described my report are one possible consequence of this vulnerability.

For a little more context in case it is relevant, in my project the zephyr kernel is using a single CPU with the "dumb" scheduler.

peter-mitsis, 2024-02-20:
I believe that @helpfulchicken is correct that some locking is required and that the described situation can indeed happen. Very nice catch!

andyross, 2024-02-20:
FWIW, the bug is absolutely correct.  The suggested fix is not.  Locking the access to the bitfield just ensures that the state word can't be corrupted by racing access, it does nothing to prevent the actual scheduler state from getting mixed up by racing contexts (e.g. consider another thread trying to suspend, or even abort, the thread you are in the process of "waking up" while you're operating on it!).

Oddly k_thread_resume() has correct locking, but k_wakeup() never got it.  To be fair: both of these are inherently race-prone APIs and I try to warn apps away from using either of them.  Any decision by a thread to "go to sleep" needs to be atomic with all changes to the state required to make that decision, and the API doesn't provide that.  Consider a condition variable for this kind of code, or restructuring the architecture to use a k_sem/k_fifo to handle work out of a "queue".

andyross, 2024-02-20:
Give me a moment, should be a quick fix.

peter-mitsis, 2024-02-20:
@andyross - Looks like we both were doing the fix at the same time. I just saw your comments now.

PR #69255 (2024-02-26): kernel: Update k_wakeup()
This commit does two things to k_wakeup():

1. It locks the scheduler before marking the thread as not suspended. As the the clearing of the _THREAD_SUSPENDED bit is not atomic, this helps ensure that neither another thread nor ISR interrupts this action (resulting in a corrupted thread_state).

2. The call to flag_ipi() has been removed as it is already being made within ready_thread().

Fixes #69148
PR #69436 (2024-03-12): [Backport v3.6-branch] kernel: Update k_wakeup()
Backport 51ae993c12629d082f847885f52249432fc31bb2 from #69255.

Fixes: #69148
closed 2024-02-26 by nashif