zephyrproject-rtos/zephyr #69175: build zephyr sample failed with zephyr 3.6.0-rc2

oska874, 2024-02-19:
I install zephyr follow instruct from zephyr [document](https://docs.zephyrproject.org/latest/develop/getting_started/index.html). Install zephyr-sdk-0.16.5, just installed arm-zephyr-eabi and register cmake package, do not install host tools.

but build failed as below:
```
west build -b pico_pi_m4  samples/hello_world
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /root/zephyrproject/zephyr/samples/hello_world
-- CMake version: 3.28.3
-- Cache files will be written to: /root/.cache/zephyr
-- Zephyr version: 3.6.0-rc2 (/root/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: pico_pi_m4
-- Found host-tools: zephyr 0.16.5 (/opt/zephyr-sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (/opt/zephyr-sdk-0.16.5)
-- Found BOARD.dts: /root/zephyrproject/zephyr/boards/arm/pico_pi_m4/pico_pi_m4.dts
-- Generated zephyr.dts: /root/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /root/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /root/zephyrproject/zephyr/build/zephyr/dts.cmake
CMake Error at /root/zephyrproject/zephyr/cmake/modules/dts.cmake:355 (message):
  command failed with return code: No such file or directory
Call Stack (most recent call first):
  /root/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  /root/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /root/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:97 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /workspace/bin/cmake-3.28.3-linux-x86_64/bin/cmake -DWEST_PYTHON=/workspace/miniconda3/envs/zephyr/bin/python -B/root/zephyrproject/zephyr/build -GNinja -S/root/zephyrproject/zephyr/samples/hello_world
```

I've tried build boards mps2-an521 and pico_pi_m4,both failed.

nordicjm, 2024-02-19:
Did you follow the install guide and install `dtc`?

henrikbrixandersen, 2024-02-19:
@oska874 Also, you _really_ shouldn't run these build commands as `root` on any system.