zephyrproject-rtos/zephyr #69190: Support Analog Devices Blackfin Embedded Processors

escherstair, 2024-02-19:
For some usage scenarios it would be useful if Zephyr would support [Blackfin Embedded Processors](https://www.analog.com/en/product-category/blackfin-embedded-processors.html) from Analog Devices.
This is a family of emebedded prcoessors, including a performant DSP engine.
The hardware is nice, but from the OS/software point of view, support from ADI is awful.
For the RTOS there is (was) old [VDK](https://www.analog.com/media/en/dsp-documentation/software-manuals/50_vdk_mn_rev_3.5.pdf) (VisualDSP Kernel) or FreeRTOS, but not for all the members of the family.

No security updates are available, and maintaining applications is a nightmare.

nordicjm, 2024-02-19:
You should probably contact adi about this

escherstair, 2024-02-19:
I'll try, but based on what happened in the past, it doesn't seem that ADI has interest in providing RTOS for BFxxx family.
It seems that it considers RTOS as "third party" core business.

escherstair, 2024-03-06:
[ADI is not interested](https://ez.analog.com/dsp/software-and-development-tools/cces/f/q-a/578285/add-support-for-zephyr-os) in this porting (at the moment).

closed 2024-03-06 by nordicjm