zephyrproject-rtos/zephyr #69195: nanopb broken in v3.5 and later

xhpohanka, 2024-02-19:
**Describe the bug**
Nanopb sample cannot be compiled

**To Reproduce**
Steps to reproduce the behavior:
1. checkout latest main or v3.5
2. build nanopb sample

```
west build -b native_posix
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /tmp/z/zephyr/samples/modules/nanopb
-- CMake version: 3.28.1
-- Found Python3: /usr/bin/python (found suitable version "3.11.6", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/honza/.cache/zephyr
-- Zephyr version: 3.6.0-rc2 (/tmp/z/zephyr)
-- Found west (found suitable version "1.1.0", minimum required is "0.14.0")
-- Board: native_posix
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.4 (/home/honza/zephyr-sdk-0.16.4)
-- Found toolchain: host (gcc/ld)
-- Found Dtc: /home/honza/zephyr-sdk-0.16.4/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /tmp/z/zephyr/boards/posix/native_posix/native_posix.dts
-- Generated zephyr.dts: /tmp/z/zephyr/samples/modules/nanopb/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /tmp/z/zephyr/samples/modules/nanopb/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /tmp/z/zephyr/samples/modules/nanopb/build/zephyr/dts.cmake

warning: NANOPB (defined at modules/nanopb/Kconfig:7) was assigned the value 'y' but got the value
'n'. Check these unsatisfied dependencies: 0 (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_NANOPB and/or look up NANOPB in the
menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration Values,
and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

Parsing /tmp/z/zephyr/Kconfig
Loaded configuration '/tmp/z/zephyr/boards/posix/native_posix/native_posix_defconfig'
Merged configuration '/tmp/z/zephyr/samples/modules/nanopb/prj.conf'
Configuration saved to '/tmp/z/zephyr/samples/modules/nanopb/build/zephyr/.config'
Kconfig header saved to '/tmp/z/zephyr/samples/modules/nanopb/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /usr/bin/ld.bfd (found version "2.41.0") 
-- The C compiler identification is GNU 13.2.1
-- The CXX compiler identification is GNU 13.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/lib/ccache/bin/gcc
-- Using ccache: /usr/bin/ccache
CMake Error at /tmp/z/zephyr/modules/nanopb/nanopb.cmake:9 (find_package):
  By not providing "FindNanopb.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Nanopb", but
  CMake did not find one.

  Could not find a package configuration file provided by "Nanopb" with any
  of the following names:

    NanopbConfig.cmake
    nanopb-config.cmake

  Add the installation prefix of "Nanopb" to CMAKE_PREFIX_PATH or set
  "Nanopb_DIR" to a directory containing one of the above files.  If "Nanopb"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  CMakeLists.txt:9 (include)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python -B/tmp/z/zephyr/samples/modules/nanopb/build -GNinja -DBOARD=native_posix -S/tmp/z/zephyr/samples/modules/nanopb
```

closed 2024-02-19 by henrikbrixandersen

NZSmartie, 2024-02-22:
How can the optional module be specified in the west manifest and not in the west configuration file?

Up until Zephyr v3.3.0, I've been using a name-allowlist to optionally include modules I need:
```yaml
manifest:
  remotes:
    - name: zephyrproject-rtos
      url-base: https://github.com/zephyrproject-rtos
  self:
    path: app

  projects:
    - name: zephyr
      remote: zephyrproject-rtos
      revision: v3.5.0
      west-commands: scripts/west-commands.yml
      import:
        name-allowlist:
          - cmsis
          - hal_nordic
          - mbedtls
          - mcuboot
          - nanopb
          - openthread
          - tinycrypt
          - zcbor
```
---
**_Update_**
Adding the following to `west.yaml` enables all optional modules. Since i'm already using allowlist for specifying which modules to include, it doesn't make sense to specify nanopb a second time... I'm leaving this here as a better explanation for others facing the same issue.

```yaml
manifest:
  # ...
  group-filter:
    - +optional # Enable all optional modules in Zephyr's submanifests. Since name-allowlist is already employed to specify exactly what's needed
  projects:
    - name: zephyr
      # ...
      import:
        name-allowlist:
```

xhpohanka, 2024-02-22:
> I'm leaving this here as a better explanation for others facing the same issue.

Thanks for pointing this out. I wanted to find the solution with `west.yml` but you were faster. I even use the same way with `name-allowlist` now.