zephyrproject-rtos/zephyr #69199: dts: nios2: dma devictree node has incorrect address or reg

cfriedt, 2024-02-19:
**Describe the bug**
The address and reg property of the dma devicetree node for altera_max10 is incorrect and it's causing CI failures.

https://github.com/zephyrproject-rtos/zephyr/blob/main/dts%2Fnios2%2Fintel%2Fnios2f.dtsi

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? Ci
 - What have you tried to diagnose or workaround this issue? Looked at the error. Mentioned it in the release channel.
 - Is this a regression? Unsure
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. Look at the errors in CI and / or build an app with altera_max10 as the target 

**Expected behavior**
Build / CI should succeed.

**Impact**
It's an annoyance, mainly because it causes unrelated PRs to fail.

**Logs and console output**
```
-- Found BOARD.dts: /__w/zephyr/zephyr/boards/nios2/altera_max10/altera_max10.dts
unit address and first address in 'reg' (0x1002c0) don't match for /soc/dma@100200
-- Generated zephyr.dts: /__w/zephyr/zephyr/twister-out/altera_max10/tests/posix/headers/portability.posix.headers.picolibc.with_posix_api/zephyr/zephyr.dts
```

[image]

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) Linux
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK
 - Commit SHA or Version used: 

**Additional context**
Just happened to see this in CI and mentioned it in the release channel on Discord while I was out. I'll mention it as well in the nios2 channel and Intel channels too

henrikbrixandersen, 2024-02-19:
The address mismatch was introduced when the `dma` node was added in b60b4be4e3d23653f3634995178d3ac6be91d933 back in March 2021.

cfriedt, 2024-02-19:
```
yr/dts.cmake
/__w/zephyr/zephyr/twister-out/altera_max10/tests/posix/headers/portability.posix.headers.with_posix_api/zephyr/zephyr.dts:66.19-71.5: Warning (simple_bus_reg): /soc/dma@100200: simple-bus unit address format error, expected "1002c0"
/__w/zephyr/zephyr/twister-out/altera_max10/tests/posix/headers/portability.posix.headers.with_posix_api/zephyr/zephyr.dts:57.20-65.5: Warning (unique_unit_address_if_enabled): /soc/i2c@100200: duplicate unit-address (also used in node /soc/dma@100200)
Parsing /__w/zephyr/zephyr/Kconfig
Loaded configuration '/__w/zephyr/zephyr/twister-out/altera_max10/tests/posix/headers/portability.posix.headers.with_posix_api/zephyr/.config'
```
Encountered in a completely unrelated PR
https://github.com/zephyrproject-rtos/zephyr/issues/69199

henrikbrixandersen, 2024-02-19:
> ```
> yr/dts.cmake
> /__w/zephyr/zephyr/twister-out/altera_max10/tests/posix/headers/portability.posix.headers.with_posix_api/zephyr/zephyr.dts:66.19-71.5: Warning (simple_bus_reg): /soc/dma@100200: [...]

Those are warnings, not errors?

nashif, 2024-03-19:
a warning, we have many of those. Considering dropping those platrforms now, closing as wontfix.

closed 2024-03-19 by nashif