zephyrproject-rtos/zephyr #69216: driver: Timer / Counter initialization incorrect for timer2 with BLE on nRF52840

carlst, 2024-02-19:
**Describe the bug**
An initialization value is incorrect if timer2 is enabled (and used in counter mode) with BLE (where timer0 and timer1 are disabled) on nRF52840.

Please also mention any information which could help others to understand
the problem you're facing:
 - `drivers/counter/counter_nrfx_timer.c` uses a Nordic HAL value to initialize `.timer`.
This is out-of-sync with devicetree information as timer0 and timer1 are disabled but
timer2 is enabled because:
[devicetree](https://docs.zephyrproject.org/latest/build/dts/api/api.html#c.DT_INST)
implies that devicetree starts indexing instances from 0 based on enabled devices, and then indexes disabled devices. So, devicetree index for timer2 (0) and HAL index for timer2 (2) differ.

 - see: commit 9a27906e38343b07cbdf1b4aca5451c5886dbd41

**To Reproduce**
Steps to reproduce the behavior:
1. Use `samples/bluetooth/broadcaster`
2. Add following to `prj.conf `
```
CONFIG_COUNTER=y
CONFIG_NRFX_TIMER2=y
```
3. Create `boards/nrf52840dk_nrf52840.overlay` with:
```
&timer2 {
  status = "okay";
};
```
4. Build and run on nRF52840DK
5. Check value of `reg` in `init_timer`. It is 0x40008000 and should be 0x4000a000

**Expected behavior**
timer2 should operate correctly as a counter.

**Impact**
Very low priority (workaround is to revert commit above).

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: zephyr-sdk-0.16.5
 - Commit SHA or Version used: 5c455aae88d14a527bdd438adf9da4958d52c17b

cvinayak, 2024-02-25:
FYI @aescolar

aescolar, 2024-02-25:
(@cvinayak thanks for the ping)
There is a fix for this ready coming in this PR #68618
CC @magp-nordic

PR #69966 (2024-03-11): drivers/counter nrfx: Fix with DT instance not matching device instance
478530ec0aa1fe5f481786c25d50f7a081b22208 introduced a bug where if the DT index while iterating its DT structure initialization does not match the actual peripheral instance, or if the device instance string is not just a simple integer, but a more complex string like "00", or "02", either the wrong peripheral address would be used, or the file would failt to compile.

Let's fix this by reverting that change, and instead, for simulation converting the hardcoded DT/real HW address to the valid addr for simulation on the fly [... cut]
PR #68618 (2024-03-12): Add missing tests alignments for nrf54l15
This PR adds missing GRTC test and counter test alignment for nRF54L15. ~~Additionally, it contains fix for getting timer's instance number in the counter driver.~~

~~Fixes #69216~~ Fixed by #69966
PR #70052 (2024-03-14): [Backport v3.6-branch] drivers/counter nrfx: Fix with DT instance not matching device instance
Backport f9d5e8458ce261d6c3574d50d3b16ea031419279~2..f9d5e8458ce261d6c3574d50d3b16ea031419279 from #69966.

Fixes #69216
closed 2024-03-11 by fabiobaltieri