zephyrproject-rtos/zephyr #69265: Can't start debug on ESP Wrover

laurenmurphyx64, 2024-02-20:
**Describe the bug**
Unable to debug on `esp_wrover_kit` on latest main (19dfdb9)

**To Reproduce**
Steps to reproduce the behavior:

```
west build -p -b esp_wrover_kit samples/hello_world
west debug
```

**Expected behavior**
Debugging works

**Impact**
Showstopper

**Logs and console output**
```
$ west debug
-- west debug: rebuilding
[1/1] cd /home/lauren/zephyrproject/zephyr/build/zephyr && /home/lau...--elf-file=/home/lauren/zephyrproject/zephyr/build/zephyr/zephyr.elf
-- west debug: using runner openocd
-- runners.openocd: OpenOCD GDB server running on port 3333; no thread info available
Open On-Chip Debugger 0.11.0+dev-00726-ge2b6f5655 (2023-11-15-11:07)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
/home/lauren/zephyrproject/zephyr/boards/xtensa/esp_wrover_kit/support/openocd.cfg:4: Error: Can't find interface/ftdi/esp32_devkitj_v1.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 26
at file "/home/lauren/zephyrproject/zephyr/boards/xtensa/esp_wrover_kit/support/openocd.cfg", line 4
GNU gdb (Zephyr SDK 0.16.4) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-espressif_esp32_zephyr-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://github.com/zephyrproject-rtos/sdk-ng/issues>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/lauren/zephyrproject/zephyr/build/zephyr/zephyr.elf...
:3333: Connection timed out.
Warning: 'flushregs', an alias for the command 'maintenance flush register-cache', is deprecated.
Use 'maintenance flush register-cache'.

Register cache flushed.
"monitor" command not supported by this target.
No hardware breakpoint support in the target.
(gdb) 
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: 19dfdb9

nashif, 2024-02-27:
@sylvioalves  can you please take a look?

marekmatej, 2024-02-28:
Hi @laurenmurphyx64, thanks for pointing this out. The Zephyr SDK version of OpenOCD is not compatible with the ESP32 SoC, nor it doesn't contain the necessary config files.
To debug the ESP32 targets you need to download and install the newest OpenOCD from https://github.com/espressif/openocd-esp32/

sylvioalves, 2024-02-28:
ESP32 boards requires custom OpenOCD as point above. Upstream version is indeed ready or almost ready to be used. Once Zephyr SDK/OpenOCD gets updated, that would make things more straightformward.  

For some reason, most all esp32-based boards documentation got this information removed:  
```c
Debugging
=========

As with much custom hardware, the ESP32S3 modules require patches to
OpenOCD that are not upstreamed yet. Espressif maintains their own fork of
the project. The custom OpenOCD can be obtained at `OpenOCD ESP32`_

The Zephyr SDK uses a bundled version of OpenOCD by default. You can overwrite that behavior by adding the
``-DOPENOCD=<path/to/bin/openocd> -DOPENOCD_DEFAULT_PATH=<path/to/openocd/share/openocd/scripts>``
parameter when building.
```
The above would do the debugging to work.

laurenmurphyx64, 2024-02-28:
@sylvioalves @marekmatej Thanks for your help! I can leave this open if you want to make changes to the docs or close it.

closed 2024-05-13 by github-actions[bot]

adrain-yu, 2024-07-01:
$ west debug
-- west debug: rebuilding
ninja: no work to do.
-- west debug: using runner openocd
-- runners.openocd: OpenOCD GDB server running on port 3333; no thread info available
Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:25)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
DEPRECATED! use 'adapter speed' not 'adapter_khz'
adapter speed: 5000 kHz
Info : Listening on port 6333 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: unable to open ftdi device with description '*', serial '*' at bus location '*'