zephyrproject-rtos/zephyr #69297: Zephyr UpdateHub sample issues

RafalHalik, 2024-02-21:
**I've been working with the OTA updatehub sample built for mimxrt1060_evk.**

I've come across some issues with json parse in the mcuboot updatehub agent. When a package is uploaded for a rollout using updatehub-ce in a docker container. The device does a check to pull the update down i get the following error in a minicom terminal window...
`` minicom -D /dev/ttyACM0`` 

`` <err> updatehub: Could not parse json ``

However the update would not fail imedietly but after a minute (interval to check for an update again) it would try pull the meta data about the update once again and fail in the same way.
I decided to by-pass the cleanup that happens if the json parse fails and now all checks to do with meta data from the json parse fail and print there error messages... 

``  <err> updatehub: Could not parse json ``
``  <err> updatehub: Could not parse json ``
``  <err> updatehub: SHA256 size is invalid ``
``  <err> eth_mcux: ENET_GetRxFrameSize return: 4001 ``

and with this change the update fails and doesn't try again after the minute.
 

**To Reproduce**
-Following the steps from https://docs.zephyrproject.org/latest/samples/subsys/mgmt/updatehub/README.html
The only difference being building for a mimxrt1060_evk board. 

- Run docker container for updatehub-ce 
 ``docker run -it -p 8080:8080 -p 5683:5683/udp --rm updatehub/updatehub-ce:latest ``
- In browser connect to ``localhost:8080`` or ``local ip``
- Create package using uhu from signed image as per the examples intructions
- Upload package ``zephyr-update.pkg`` on the updatehub dashboard
- Create rollout and the update never fails nor completes
- For update to fail with previously mentioned errors lines 878, 884, 893, 906, 922
 need to be commented out, can be found at ``~zephyproject/zephyr/subsys/mgmt/updatehub/updatehub.c``
  

**Expected behaviour**
- Update never completing, fail at a json parse which can be seen through a minicom -D /dev/ttyACM0
- If code commented out update failing and reporting back to updatehub that it failed

**Impact**
Due to this issue I haven't been able to continue on with working with this sample

**Environment**
 - OS: Linux
 - Toolchain: Zephyr SDK 
 - Zephyr version 3.5.0

nashif, 2024-02-27:
@nandojve can you please take a look?

nandojve, 2024-02-27:
> @nandojve can you please take a look?

Hi @nashif ,

Yes, It is on the queue.

nandojve, 2024-04-23:
Hi @RafalHalik,

This seems to be related to changes introduced by #50816. We need to understand why this is happening to fix it.

closed 2024-05-02 by carlescufi

nandojve, 2024-05-03:
Hi @RafalHalik ,

With the #72300 the UpdateHub is fully operational again.

RafalHalik, 2024-05-07:
Hey @nandojve 

That's great news thankyou for looking into this

nandojve, 2024-05-07:
Hi @RafalHalik ,

The backport is already scheduled for 3.5/3.6. I'll add the missing parts and fix commit msgs soon.
Thank you for reporting : )

PR #72036 (2024-05-02): mgmt: updatehub: Fix json arrays
After the changes introduced by #50816 the UpdateHub could not decode anymore the JSON object. This introduce missing parsing definitions to allow JSON parser undertood the correct UpdateHub probe object.

Fixes #69297

Before evaluate it is recommended take a look at [UpdateHub zephyr guide](https://docs.updatehub.io/zephyr-project/zephyr-project-guide) and make sure your device have a network connection. Remember that by default UpdateHub sample get IP automatically.

Steps to Evaluate:

Run updatehub-ce docker
```
sudo docker run -it -p 8080:8080 -p 5683:568 [... cut]
commit 5fb62ca9 (2024-05-02): mgmt: updatehub: Fix json arrays

After the changes introduced by #50816 the UpdateHub could not decode
anymore the JSON object. This introduce missing parsing definitions
to allow JSON parser undertood the correct UpdateHub probe object.

Fixes #69297
commit 46523e56 (2024-05-07): mgmt: updatehub: Fix mark for update

This fixes compatibility with recent bootutils API.

Fixes #69297

(cherry picked from commit 94cd46d6efe2be3b0beea8cd470297b6dcfbd913)
commit 7250b238 (2024-05-07): mgmt: updatehub: Fix json arrays

After the changes introduced by #50816 the UpdateHub could not decode
anymore the JSON object. This introduce missing parsing definitions
to allow JSON parser undertood the correct UpdateHub probe object.

Fixes #69297

(cherry picked from commit 5fb62ca960e1b5ad6bd31a9c81231cb455bd2559)