zephyrproject-rtos/zephyr #69298: doc: Documentation issue in 'develop/application/VERSION'

schnitzlein, 2024-02-21:
Application VERSION File parsing

throws this error:

`zephyr/include/generated/app_version.h:19:27: error: too many decimal points in number`

Unfortunatly I am not able to generate a VERSION, and the usage in main is used also as described.
I simply want to keep tracking on progress and mark floating hardware with version information using the VERSION Feature.
But it simply is not working.

<< The VERSION File is written as described in: https://docs.zephyrproject.org/latest/build/version/index.html#app-version-details >>
<< With simply data filed>>
`VERSION_MAJOR = 2
VERSION_MINOR = 0
PATCHLEVEL = 0
VERSION_TWEAK = 0
EXTRAVERSION = develop`

**Environment is Linux with latest Zephyr rtos**

* Page: `develop/application/index`
* Version: main
* SHA-1: d2001a6fab85af0bde533e5613f12f98016d11ec

nordicjm, 2024-02-21:
Tested on `main` with the above described file and working fine

schnitzlein, 2024-02-22:
It is the "APP_BUILD_VERSION" define which have a unquoted string, this throws the error.
I use now the "APP_VERSION_STRING" it works fine.
Thanks

closed 2024-02-22 by schnitzlein

nordicjm, 2024-02-22:
> It is the "APP_BUILD_VERSION" define which have a unquoted string, this throws the error. I use now the "APP_VERSION_STRING" it works fine. Thanks

That is because `APP_BUILD_VERSION` is not quoted, to use that as a string you would need to use `STRINGIFY(APP_BUILD_VERSION)`

schnitzlein, 2024-03-05:
Thanks this actually worked! :) 
I couldn't figure out how to make the unquoted variable to a String.
The STRINGIFY Macro works like a charm, this is exactly what I was looking for.

You saved me from writing again a solution and inventing the wheel for the 100 times.
Now I can throw my dirty solution away where I basically build one with pristine, copy the app_version.h 
append my variables to it.
Build again without pristine build and flash this.
It was a really bad solution.