zephyrproject-rtos/zephyr #69313: RISC-V memory protection: Incorrect configuration of PMP when enabling CONFIG_PMP_STACK_GUARD and CONFIG_USERSPACE

ASaltFishy, 2024-02-22:
**Describe the bug**
Bug code is as follows.
When I enabled  CONFIG_PMP_STACK_GUARD and CONFIG_USERSPACE in userspace hello world demo, both `z_riscv_pmp_stackguard_prepare` and `z_riscv_pmp_usermode_prepare` configured PMP entries starting from the entry specified in `z_riscv_pmp_thread_init`. Therefore, the PMP entries configured in` z_riscv_pmp_usermode_prepare` will override the entries configured in `z_riscv_pmp_stackguard_prepare`, causing failure of user stack guard.
``` 
// Function arch_user_mode_enter In arch/riscv/core/thread.c
#ifdef CONFIG_USERSPACE
FUNC_NORETURN void arch_user_mode_enter(k_thread_entry_t user_entry,
					void *p1, void *p2, void *p3)
{
	...

#ifdef CONFIG_PMP_STACK_GUARD
	/* reconfigure as the kernel mode stack will be different */
	z_riscv_pmp_stackguard_prepare(_current);
#endif

	/* Set up Physical Memory Protection */
	z_riscv_pmp_usermode_prepare(_current);
	z_riscv_pmp_usermode_enable(_current);

      ...
}

#endif /* CONFIG_USERSPACE */
```

arch: RISC-V
board: qemu_riscv64 
app: [Userspace Hello World Demo](https://docs.zephyrproject.org/3.5.0/samples/userspace/hello_world_user/README.html)
prj.conf:
```
CONFIG_USERSPACE=y
CONFIG_ASSERT=y
CONFIG_APPLICATION_DEFINED_SYSCALL=y

CONFIG_SMP=y
CONFIG_LOG=y
CONFIG_LOG_MODE_MINIMAL=y
CONFIG_MAIN_STACK_SIZE=2560
CONFIG_HW_STACK_PROTECTION=y
```

**Environment:**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Version:3.5.0

**To Reproduce**
Steps to reproduce the behavior:
Directory: zephyr-spmp/samples/userspace/hello_world_user
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=qemu_riscv64 ..
3. ninja run
4. Find error

fkokosinski, 2024-02-26:
Hi @ASaltFishy, thanks for filing this report.

I can't seem to reproduce the issue that you've described here. The steps you've described in `To reproduce` weren't sufficient to trigger it, so I've additionally tried:
1. building and running with: 
```west build -p -b qemu_riscv64 samples/userspace/hello_world_user -t run -- -DCONFIG_HW_STACK_PROTECTION=y```
2. modifying the project config to match what you've pasted in the snipper

None of these would trigger this issue either. Could you describe the steps necessary to reproduce it in more detail? Also, please note that Zephyr v3.6.0 has just been released, and v3.5.0 is not LTS (I've tried reproducing this issue on both versions, though). Thanks

ASaltFishy, 2024-02-27:
Hi @fkokosinski, thanks for comment.

As the code shown below, the **index** at function `z_riscv_pmp_stackguard_prepare` and `z_riscv_pmp_usermode_prepare` is the same. When calling these two functions one after the other, the configuration in the latter will override that of the former.
``` c
/* 
      arch/riscv/core/pmp.c
*/
void z_riscv_pmp_stackguard_prepare(struct k_thread *thread)
{
	unsigned int index = z_riscv_pmp_thread_init(PMP_M_MODE(thread));
	uintptr_t stack_bottom;

	/* make the bottom addresses of our stack inaccessible */
	stack_bottom = thread->stack_info.start - K_KERNEL_STACK_RESERVED;
#ifdef CONFIG_USERSPACE
	if (thread->arch.priv_stack_start != 0) {
		stack_bottom = thread->arch.priv_stack_start;
	} else if (z_stack_is_user_capable(thread->stack_obj)) {
		stack_bottom = thread->stack_info.start - K_THREAD_STACK_RESERVED;
	}
#endif

void z_riscv_pmp_usermode_prepare(struct k_thread *thread)
{
	unsigned int index = z_riscv_pmp_thread_init(PMP_U_MODE(thread));

	LOG_DBG("pmp_usermode_prepare for thread %p", thread);

	/* Map the usermode stack */
	set_pmp_entry(&index, PMP_R | PMP_W,
		      thread->stack_info.start, thread->stack_info.size,
		      PMP_U_MODE(thread));

	thread->arch.u_mode_pmp_domain_offset = index;
	thread->arch.u_mode_pmp_end_index = index;
	thread->arch.u_mode_pmp_update_nr = 0;
        ...
}

```
However, I found that stack guard is not required to catch stack overflows in user mode on https://docs.zephyrproject.org/latest/kernel/usermode/memory_domain.html. 
So the configuration of pmp in above text is correct——no pmp entry is needed for user stack guard in usermode and overwriting in `z_riscv_pmp_usermode_prepare` will be ragarded as the correct behavior. (By the way, why don't we just delete `z_riscv_pmp_stackguard_prepare(_current);` line as the config in it will be overwritten latter and not used in userspace? )

I will close this issue as there's no bug in it. Thanks for your support)!

closed 2024-02-27 by ASaltFishy