zephyrproject-rtos/zephyr #69317: hwmv2: Mismatch between SOC_SERIES name and value

nordicjm, 2024-02-22:
For hwmv2, there should be a match between the names and values of SoCs, this should take the form of:
```
config SOC_SERIES
        default "nrf51" if SOC_SERIES_NRF51
        default "nrf52" if SOC_SERIES_NRF52
```

The following is wrong:
```
config SOC_SERIES
        default "nrf51" if SOC_SERIES_NRF51X
        default "nrf52" if SOC_SERIES_NRF52X
```

@manuargue has helpfully put together a script which lists the mismatches for SOC_SERIES:

- soc/andes/ae350/Kconfig.soc @nordicjm
  - [ ] default string != option name ('ae350', 'ANDES_AE350')
- soc/nordic_nrf/Kconfig.soc @nordicjm
  - [ ] default string != option name ('nrf51', 'NRF51X')
  - [ ] default string != option name ('nrf52', 'NRF52X')
  - [ ] default string != option name ('nrf53', 'NRF53X')
  - [ ] default string != option name ('nrf54h', 'NRF54HX')
  - [ ] default string != option name ('nrf54l', 'NRF54LX')
  - [ ] default string != option name ('nrf91', 'NRF91X')
- soc/st/stm32/stm32f0/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32f0', 'STM32F0X')
- soc/st/stm32/stm32u5/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32u5', 'STM32U5X')
- soc/st/stm32/stm32h5/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32h5', 'STM32H5X')
- soc/st/stm32/stm32l4/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32l4', 'STM32L4X')
- soc/st/stm32/stm32l5/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32l5', 'STM32L5X')
- soc/st/stm32/stm32wl/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32wl', 'STM32WLX')
- soc/st/stm32/stm32h7/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32h7', 'STM32H7X')
- soc/st/stm32/stm32c0/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32c0', 'STM32C0X')
- soc/st/stm32/stm32l1/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32l1', 'STM32L1X')
- soc/st/stm32/stm32g4/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32g4', 'STM32G4X')
- soc/st/stm32/stm32f1/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32f1', 'STM32F1X')
- soc/st/stm32/stm32f3/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32f3', 'STM32F3X')
- soc/st/stm32/stm32f4/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32f4', 'STM32F4X')
- soc/st/stm32/stm32f2/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32f2', 'STM32F2X')
- soc/st/stm32/stm32wb/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32wb', 'STM32WBX')
- soc/st/stm32/stm32mp1/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32mp1', 'STM32MP1X')
- soc/st/stm32/stm32f7/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32f7', 'STM32F7X')
- soc/st/stm32/stm32l0/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32l0', 'STM32L0X')
- soc/st/stm32/stm32g0/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32g0', 'STM32G0X')
- soc/st/stm32/stm32wba/Kconfig.soc @erwango
  - [ ] default string != option name ('stm32wba', 'STM32WBAX')
- soc/microchip/mec/Kconfig @fabiobaltieri
  - [ ] default string != option name ('mec152x', 'MEC1501X')
- soc/raspberry_pi/rp2/Kconfig.soc @nordicjm
  - [ ] default string != option name ('rp2', 'RP2XXX')
  - [ ] default string != option name ('rp2040', 'RP2XXX')
- soc/intel/intel_adsp/ace/Kconfig.soc @golowanow
  - [ ] default string != option name ('ace', 'INTEL_ACE')
  - [ ] default string != option name ('intel_ace15_mtpm', 'INTEL_ACE')
- soc/intel/intel_adsp/cavs/Kconfig.soc @golowanow
  - [ ] default string != option name ('cavs', 'INTEL_ADSP_CAVS')
- soc/xilinx/zynq7000/xc7zxxx/Kconfig.soc @nordicjm
  - [ ] default string != option name ('xc7zxxx', 'XILINX_XC7ZXXX')
  - [ ] default string != option name ('xc7zxxxs', 'XILINX_XC7ZXXXS')
- soc/nxp/s32/s32k1/Kconfig.soc @manuargue
  - [ ] default string != option name ('s32k1', 'S32K1XX')
- soc/nxp/s32/s32k3/Kconfig.soc @manuargue
  - [ ] default string != option name ('s32k3', 'S32K3XX')
- soc/nxp/kinetis/k2x/Kconfig.soc @dleach02
  - [ ] default string != option name ('k2x', 'KINETIS_K2X')
- soc/nxp/kinetis/ke1xf/Kconfig.soc @dleach02
  - [ ] default string != option name ('ke1xf', 'KINETIS_KE1XF')
- soc/nxp/kinetis/k8x/Kconfig.soc @dleach02
  - [ ] default string != option name ('k8x', 'KINETIS_K8X')
- soc/nxp/kinetis/kwx/Kconfig.soc @dleach02
  - [ ] default string != option name ('kwx', 'KINETIS_KWX')
- soc/nxp/kinetis/k6x/Kconfig.soc @dleach02
  - [ ] default string != option name ('k6x', 'KINETIS_K6X')
- soc/nxp/kinetis/kl2x/Kconfig.soc @dleach02
  - [ ] default string != option name ('kl2x', 'KINETIS_KL2X')
- soc/nxp/kinetis/kv5x/Kconfig.soc @dleach02
  - [ ] default string != option name ('kv5x', 'KINETIS_KV5X')
- soc/arm/designstart/Kconfig.soc @nordicjm
  - [ ] default string != option name ('designstart', 'ARM_DESIGNSTART')

nordicjm, 2024-02-22:
@fabiobaltieri/@jvasanth1 I think yours is the exception

erwango, 2024-02-22:
@nordicjm This is doable but my initial question is what is the use of Kconfig symbol `SOC_SERIES` ?

nordicjm, 2024-02-22:
> @nordicjm This is doable but my initial question is what is the use of Kconfig symbol `SOC_SERIES` ?

It should match up with what is in soc.yml. Can think of it as how boards no longer have `bool` in their `Kconfig.<board>` files but the build system generates it automatically, that too should be possible for SOC fields in future e.g. `SOC`, `SOC_FAMILY`, `SOC_SERIES`.

Actually here is a good example, take this `soc.yml` file https://github.com/zephyrproject-rtos/zephyr/blob/collab-hwm/soc/aspeed/soc.yml:
```
series:
- name: ast10x0
  socs:
  - name: ast1030
```
and `Kconfig.soc` files https://github.com/zephyrproject-rtos/zephyr/blob/collab-hwm/soc/aspeed/Kconfig.soc and https://github.com/zephyrproject-rtos/zephyr/blob/collab-hwm/soc/aspeed/ast10x0/Kconfig.soc:
```
config SOC_FAMILY_ASPEED
	bool

config SOC_FAMILY
	default "aspeed" if SOC_FAMILY_ASPEED

config SOC_SERIES_AST10X0
	bool
	select SOC_FAMILY_ASPEED
	help
	  Enable support for ASPEED AST10X0 series

config SOC_AST1030
	bool
	select SOC_SERIES_AST10X0
	help
	  AST1030

config SOC_SERIES
	default "ast10x0" if SOC_SERIES_AST10X0

config SOC
	default "ast1030" if SOC_AST1030
```
The soc.yml file here is wrong because it defines a soc series and soc but does not define a soc family, so if the build system parses all soc.yml files and generated all the `SOC_FAMILY_<X>` symbols automatically then this would fail for this board because the soc.yml file does not define it when it should, indicating an error

tejlmand, 2024-02-22:
Please see this PR which introduces SoC name checking in existing compliance check: https://github.com/zephyrproject-rtos/zephyr/pull/69309

tejlmand, 2024-02-22:
>  that too should be possible for SOC fields in future e.g. SOC, SOC_FAMILY, SOC_SERIES.

See reason here why this haven't been implemented so far: https://github.com/zephyrproject-rtos/zephyr/pull/69309#issuecomment-1958907445

golowanow, 2024-02-22:
> soc/intel/intel_adsp/ace/Kconfig.soc 
> default string != option name ('intel_ace15_mtpm', 'INTEL_ACE')

@nordicjm 
this one is for toolchain selection

https://github.com/zephyrproject-rtos/zephyr/blob/d6cb9c967227b05e59a3f8e5a619fa695c174594/soc/intel/intel_adsp/ace/Kconfig.soc#L13-L14

fabiobaltieri, 2024-02-22:
> @fabiobaltieri @jvasanth1 I think yours is the exception here and the name is incorrect

yeah that one gave me some head scratching, initially tried to rename the directory to match and then bumped into a dependency with the HAL `modules/hal/microchip/mec/CMakeLists.txt` and figure I'd leave it alone, @nordicjm do you think it's worth fixing it now a the cost of having a dependency on a hal change?

erwango, 2024-02-22:
@nordicjm STM32 https://github.com/zephyrproject-rtos/zephyr/pull/69325

nordicjm, 2024-02-22:
> yeah that one gave me some head scratching, initially tried to rename the directory to match and then bumped into a dependency with the HAL `modules/hal/microchip/mec/CMakeLists.txt` and figure I'd [...]

HAL update is fine, think we have about 5 or 6 repos that have changes for hwmv2 that will need merging

fabiobaltieri, 2024-02-22:
> HAL update is fine, think we have about 5 or 6 repos that have changes for hwmv2 that will need merging

Cool, @jvasanth1 @albertofloyd is this something you could look into? I'm OOO at the moment.

fabiobaltieri, 2024-02-27:
https://github.com/zephyrproject-rtos/zephyr/pull/69528 for the MEC stuff

ajf58, 2024-12-17:
I don't think this is an issue for Raspberry Pi's SoCs on the `HEAD` of `main` https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/raspberrypi/rp2xxx/Kconfig.soc

butok, 2026-01-16:
> [@mmahadevan108](https://github.com/mmahadevan108) [@butok](https://github.com/butok) it's been almost 2 years, can you look at the above and submit PRs fixing your boards/socs please

I see it first time :)

dcpleung, 2026-01-16:
https://github.com/zephyrproject-rtos/zephyr/pull/102421 for `intel_adsp`.

closed 2026-01-29 by carlescufi

nordicjm, 2026-02-03:
@butok any update on submitting a fix?

butok, 2026-02-04:
> [@butok](https://github.com/butok) any update on submitting a fix?

Not yet. But as you are pushing ;) I have to look at it.

PR #69335 (2024-02-23): hwmv2: soc: intel_adsp: Align SoC Series name and value
Align `INTEL_ADSP` SoC family to the new HWMv2 compliance check requirements to match SoC series name and value for:
* `cavs` -> `intel_adsp_cavs`
* `ace` -> `intel_adsp_ace` 

resolves #69317