zephyrproject-rtos/zephyr #69318: CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC is not right for acrn_defconfig

bensenq, 2024-02-22:
**Describe the bug**
defconfig for x86 acrn boards declares:
```
# boards/x86/acrn/acrn_defconfig
CONFIG_APIC_TSC_DEADLINE_TIMER=y
CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=25000000
``
This is definitely not right as TSC frequency is the CPU Core Freq which should not be 25MHz small.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? x86
 - What have you tried to diagnose or workaround this issue? Runing a sleep test app.

**Impact**

Time/timeout not work correctly.

jhedberg, 2024-02-22:
@bensenq would you like to submit a pull request to fix this?

jhedberg, 2024-02-22:
@nashif @dcpleung @najumon1980 fyi

najumon1980, 2024-02-23:
Yes this is wrong and TSC APIC timer frequency will be the processor’s bus clock or core crystal clock frequency divided by the value specified in the divide configuration register. But core clock frequency varies from platform to platform and even with different SKUs (for Intel platforms it usually from ~1GHz to ~4GHz).  But even with same platform type (for different SKU) this can be changed. **So best way is read via CPUID/MSR and then configure dynamically instead of kernel/app using via static macro CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC.** But this might need a wider discussion.

najumon1980, 2024-02-23:
it seems dynamic reading supported if CONFIG_TIMER_READS_ITS_FREQUENCY_AT_RUNTIME enabled. But looks currently it not implemented for APIC timer driver. Also the kernel/test codes still using CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC instead of using sys_clock_hw_cycles_per_sec() and this also need to fix.

bensenq, 2024-03-01:
> @bensenq would you like to submit a pull request to fix this?

why not just use HPET or APIC Timer? It seems to be more stable.
```
# TSC on this board is 1.9 GHz, HPET and APIC are 19.2 MHz
config SYS_CLOCK_HW_CYCLES_PER_SEC
        default 1900000000 if APIC_TSC_DEADLINE_TIMER
        default 1900000000 if APIC_TIMER_TSC
        default 19200000
```

najumon1980, 2024-03-02:
Created a PR:https://github.com/zephyrproject-rtos/zephyr/pull/69705 for the issue and please have a look:

najumon1980, 2024-03-02:
> why not just use HPET or APIC Timer? It seems to be more stable.
>
> ```
> # TSC on this board is 1.9 GHz, HPET and APIC are 19.2 MHz
> config SYS_CLOCK_HW_CYCLES_PER_SEC
>         default 1900000000 if [...]

APIC/TSC timer clock frequency will be varied with platform to platform as well SKUs within same platforms. So we need to use run time support for Intel platforms.

closed 2024-05-04 by jhedberg