zephyrproject-rtos/zephyr #69320: doc: Documentation issue in 'services/sensing/index'

bmihelcic, 2024-02-22:
**Describe the bug**

Code snippet under [this paragraph](https://docs.zephyrproject.org/latest/services/sensing/index.html#sensor-types-and-instance) has a typo:

sensing_open_sensor_by_dt(DEVICE_DT_GET(**DT_NODELABLE**(base_accel)), cb_list, handle);

**Environment**

* Page: `services/sensing/index`
* Version: main
* SHA-1: 97b9bce995c0df59fe7390650e13fba5b062e5d9

kartben, 2024-02-22:
Thanks for the report! Would certainly welcome your first PR, after your first issue ;-)

bmihelcic, 2024-02-22:
It will be my pleasure.
Opened a PR [#69357](https://github.com/zephyrproject-rtos/zephyr/pull/69357)
Let me know if I missed some conventions.

closed 2024-02-27 by aescolar