zephyrproject-rtos/zephyr #69329: hwmv2: west boards output with identifiers is a python array to wrong pipe

nordicjm, 2024-02-22:
It seems that `west boards --format "{identifiers}"` gives a python array output:
```
['stm32wb5mmg/stm32wb55xx']
['em_starterkit/emsk_em7d', 'em_starterkit/emsk_em9d', 'em_starterkit/emsk_em11d']
['emsdp/emsdp_em4', 'emsdp/emsdp_em5d', 'emsdp/emsdp_em6', 'emsdp/emsdp_em7d', 'emsdp/emsdp_em7d_esp', 'emsdp/emsdp_em9d', 'emsdp/emsdp_em11d']
['hsdk/arc_hsdk', 'hsdk/arc_hsdk/2cores']
['hsdk4xd/arc_hsdk4xd']
['iotdk/arc_iot']
['nsim/nsim_em', 'nsim/nsim_em7d_v22', 'nsim/nsim_em11d', 'nsim/nsim_hs', 'nsim/nsim_hs/sram', 'nsim/nsim_hs/smp', 'nsim/nsim_hs/flash_xip', 'nsim/nsim_hs/mpuv6', 'nsim/nsim_hs/hostlink', 'nsim/nsim_hs5x', 'nsim/nsim_hs5x/smp', 'nsim/nsim_hs5x/smp/12cores', 'nsim/nsim_hs6x', 'nsim/nsim_hs6x/smp', 'nsim/nsim_hs6x/smp/12cores', 'nsim/nsim_sem', 'nsim/nsim_sem/mpu_stack_guard', 'nsim/nsim_vpx5']
['robokit1/same70q21b']
['tlsr9518adk80d/tlsr9518']
```

Which does not go to stdout, it does not include any hwmv1 boards either

nordicjm, 2024-02-23:
Yes the actual issue here is the output of `west boards --format="{name} {identifiers}"` is wrong, identifiers is outputting board names along with identifiers when it should just be identifiers, so that is the bug that needs fixing

aescolar, 2024-03-22:
Reopened as the commit that fixed this was reverted

commit 66b475a3 (2024-03-22): scripts: remove board name from the qualifiers in list_boards.py

Fixes: #69329

The board name was printed as part of printing board qualifiers because
those was being concatenated in the `board_v2_qualifiers()` method.

Keep the qualifiers separated from the board name and let the caller
concatenate the strings when required.

Completion scripts are also updated to handle the corrected behaviour.
PR #70619 (2024-03-26): Board listing fixes
Re-applies #69904 and re-fixes #70614

----

* Error out if multiple boards use the same name
* Ignore duplicate roots in list_boards/hardware

---

(edit by aescolar) Fixes: #69329
commit 88d8b238 (2024-03-26): scripts: remove board name from the qualifiers in list_boards.py

Fixes: #69329

The board name was printed as part of printing board qualifiers because
those was being concatenated in the `board_v2_qualifiers()` method.

Keep the qualifiers separated from the board name and let the caller
concatenate the strings when required.

Completion scripts are also updated to handle the corrected behaviour.

(cherry picked from commit 66b475a3aa4d73f4ddd5be5a7a1e0f3c7028e539)
closed 2024-03-26 by nashif