zephyrproject-rtos/zephyr #69343: System freezing in case of k_sleep() usage

SergeyStarovoitov, 2024-02-22:
**Describe the bug**

_Application_: commercial project.

_Short Description of the Application:_ accumulation of the BLE advertising data and sending it to another chip by UART (for interconnection between chips, we use nRF RPC with a custom transport layer). The system is protected from freezing by `task_wdt`.

_System behaviour_

1. After some period of time (it depends; it might be just an hour or even a few days), the system is partially frozen when calling `k_msleep()`.
2. The standard `task_wdt` feature does not reboot the MCU in this case.

**To Reproduce**
Unknown. The issue may appear in just an hour of uptime or a few days of uptime.

**Expected behavior**
The threads and threads watchdog shouldn't freeze permanently in case of `k_sleep()` usage.

**Impact**
Showstopper. The system is permanently stuck (a manual reboot is needed).

**Environment**
 - MCU: nRF52840
 - Toolchain: nRF Connect SDK 2.3.0 / 2.5.2 (the same behavior on both versions)

**Additional Notes**

 * Threads that use `k_msleep()` totally freeze (we see it because it has a light indication).
 * `task_wdt` doesn't reboot the system (it's strange because we have a task watchdog for each user-defined thread).
 * Threads that use, for example, `k_sem_take()` work correctly because they are given after some period of time by `k_sem_give()`.
 * UART communication works fine (we use the UART async API).
 * Via UART, we transmit a timestamp, which is gotten by `k_uptime_get()`, and the timestamp is changing (so at least the RTC1 is counting).
 * Also, we have one thread that calls `sys_reboot()` in the case of the specific UART command, but before it, it's waiting 1 second using `k_msleep()`. So this thread is also getting stuck if another thread is already stuck as described above (we send the command via UART, but the system doesn't reboot).

Since the nRF Connect SDK 2.5.2 was developed for Zephyr 3.5.0 (3.4.99-ncs1). We are concerned that migrating to upstream main may present issues since nRF SDK was not tested with this build.

nordicjm, 2024-02-22:
> Toolchain: nRF Connect SDK 2.3.0 / 2.5.2 (the same behavior on both versions)

This is for reporting problems in zephyr, not ncs, problems in ncs should be reported on https://devzone.nordicsemi.com/ unless you have reproduced the issue using native upstream zephyr. If you can then please reopen the issue

closed 2024-02-22 by nordicjm

Geliozzz, 2025-05-23:
I use Zephyr OS build v4.1.0-6-g6e19c9aa6203 and have the same problem with k_msleep(). 
@SergeyStarovoitov , thank you for report.