zephyrproject-rtos/zephyr #69372: Atmel SAM0 ADC not working

cperkulator, 2024-02-23:
**Describe the bug**
Setting up analogs on a custom board using an ATSAMD21G part.

reduced DTS:

      zephyr,user {
	      io-channels = <&adc 0>;
      };
      
      &adc {
	  #address-cells = <1>;
	  #size-cells = <0>;
          status = "okay";
      
	      channel@0 {
		      reg = <0>;
		      zephyr,gain = "ADC_GAIN_1";
		      zephyr,reference = "ADC_REF_INTERNAL";
		      zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
		      zephyr,resolution = <12>;
		      zephyr,input-positive = <0>;
              status = "okay";
	  };
  };
Through some debugging I've found that the adc_sam0_init function calls pinctrl_lookup_state which returns -ENOENT and causes everything to break.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
    - Atmel SAM0
 - What have you tried to diagnose or workaround this issue?
    - debugged code, it appears that the pinctrl function is not working
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. configure ADC 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

attie-argentum, 2024-02-27:
Hi @cperkulator, thanks for the report... apologies for my delayed response.

Have you provided the pinctrl info as well? (example below)

```c
&adc {
    #address-cells = <1>;
    #size-cells = <0>;

    status = "okay";

    pinctrl-0 = <&adc_default>;
    pinctrl-names = "default";

    adc_vbat: channel@6 {
        reg = <6>;
        zephyr,gain = "ADC_GAIN_1";
        zephyr,reference = "ADC_REF_VDD_1";
        zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
        zephyr,resolution = <12>;
        zephyr,input-positive = <6>;
        zephyr,vref-mv = <ADC_REF_VDD_1_MV>;
    };

    adc_vsys: channel@17 {
        reg = <17>;
        zephyr,gain = "ADC_GAIN_1";
        zephyr,reference = "ADC_REF_VDD_1";
        zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
        zephyr,resolution = <12>;
        zephyr,input-positive = <17>;
        zephyr,vref-mv = <ADC_REF_VDD_1_MV>;
    };
};

&pinctrl {
    adc_default: adc_default {
        group1 {
            pinmux =
                <PA6B_ADC_AIN6>,
                <PA9B_ADC_AIN17>;
        };
    };
};
```

nandojve, 2024-04-28:
Hi @cperkulator,
Did comments from @attie-argentum solve your issue?

PeeJay, 2024-05-03:
!@#$% I just spend 2 days chasing down the same problem, no pinctrl! Doh! Added it in and it works.
None of the (sam) ADC sample overlays that I looked at had it, I just now see that some do and some don't.

nandojve, 2024-05-03:
Hi @PeeJay ,

Good to know that it worked. We encourage people to contribute so feel free to send overlays patches to help keep everything in shape. Those are the good ones to start contribute : )

closed 2024-05-03 by nandojve