zephyrproject-rtos/zephyr #69383: NVS: add support for flash devices without erase function

rghaddab, 2024-02-23:
## Introduction

This RFC proposes a new method of recognizing  erased ATEs without having to erase entire sectors to handle NVS reads/writes operations.
This methhod is based on adding what is called a "cycle counter" that is incremented, for a given sector, each time we want to delete all the ATEs in that sector.

### Problem description

NVS filesystem operations are based on the fact that all memory cells are erased and in a known state before doing a write operation.
For non erasable flash devices like RRAMs this operation is translated into memory writes which is a costly operation in terms of memory cells life time.
One of the biggest advantage of using NVS for these devices is its wear leveling capabilities, but the amount of write operations that are added in order to erase sectors is affecting the life time of memory cells.
For normal operations, comparing flash devices that have normal erase functions with flash devices that should do write operations to erase memory cells, we get almost double of the number of write operations for the latter. Which means half of the life time expectancy.

### Proposed change

The proposed change will add a "cycle counter" parameter for each ATE that defines if a given ATE is still valid (cycle_counter == sector_cycle_counter) or erased/non valid (cycle_counter != sector_cycle_counter).
The change is also retro-compatible as it keeps the old mechanism of recognizing the erased cells using erase_value.

## Detailed RFC

### Proposed change (Detailed)

The main purpose of the proposed changes is to get rid of erasing the sectors and all parts of the code that compares the memory cells to a know value (erase_value) for all the NVS filesystem operations.
It also introduce these changes only for devices that needs this kind of feature while keeping the same operations for all other devices.
Please note that this new proposed method of recognizing erased cells could work as well for all the types of flash devices (erasable and non erasable). But it is limited for the moment for only non erasable devices to limit any risk of breaking systems that run using the old method.

**Global changes :**
- To keep the "nvs_ate" structure size unchanged we use the member "part" which is unused and we replace it by the cycle_counter
struct nvs_ate {
	...
	uint8_t part; => uint8_t cycle_cnt;
	...
} __packed;

- for flash_parameters we introduce a new flag : no_erase which is only meaningful for non erasable devices and defines whether a flash device is non erasable
- two CONFIGs are added : 
config NVS_FLASH_HAS_NO_ERASE
config NVS_FLASH_HAS_ERASE

These configs defines whether or not the NVS filesystem needs to use old operation method, new operations or both of them.
CONFIG_NVS_FLASH_HAS_ERASE is by default y and is defined when the other config NVS_FLASH_HAS_NO_ERASE is not defined.

**nvs.c changes :**
A new special ATE is introduced called empty ATE.
This ATE is the one holding the leading cycle counter for a given sector.
When a sector is being erased the empty ATE cycle counter is increased to make all the other ATEs in that sector invalid
When a sector is closed the close ATE cycle counter will get the same cycle counter as the empty ATE

The sector ATEs organization should look like this at anytime :
    _________________________________________________________
    |                  |                            |                    |                      |                    |
    |... DATA ... | GC_DONE ATE  | ... ATEs ...   | Empty ATE   | Close ATE  |
    |_________|______________|__________|___________|__________|

A sector can have multiple status :
- Open sector : empty_ate.cycle_cnt == close_ate.cycle_cnt + 1
- Closed sector : empty_ate.cycle_cnt == close_ate.cycle_cnt
- Never used Sector : Empty ATE or Close ATE invalid

In the proposed changes the startup function for non erasable devices is "nvs_startup_noerase"
This function has almost the same operations as the old nvs_startup function but for readability purpose I prefer to separate it in a separate function as there are some differences between operation for erasable and non erasable devices.

### Dependencies

These changes depends on this [PR](https://github.com/zephyrproject-rtos/zephyr/pull/67687/) which should introduce the flag to reconize erasable and non erasable devices

Laczen, 2024-02-23:
@rghaddab, thank you for this RFC. It would be a nice enhancement of nvs could be used without the need for erasing.

As a first remark i would say the config naming is badly chosen, the `FLASH` should be removed  and it should just be `NVS_ERASE` and `NVS_OVERWRITE`.

Before going through the PR, a question arises about the `empty_ate`, is this being written each time a valid ate is written ? Has it been evaluated that even the write of a `empty_ate` can fail?

rghaddab, 2024-02-23:
> @rghaddab, thank you for this RFC. It would be a nice enhancement of nvs could be used without the need for erasing.
> 
> As a first remark i would say the config naming is badly chosen, the `FLASH` should be removed and it should just be `NVS_ERASE` and `NVS_OVERWRITE`.
I will change the CONFIG names.

> 
> Before going through the PR, a question arises about the `empty_ate`, is this being written each time a valid ate is written ? Has it been evaluated that even the write of a `empty_ate` can fail?
Empty ATE is not written each time a valid ATE is written, otherwise it won't make sense for wear leveling feature.
Empty ATE is only written when erasing a sector or when it is the first time using a sector.
I made sure that each ATE address is written at most once for each sector write cycle.

Regarding the fail of empty_ate write, I also took that into account.
When an empty ATE write fails, it means that trying to erase the sector failed.
On the next startup we have two cases :
- the empty ATE was partially written => the empty ATE is invalid (wrong CRC) => we fix the empty ATE if the close ATE is valid (note that writing the close ATE doesn't happen at the same time as writing the empty ATE, so both ATE writes could not fail at the same time)
- the empty ATE write didn't happen at all => the empty ATE cycle counter still have the close ATE cycle counter => the sector is considered as closed. If we find a GC_DONE in the previous sector it means that a GC has finished and we can erase the sector by adding an empty ATE, otherwise we must start again the GC.

Laczen, 2024-02-26:
@rghaddab, regarding the `empty_ate`, what would happen to the nvs storage solution if the crc of the `empty_ate` would be bad for a different reason (not after an incomplete write) ?

The use of a Kconfig option also limits the use of multiple nvs storage solutions when devices with and without erase are used, how would this be handled ?

rghaddab, 2024-02-26:
> @rghaddab, regarding the `empty_ate`, what would happen to the nvs storage solution if the crc of the `empty_ate` would be bad for a different reason (not after an incomplete write) ?

Whether it is after an incomplete write or for any other reason, at startup NVS always checks for close sector followed by an open one and an empty sector :
If the empty ATE is invalid we fix it at startup when the close ATE is valid :
```
			/* we have to check if the close_ate or
			 * the empty_ate is valid.
			 * If one of them is valid it means that
			 * something wrong happened and either
			 * the close_ate or empty_ate write failed
			 * Let's fix the wrong ATE
			 */
			if (nvs_close_ate_valid(fs, &close_ate)) {
				/* update ate_wra to point to the address
				 * of empty ATE
				 */
				fs->ate_wra = addr - ate_size;
				rc = nvs_fix_empty_ate(fs, close_ate.cycle_cnt);
				if (rc) {
					goto end;
				}
				break;
			}
```
Or if the next sector (the empty one) has invalid empty ATE, then we create a new one.

> 
> The use of a Kconfig option also limits the use of multiple nvs storage solutions when devices with and without erase are used, how would this be handled ?

The Kconfig options can be both enabled. In this case the NVS driver can handle both types of storage based on the flag that is found on flash_parameters :

```
#if defined(CONFIG_NVS_OVERWRITE) && defined(CONFIG_NVS_ERASE)
#define NVS_DEVICE_HAS_NO_ERASE(fs) ((fs)->flash_parameters->no_erase)
#else
#define NVS_DEVICE_HAS_NO_ERASE(fs) IS_ENABLED(CONFIG_NVS_OVERWRITE)
#endif
```

thedjnK, 2025-01-14:
@rghaddab looks like this can be closed too?

closed 2025-01-15 by rghaddab