zephyrproject-rtos/zephyr #69386: LE Audio: bt_bap_broadcast_sink_cb.syncable should provide the BIGInfo

Thalley, 2024-02-23:
**Is your enhancement proposal related to a problem? Please describe.**
The `syncable` callback only provides the encryption field of the BIGInfo, but since the BIGInfo also contains information about SDU sizes and other useful values, it makes sense to provide the entire struct. 

**Describe the solution you'd like**
Modify `bt_bap_broadcast_sink_cb.syncable` to provide the BIGInfo instead of just the encryption field. 

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

ChangNice, 2024-02-27:
Hi, Emil you could assign this issue to me.

Thalley, 2024-05-15:
Implemented by https://github.com/zephyrproject-rtos/zephyr/pull/69729

closed 2024-05-15 by Thalley