zephyrproject-rtos/zephyr #69391: tests: net: NXP: mimxrt1170_evk_cm7: 100M phy does not support PHY_BASICSTATUS_REG read

hakehuang, 2024-02-23:
**Describe the bug**

mimxrt1170_evk_cm7, if run the tests/net/hostname test, there is a hardfault in the end of test SUCCESS, and it happens all the time, since for this case.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 mimxrt1170_evk_cm7, vmu_rt1170
 - What have you tried to diagnose or workaround this issue?
 - we find it is below code issue, if disable this, test case pass without hardfault.

```
	case eth_mcux_phy_state_restart:
		/* Start reading the PHY basic status. */
#if !defined(CONFIG_ETH_MCUX_NO_PHY_SMI)
		ENET_StartSMIRead(context->base, context->phy_addr,
				  PHY_BASICSTATUS_REG,
				  kENET_MiiReadValidFrame);
#endif
```

 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
not a regression but a long exiting hidden issue

**To Reproduce**
Steps to reproduce the behavior:
1. cd tests/net/hostname
2. west build -b mimxrt1170_evk_cm7
3. west flash
4. See error

**Expected behavior**

Test PASS without error

**Impact**

phy restart for mimxrt1170 series

**Logs and console output**

```
*** Booting Zephyr OS build v3.6.0-rc3-5-g53f527cc2dc3 ***
Running TESTSUITE net_hostname
===================================================================
START - test_hostname_get
 PASS - test_hostname_get in 0.001 seconds
===================================================================
START - test_hostname_set
 PASS - test_hostname_set in 0.001 seconds
===================================================================
TESTSUITE net_hostname succeeded

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [net_hostname]: pass = 2, fail = 0, skip = 0, total = 2 duration = 0.002 seconds
 - PASS - [net_hostname.test_hostname_get] duration = 0.001 seconds
 - PASS - [net_hostname.test_hostname_set] duration = 0.001 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: c6a0885e6eb103791a8039b84fb8efeb
PROJECT EXECUTION SUCCESSFUL
E: ***** USAGE FAULT *****
E:   Unaligned memory access
E: r0/a1:  0x80000800  r1/a2:  0x80001da8  r2/a3:  0x40424001
E: r3/a4:  0x61060000 r12/ip:  0x00018510 r14/lr:  0x3000f35f
E:  xpsr:  0x01000000
E: Faulting instruction address (r15/pc): 0x3000a308
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x80001da8 (sysworkq)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: v3.6.0-rc3-11-gdc8dff0fce0

hakehuang, 2024-02-23:
@dleach02

decsny, 2024-02-29:
This driver is going to be deprecated asap before 3.7

decsny, 2024-04-24:
closing as the driver is deprecated

closed 2024-04-24 by decsny